springfall2008/batpred #2291: IBoost continues to run beyond it's planned time

timparker71, 2025-04-25:
My iBoost is all set to run for no more than 60mins when import rates are below 9p. 
The plan is generated correctly with iBoost scheduled at the start of the overnight cheap rate (8.5p 00:30-05:00) for 45mins, however the iBoost switch now stays on till the end of the cheap rate (05:30), then switches off. So it runs for 5 hours rather than the planned 45mins.

Predbat version 8.18.10

Thanks

Tim

[predbat.3.log](https://github.com/user-attachments/files/19906069/predbat.3.log)

[Predbat_debug.yaml.txt](https://github.com/user-attachments/files/19906120/Predbat_debug.yaml.txt)

[image]

springfall2008, 2025-04-25:
Where do you see the iboost running, can you show what you are looking at?

springfall2008, 2025-04-25:
Oh wait I can see it now, appears to be a rounding issue of some sort

timparker71, 2025-04-25:
The highlighted bit from Saturday 00:30.
It was the same last night in the plan, just 45mins of iBoost which is correct, but in reality it didn't switch off till 0530

springfall2008, 2025-04-25:
Fix on 'main' for testing

springfall2008, 2025-04-27:
Fix released

closed 2025-04-27 by springfall2008