springfall2008/batpred #2296: Iboost never come active.

mbuhansen, 2025-04-27:
I have run with you predbat for a month now, i have try every thing to get i boost working. But i cant get iboost active to come TRUE, and predbat don't calculate to use iboost.
I am from denmark, so i a m running with energi data service from mtrab, and have set manual gas rates because i use pellets burner, so the price on 72 p is the price/kwh heat.
I have activate iboost solar, but iboost dont come active when battery i fully charged. i have also set iboost active when export price is below gas rate.

Can you se what is wrong and why my iboost now come active.??

[image]

[predbat_debug.txt](https://github.com/user-attachments/files/19928537/predbat_debug.txt)

[predbat (2).log](https://github.com/user-attachments/files/19928513/predbat.2.log)

springfall2008, 2025-04-27:
Have you tried predbat_iboost_rate_threshold of higher e.g. 500?

mbuhansen, 2025-04-27:
Thank you , no i don't and this was the trick, now predbat want to set iboost active.

[image]

closed 2025-04-27 by mbuhansen