springfall2008/batpred #2299: predbat stop calculate discharge plan.

mbuhansen, 2025-04-27:
After you update apps.yaml to handle inverter_limit_charge and inverter_limitdischarge from sensor, i start to use this.
But When my battery was charge to 100% inverter set the charge rate sensor to 0, when this happen predbat stop to calculate a discharge plan. I have see this behavior a couple off time, to day i change inverter_limit_charge to fixed value again, and predbat calculate now discharge plan .

don't now about this is a bug..??

springfall2008, 2025-04-27:
Please share more details, what is in your apps.yaml?

mbuhansen, 2025-04-27:
here is some picture with fixed value.

[image]
[image]

And here i have change back to sensor use on charge and discharge limit.
but this only happen when charge go to zero.

[image]
[image]

mbuhansen, 2025-04-27:
[debug log.txt](https://github.com/user-attachments/files/19930910/debug.log.txt)
[predbat (3).log](https://github.com/user-attachments/files/19930911/predbat.3.log)

chalfontchubby, 2026-08-20:
Thanks for the report. This issue has been open for over a year with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-08-28).

mbuhansen, 2026-08-21:
No problem is not pending anymore, i close this issue. Thanks.

closed 2026-08-21 by mbuhansen