springfall2008/batpred #2303: No Freeze Export Slots Planned

nbullus, 2025-04-28:
**Describe the bug**
No Freeze Export slots have been planned for some time when there is solar surplus.

**Expected behaviour**
More cost effective to Freeze the battery and export the surplus solar. Instead it charges battery to 100% using this surplus solar and then sells it back to the grid later in the day before the next cheap rate period. This incurs additional charge/discharge losses which could be avoided by just exporting direct.

**Predbat version**

8.19.1

**Environment details**

- GE AIO
- Standard HAOS installer

**Screenshots**

Plan for tomorrow:

[image]

**Log file**
Debig log file provided.

**Predbat debug yaml file**

[predbat_debug_14_35_00.yaml.txt](https://github.com/user-attachments/files/19941062/predbat_debug_14_35_00.yaml.txt)

nbullus, 2025-04-28:
Interesting that when I drop the minimum battery charge rate from 100w to 30w in the Apps.yaml, the Freeze Export slots appear as expected. 

[image]

I would expect that even when set at 100w minimum charge, it should still be Freezing from 07:30 onwards as PV minus house load is still > 100w.

gcoan, 2025-04-28:
I too am seeing less freeze export occurring in the plan with 8.19.1, today the batteries charged to 100% which they didn't need to, as the export ll then need to be exported.  Same appears to be planned for tomorrow:

[image]

I tried changing input_number.predbat_metric_min_improvement_export_freeze from 10p to 5p and it pushes the overnight export back to earlier in the morning, and its now not an export to empty, but still charges to 100%:

[image]

And setting the metric to 2p:

[image]

Personally I'd like to see Predbat discharging the batteries to empty in the morning so that I can trickle charge to reduce clipping,  then charge to a level sufficient to run me through the night,  and freeze export the rest.  Freeze exporting planned for the start and end of the day when the sun (and thus clipping) is less, letting the batteries charge during the peak of the day

debug file:
[predbat_debug.txt](https://github.com/user-attachments/files/19949018/predbat_debug.txt)

nbullus, 2025-05-02:
Just seems random. One day you get get a clock of Freeze Export as expected, the next day it has nothing. Seems to be a problem with consistency.

chalfontchubby, 2026-09-17:
Could you confirm whether this is still an issue? A lot has changed and in order to debug abything now we would need a new debug.yaml please.

If it's fixed (or we don't hear back to the contrary) I'll close this ticket in around a week so we can focus on live issues. Thanks.

nbullus, 2026-09-17:
Unable to confirm as I’m no longer on this tariff.

closed 2026-09-18 by chalfontchubby