springfall2008/batpred #2305: Exclude Consumption from History

buttash, 2025-04-29:
**Is your feature request related to a problem? Please describe.**
We have a kiln and the consumption is monitored using a Shelly PM1 Pro.  I have creatd CSV profiles for various programmes and have written a Pythion script to create a sensor to be picked up by Predbat's load_forecast feature.  I set the hour and the profile and can see that Predbat is pickingt this information up inthe plan.  A single fire can cobnsumed up to 22kWh.
**Describe the solution you'd like**
Since the kiln firing is performed on an ad hoc basis, I find that the Predbat historic plan is incorrect.  If I fire last Tuesday then this Tuesday's consumption is c20kWh over what I'd normally consume.

I'd like to be able to exclude the kiln firing consumption from Predbat's record of historical consumption.  

Since my kiln on a separate sensor, it would be great to have an exclude_consumption in Predbat's app.yaml which takes an optional list of sensors to exclude from the historical consumption.

This option might have a general application - for example a heat pump warming a pool...

gcoan, 2025-04-29:
These features already exist in predbat, you can use car charging filtering to exclude certain load sensors

See https://springfall2008.github.io/batpred/apps-yaml/#car-charging-filtering

buttash, 2025-04-29:
Missed that - I will configure and check it out.  Thanks.

closed 2025-04-29 by buttash