springfall2008/batpred #2307: Load remaining today mismatch with plan

nbullus, 2025-04-29:
**Describe the bug**
I could not see a Predbat 'remaining load today' sensor/attribute available so have created my own template sensor. Unfortunately it fails to tally with the Predbat load remaining in the plan.

Both of the following Predbat sensors have a useful 'today' attribute indicating total load today (assumed to be actual load so far + remaining prediction):

predbat.best_load_energy
predbat_load_energy

Both sensors have the same value in the 'today' attribute of 15.172 :

[image]

When I subtract the GE house energy used today (8.8 kWh) from the above attribute (15.172 kWh) it leaves a bigger value (6.37 kWh) than what load is left remaining in the plan (5.79 kWh):

[image]

**Expected behaviour**

'total predicted load energy today' - 'load energy used so far' = 'remaining load in plan for today'

**Predbat version**

8.19.1

**Environment details**

- GE AIO
- Standard HAOS 

**Screenshots**

See above

**Log file**

Debug file below

**Predbat debug yaml file**

[predbat_debug_17_00_00.yaml.txt](https://github.com/user-attachments/files/19962793/predbat_debug_17_00_00.yaml.txt)

nbullus, 2025-04-29:
> RTFM
> 
> https://springfall2008.github.io/batpred/output-data/#baseline-data
> 
> "NB: All of Predbat's forecasts are from midnight today to the forecast_hours duration (set in apps.yaml) into the future and shouldn't be confused with 'today' figures.
> 
> e.g. predbat.pv_energy is the actual PV energy from midnight today, and for the predicted forecast_hours (typically 48) ahead so will be much larger than sensor.solcast_pv_forecast_today which is today's Solcast PV forecast."

My forecast_hours is also 48 and the sensor does show it going 48 hours, but the 'today' attribute should filter this forecast down to today only (00:00-00:00). The 'today' attribute of the two aforementioned sensors does not span 48 hours otherwise the load value would be way higher than a 15.1 kWh.

gcoan, 2025-04-30:
OK, sorry, I didn't see the bit about attribute you were reading

The difference is almost certainly due to your load scaling.

Predbat calculates a load and a load 10 value, and uses these in calculating the remaining load today.

What you are looking at in the plan is just the load value.  If you turn html debug on you will see the load 10 values as well, and the sum of those should be the load remaining today.

nbullus, 2025-04-30:
> OK, sorry, I didn't see the bit about attribute you were reading
>
> The difference is almost certainly due to your load scaling.
>
> Predbat calculates a load and a load 10 value, and uses these in [...]

Hi Geoff. I thought this might be the answer, but it still doesn't balance even if you include the 'load scaling PV10' (mine is standard 1.1 setting). If I add 10% onto the standard plan values it is still out by 0.5 kWh already today:

Plan Today = 15.125 kWh
Actual Load = 6.1 kWh
Plan Remaining = 8.74 kWh 
Plan Remaining (10%) = 9.61 kWh 

Plan Today minus Actual Load = 9.02 kWh which matches neither the Plan Remaining or if 10% value included

chalfontchubby, 2026-08-20:
Thanks for the report. This issue has been open for over a year with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-08-28).

nbullus, 2026-08-21:
Hi. You can close this now as it relates to remaining plan values which has been recently fixed. Nigel

closed 2026-08-21 by gcoan