springfall2008/batpred #2312: Export Before Off-Peak Rate Stopping Early

nbullus, 2025-04-30:
**Describe the bug**

The nightly export before the start of the next off-peak charge period is stopping the export early with 16% of the battery left instead of the normal 4% SOC keep.

**Expected behaviour**

This has been working fine but intermittently it changes the plan and stops export early. Should export down to 4% as was working consistently on v8.18.12.

Reverted back to v8.8.12 for now as it is a more stable version (see below screenshots).

**Predbat version**

8.19.1

**Environment details**

- GE AIO
- Standard HAOS

**Screenshots**

v8.19.1

[image]

v8.18.12

[image]

**Log file**
 
See below

**Predbat debug yaml file**

[predbat_debug_18_20_00.yaml.txt](https://github.com/user-attachments/files/19981739/predbat_debug_18_20_00.yaml.txt)

springfall2008, 2025-05-02:
Oddly I re-ran on v8.19.1 and it all seems fine

springfall2008, 2025-05-02:
The reason is the two plans are almost identical, one exports more earlier but then that will mean less export during the cheap period due to more charging time required.

nbullus, 2025-05-02:
Seems to change its mind. Some days it’s correct and sticks, other times it flips. Earlier version tends to stick better than latest version. Previous versions always exported earlier down to reserve.

springfall2008, 2025-05-03:
The point is it doesn't really matter which way you go, if you export down to reserve you have to charge more, meaning less time to export during the cheap charge period. At the end of the charge period the cost is the same either way.

chalfontchubby, 2026-08-20:
Thanks for the report. This issue has been open for over a year with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-08-28).

nbullus, 2026-08-21:
Hi. Just close it now as it’s so old and I’m no longer on that tariff so unable to confirm if it’s fixed.

closed 2026-08-21 by chalfontchubby