springfall2008/batpred #2314: No load data

b1nuzz, 2025-04-30:
**Describe the bug**
Predbat is slowly losing load data

**Expected behaviour**
Load data has been stable but over the last month or so, it has started to disappear. The load sensor is reporting with no issues. If I adjust PB to only look at 1 day history, it will offer load data for 24 hours, and then it drops to 0.0 again. If I offer (up to) 7 days, it is immeditely 0.0 in the load data. 

**Predbat version**

Latest

**Environment details**

- GE Hybrid Inverter
- Standard HAOS installer
- Anything else?

**Screenshots**

**Log file**
[predbat (1).log](https://github.com/user-attachments/files/19985034/predbat.1.log)

**Predbat debug yaml file**

gcoan, 2025-04-30:
looking at the logfile would suggest a problem with your source load sensor:

```

2025-04-30 19:50:01.855545: Historical data totals for days [7] are [0.18] - min 0.18
2025-04-30 19:50:01.855616: Warn: Historical day 7 has 5 minutes of gap in the data, filled from 0.18 kWh to make new average 0.18 kWh (percent 100%)
2025-04-30 19:50:01.856731: Historical days now [7] weight [1]
2025-04-30 19:50:01.865250: Today's load divergence 193.78 % in-day adjustment 189.09 % damping 0.95x
2025-04-30 19:50:01.865272: Today's predicted so far 0.48 kWh with 39.23 kWh car/iBoost excluded and 2.48 kWh import ignored and 0 forecast extra.
2025-04-30 19:50:01.865282: Today's actual load so far 7.77 kWh with 0.0 kWh Car/iBoost excluded and 2.23 kWh import ignored.
```

I notice you are only taking data from a single day, a week ago, so if there are problems with that data then it will significantly affect the forecast.  Is it possible to increase days_previous to average over more days?

Can you include a copy of your apps.yaml and show the graph for the sensor being used for house load, showing the data for the past 8 days to confirm it is OK.

lewiswatt, 2025-04-30:
I was out of town for a long weekend and thought the zero load from my system was a symptom of that unroutine behaviour, however it's not recovering, and this post describes exactly what I'm also seeing on my system.

I've not touched any settings within Predbat in many weeks. I double-checked my apps.yaml and confirmed that it was pointed at the correct sensor which is providing history for 14 days.

gcoan, 2025-05-01:
Can you provide an updated logfile and apps.yaml configuration please?

I would double check the car charging configuration in predbat, the sensor that is configured to report car charging load, and the car charging limit because in the first logfile it appeared that a lot of house load was being ignored because predbat has treated it as car charging

b1nuzz, 2025-05-01:
[image]
[image]

[image]

IMG - Current plan (with 7 days of load data) showing no load data. 
IMG - Load sensor being used in apps.yaml for the data
IMG - Car charging sensor for 7 days

[predbat.log](https://github.com/user-attachments/files/19999270/predbat.log)
Updated log file with 7 days now active again. 

 apps.yaml here: https://pastebin.com/yhjQZE6L

From what I can see, the car sensor is working correctly. 

Thanks for your help.

lewiswatt, 2025-05-03:
Thought the problem was resolving itself, as I could see that the plan was looking normal for the next day, however when I've woken up this morning, it's completely failed again.

[predbat.log](https://github.com/user-attachments/files/20022916/predbat.log)
[apps.yaml.txt](https://github.com/user-attachments/files/20022917/apps.yaml.txt)

[image]

springfall2008, 2025-05-03:
If you look at the log the load predictions seem okay, but all your load has been excluded as Car/iBoost.

Can you check if the givenergy sensor you have cnfigured is correct, maybe show it on the above chart?

  car_charging_energy:
    - 're:(sensor.givenergy_ev_power_watts)'

2025-05-03 10:25:03.288073: Model filter enabled - Discarding day 4 as it is the lowest of the 4 datapoints
2025-05-03 10:25:03.288206: Warn: Historical day 1 has 65 minutes of gap in the data, filled from 5.23 kWh to make new average 5.48 kWh (percent 95%)
2025-05-03 10:25:03.288547: Warn: Historical day 2 has 55 minutes of gap in the data, filled from 9.3 kWh to make new average 9.67 kWh (percent 96%)
2025-05-03 10:25:03.291165: Warn: Historical day 3 has 95 minutes of gap in the data, filled from 11.73 kWh to make new average 12.55 kWh (percent 93%)
2025-05-03 10:25:03.295540: Historical days now [1, 2, 3] weight [1, 1, 1]
2025-05-03 10:25:03.320761: Today's load divergence 200.0 % in-day adjustment 190.0 % damping 0.9x
2025-05-03 10:25:03.320822: Today's predicted so far 0 kWh with 4.92 kWh car/iBoost excluded and 0.8 kWh import ignored and 0 forecast extra.
2025-05-03 10:25:03.320836: Today's actual load so far 2.59 kWh with 5.55 kWh Car/iBoost excluded and 2.62 kWh import ignored.

lewiswatt, 2025-05-03:
[image]
Additional Context

lewiswatt, 2025-05-04:
Is there a clean way I can purge the load data history from predbat? I'm not seeing the behaviour improve, and it's randomly providing the correct loads, whilst in other plan calculations those readings are completely omitted. It's getting to the point now where it's planning an export at 17:00 which will drain my battery a good 7hrs early.

b1nuzz, 2025-05-04:
Yeah my experience is exactly the same. I’ve even changed sensor for the car load but that doesn’t make any difference. 
I’m almost certain it is car load related, BUT, I can’t fix it. 
It doesn’t matter whether I look at 1 day or 7 day load average, or anything in between, it always ends ups having a 0 load predication eventually.

gcoan, 2025-05-04:
> Is there a clean way I can purge the load data history from predbat? I'm not seeing the behaviour improve, and it's randomly providing the correct loads, whilst in other plan calculations those [...]

Don't really understand what you mean by 'purge the load data history from Predbat'.

It's up to you as to what days_previous of load_history you choose, so if you want to exclude some days, then set days_previous accordingly.

But I think I have found what could be your problem, in your apps.yaml:

```
  # car_charging_energy defines an incrementing sensor which measures the charge added to your car
  # is used for car_charging_hold feature to filter out car charging from the previous load data
  # Automatically set to detect Wallbox and Zappi, if it doesn't match manually enter your sensor name
  # Also adjust car_charging_energy_scale if it's not in kwH to fix the units
  # car_charging_energy: 're:(sensor.car_integral_charge_estimate)'

...

  car_charging_energy:
    - 're:(sensor.givenergy_ev_power_watts)'
```

Per the car charging documentation and the notes in apps.yaml, car_charging_energy must be a sensor in kWh.  The EV sensor you are using appears to be in watts so your EV load would be causing your entire house load to be excluded.

Have you set car_charging_energy_scale appropriately?

lewiswatt, 2025-05-04:
This sensor was not scaled correctly, and now has been rectified. I did switch that sensor around a month ago so I'm hoping this resolves it. 

However, after waiting for several plan refreshes I'm still seeing the same zero load behaviour. I'm assuming I'll need to either wait the 4 days for it to be averaged out again?

gcoan, 2025-05-04:
> Is this a recent change, because I haven't been experiencing this for that last 12 months.

no the car energy sensor and being able to scale it has been in predbat for ages

> However, after waiting for several plan refreshes I'm still seeing the same zero load behaviour. I'm assuming I'll need to either wait the 4 days for it to be averaged out again?

no you shouldn't have to, predbat recalculates the load every other time it runs (i.e. every 10 minutes).

can you try commenting out car_charging_energy from apps.yaml and see what that does to the plan to confirm whether the car_charging is what's causing the zero load.

would be useful then to see the plan again and logfile

lewiswatt, 2025-05-04:
I swapped back to the old sensor that was intermittent and the plan was normal. I then realised that the sensor I used before was a total energy and not an instantaneous energy. Seems to have resolved it by using the new total energy sensor.

Apologies for hijacking your issue @b1nuzz. Hopefully some of it helps.

b1nuzz, 2025-05-04:
Although it’s never been an issue until recently, my EV load sensor was an instantaneous sensor in watts. 
I’ve made a helper so that I can reset this daily, and adjusted the load scale to 0.001 so that it converts from watts to kW. 

Hopefully that fixes it.

gcoan, 2025-05-04:
> I swapped back to the old sensor that was intermittent and the plan was normal. I then realised that the sensor I used before was a total energy and not an instantaneous energy. Seems to have resolved it by using the new total energy sensor.

> Although it’s never been an issue until recently, my EV load sensor was an instantaneous sensor in watts.

Per the documentation:

car_charging_energy - Set in apps.yaml to point to a Home Assistant entity which is the daily incrementing kWh data for the car charger. ...
Note that this must be configured to point to an 'energy today' sensor in kWh not an instantaneous power sensor (in kW) from the car charger.

You need to configure predbat to point to a Today energy sensor, not a Total sensor or an instantaneous power sensor.  And must be in kWh not W !

chalfontchubby, 2026-09-17:
Could you confirm whether this is still an issue? A lot has changed and in order to debug abything now we would need a new debug.yaml please.

If it's fixed (or we don't hear back to the contrary) I'll close this ticket in around a week so we can focus on live issues. Thanks.

lewiswatt, 2026-09-17:
Swapped to ML Prediction many months ago. Before swapping to ML, issues was resolved from my side.

closed 2026-09-17 by chalfontchubby