springfall2008/batpred #2315: New Predbat 8.19.1 install - Battery SOC not forecast in monitor only mode

jambees, 2025-05-01:
**Describe the bug**
After setting up Predbat and admittedly getting a few sensors incorrect in the apps.yaml config file several times (some sensors for the Fox setup were in kw rather than the needed w etc) I eventually got what I think is a correct apps.yaml. However the Battery SOC chart shows historical SOC on the left but has no forecast at all - simply a horizontal future line. The plan page also shows no change to the values in the SOC column for each timeslot.

I tried changing the mode from "monitor" to "control charge" in the config web page and then the next time it ran the plan it caused a proper battery SOC chart to be produced with a forecast. The plan page also showed battery SOC values in each timeslot. 

I tried changing the mode back from "control charge" to "monitor" again in the config web page and this still showed a proper forecast SOC chart and plan SOC values. As I'd been changing it in the config page without a restart I thought I'd test restarting the Predbat addon with monitor mode in the apps.yaml still. The plan showed changing battery SOC values and the SOC chart still had a forecast.

I wondered if maybe it was because I was changing mode in the config page that was making it work so I restarted the Predbat addon with the same mode of monitor. Still had an SOC forecast in the plan and a proper SOC chart.

**Expected behaviour**
When Predbat is first started after an install with it in monitor mode, it should still forecast the battery SOC based on it's knowledge from load sensors and PV sensors (and anything else).

**Predbat version**
v8.19.1

**Environment details**
- Inverter and battery setup - Fox H series 5kw G2 inverter with Fox EQ4800 energy cube batteries and RS485 modbus setup+integration
- Standard HAOS

**Screenshots**
I have screenshots from yesterday and today that show the sort of problem (but the yesterday ones were while a few sensors were still in kw rather than w).
IMG_0059 - Plan from yesterday with no SOC changing in the plan
IMG_0063 - SOC chart from yesterday with no forecast
IMG_0066 - SOC chart this evening after changing monitor mode in the config web page to control charge
IMG_0067 - SOC chart after changing mode back to monitor and letting plan run again.
IMG_0068 - Plan after changing back to monitor mode and plan running again.
Apologies they're photos of a monitor - it's all I had as evidence while trying to problem solve.

**Log file**

[predbat.log](https://github.com/user-attachments/files/20003083/predbat.log)

Log attached.

**Predbat debug yaml file**
Debug file available - produced by clicking debug button about 45mins after discovering the change of mode seemed to fix the SOC forecast.

phiebs, 2025-05-01:
I posted in Q&A as I wasn't sure if it was a bug but I feel this may be related - I noticed your import prices are all highlighted in green. I've been having the same issue recently, presumed it was just a display issue but occasionally have problems with the plan (importing at high IOG rate when not needed). I've been running predbat for a couple of years and haven't changed much other than updating versions. I wonder if somehow predbat is almost completely ignoring the import rates and that's why your plan isn't working?

gcoan, 2025-05-01:
Its hard to tell from the description of what was happening what the cause of the problem is.

It could well have been because some of your sensors were set wrongly that the SoC wasn't projected, the screenshots you've got from yesterday are certainly weird, you have a projected load and cost being incurred, but the status indicated that the battery was not moving at all.

Certainly Predbat when it starts up in monitor mode should be projecting the SoC changes,  but it sounds like it is doing this now.

Are you able to recreate the issue, shutting down predbat, starting it in monitor mode and seeing the behaviour you had before?  If so, then the logfile and your apps.yaml would enable the issue to be tracked down

jambees, 2025-05-01:
> Certainly Predbat when it starts up in monitor mode should be projecting the SoC changes, but it sounds like it is doing this now.
>
> Are you able to recreate the issue, shutting down predbat, [...]

Unfortunately I can't recreate the issue now. It is still in monitor mode as I never changed the apps.yaml file and the config page confirms it is in monitor mode. I have restarted the addon again and it still shows proper battery SOC charts and confirmed still correct after another plan run.

Despite me trying to correct some sensor allocations, I don't think that was the cause. Given the last thing I changed between it not working and suddenly working was changing the working mode to something other than monitor it seems that is the clue for what might have happened. It is as if that triggered something to happen that corrected the issue with SOC forecasts not being carried out.

jambees, 2025-05-02:
> I posted in Q&A as I wasn't sure if it was a bug but I feel this may be related - I noticed your import prices are all highlighted in green. I've been having the same issue recently, presumed it was [...]

I hadn't noticed that initially as I didn't know what the colours in the different cells was supposed to signify yet (still learning and readingthe docs/videos). But it does seem a bit odd that all of the peak and off-peak rates are marked green but that could be a feature of monitor. In fact I just did a test changing modes and in monitor mode the import price for all timesslots is green but in control charge mode they get coloured as I'd expect with off-peak rates green and peak rates red. This kind of makes sense as in monitor mode there's nothing it can do so all slots have to be acceptable.

Monitor mode:
[image]

Control charge mode:
[image]

phiebs, 2025-05-02:
It could be a complete red (or green!) herring and I don't want to cloud your initial issue - I've just seen my plan change from red/green to all green recently without me changing the mode and thought it *may* be relevant. I've rolled back to v8.18.5 and don't get this behavior anymore.

closed 2026-08-28 by chalfontchubby