springfall2008/batpred #2316: 8.19.1 throwing errors since 1.45am

pacemaker82, 2025-05-02:
**Describe the bug**

Something related to 
_"Inverter 0 unable to read charge window time as neither REST, charge_start_time or charge_start_hour are set"_

This started at 1.45am, and has continued since. I obviously took no action at this time as was asleep.

**Log file**

2025-05-02 08:00:14.849805: Warn: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
2025-05-02 08:00:14.880280: Warn: record_status Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
2025-05-02 08:00:14.880832: Inverter 0 SOC: 0.0kW 0% Current charge rate 156000000W Current discharge rate 156000000W Current power 0.0W Current voltage 52.0V
2025-05-02 08:00:14.880991: Error: Inverter 0 unable to read charge window time as neither REST, charge_start_time or charge_start_hour are set
2025-05-02 08:00:14.911153: Warn: record_status Error: Inverter 0 unable to read charge window time as neither REST, charge_start_time or charge_start_hour are set
2025-05-02 08:00:14.911436: Error: Exception raised 
2025-05-02 08:00:14.915024: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 1392, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 616, in update_pred
    self.fetch_inverter_data()
  File "/config/execute.py", line 647, in fetch_inverter_data
    inverter.update_status(self.minutes_now)
  File "/config/inverter.py", line 906, in update_status
    raise ValueError
ValueError

2025-05-02 08:00:14.949451: Info: record_status Error: Exception raised 
2025-05-02 08:00:14.949726: Error: 

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

pacemaker82, 2025-05-02:
Looks to be related to GIVTCP at 1.45am, after Predbat set the charge schedule... any reason this would randomly happen?

I am restarting GivTCP to see if that helps (v3.1.6)

```
2025-05-02 01:44:51,920 - GivTCP - write       -  [INFO    ] - Setting Charge Slot 1 to: 01:30 - 02:00 was a success
2025-05-02 01:44:52,415 - GivTCP - write       -  [INFO    ] - Setting Charge Schedule to enable was a success
2025-05-02 01:50:14,410 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-02 01:50:14,436 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-02 01:50:14,495 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-02 01:50:14,520 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-02 01:55:14,990 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
```

pacemaker82, 2025-05-02:
Restarting GivTCP has not resolved the issue

```
2025-05-02 08:15:14,880 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-02 08:15:14,905 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-02 08:15:14,959 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-02 08:15:14,981 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
```

pacemaker82, 2025-05-02:
I've not been able to resolve the issue with GivTCP 3.1.6, even after hard reboot of HA device. Therefore rolled back to GivTCP 2.4.9 add on, restarted Predbat and I have a plan again. Will stay on 2.4.9 I think ...

SwiftRR, 2025-05-04:
I have been experiencing similar.
I first had issues with givTCP 3.1.6 but I have been keeping abreast of prepay updates to 8.19.4.
Predbat has been crashing and/or freezing.
I have now gone back to v8.19.1.

I may downgrade givTCP back to 2.4.9 but I have only just got it back after restoring from a backup.

Rob