springfall2008/batpred #2321: Since lates upgrade to 18.19.3 below error in monitor mode

BondBug, 2025-05-03:
Hi I am getting the below error when set to Monitor Mode since latest update
Home Assistant Raspberry Pi 5 

 Add-on: Predbat
 Home Battery Prediction and Control
-----------------------------------------------------------
 Add-on version: 1.2.7
 You are running the latest version of this add-on.
 System: Home Assistant OS 15.2  (aarch64 / raspberrypi5-64)
 Home Assistant Core: 2025.4.4
 Home Assistant Supervisor: 2025.04.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[21:35:52] INFO: Configuring timezone (Europe/London)...
Bootstrap Predbat
Startup
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Predbat files are installed correctly for version v8.19.3
**** Starting Standalone Predbat ****
2025-05-03 21:35:52.814732: Loading apps.yaml
2025-05-03 21:35:52.864119: Info: Add-on slug is 6adb4f0d_predbat
2025-05-03 21:35:52.873751: Info: Connected to Home Assistant at http://supervisor/core
2025-05-03 21:35:52.873873: Creating task: <coroutine object HAInterface.socketLoop at 0x7f8463cfe0>
2025-05-03 21:35:52.874340: Info: Web Socket task started
2025-05-03 21:35:52.874801: Creating task: <coroutine object WebInterface.start at 0x7f8466e980>
2025-05-03 21:35:52.874920: Info: Start socket for url http://supervisor/core/api/websocket
Web interface started
2025-05-03 21:35:52.885139: Info: Web Socket active
Watching ['./futurerate.py', './prediction.py', './config.py', './ha.py', './unit_test.py', './predheat.py', './octopus.py', './db_manager.py', './hass.py', './userinterface.py', './utils.py', './solcast.py', './download.py', './alertfeed.py', './apps.yaml', './inverter.py', './plan.py', './execute.py', './gecloud.py', './compare.py', './fetch.py', './web.py', './output.py', './predbat.py', './energydataservice.py'] for changes
2025-05-03 21:35:58.531297: Error: Exception raised cannot access local variable 'best_soc_min' where it is not associated with a value
2025-05-03 21:35:58.534713: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 1331, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 628, in update_pred
    recompute = self.calculate_plan(recompute=recompute)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/plan.py", line 850, in calculate_plan
    self.optimise_all_windows(metric, metric_keep, debug_mode)
  File "/config/plan.py", line 2777, in optimise_all_windows
    best_price, best_price_export, best_metric, best_cost, best_keep, best_soc_min, best_cycle, best_carbon, best_import, best_battery_value = self.optimise_levels_pass(best_metric, metric_keep, debug_mode)
                                                                                                                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/plan.py", line 2769, in optimise_levels_pass
    return best_price, best_price_export, best_metric, best_cost, best_keep, best_soc_min, best_cycle, best_carbon, best_import, best_battery_value
                                                                             ^^^^^^^^^^^^
UnboundLocalError: cannot access local variable 'best_soc_min' where it is not associated with a value
2025-05-03 21:35:58.549564: Info: record_status Error: Exception raised cannot access local variable 'best_soc_min' where it is not associated with a value
2025-05-03 21:35:58.549785: Error: cannot access local variable 'best_soc_min' where it is not associated with a value
Traceback (most recent call last):
  File "/config/hass.py", line 191, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 1337, in update_time_loop
    raise e
  File "/config/predbat.py", line 1331, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 628, in update_pred
    recompute = self.calculate_plan(recompute=recompute)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/plan.py", line 850, in calculate_plan
    self.optimise_all_windows(metric, metric_keep, debug_mode)
  File "/config/plan.py", line 2777, in optimise_all_windows
    best_price, best_price_export, best_metric, best_cost, best_keep, best_soc_min, best_cycle, best_carbon, best_import, best_battery_value = self.optimise_levels_pass(best_metric, metric_keep, debug_mode)
                                                                                                                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/plan.py", line 2769, in optimise_levels_pass
    return best_price, best_price_export, best_metric, best_cost, best_keep, best_soc_min, best_cycle, best_carbon, best_import, best_battery_value
                                                                             ^^^^^^^^^^^^
UnboundLocalError: cannot access local variable 'best_soc_min' where it is not associated with a value

**Environment details**

- Inverter and battery setup
- Standard HAOS installer or Docker
- Anything else?

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

johnleeshore, 2025-05-04:
Yes I’m getting the same error with 18.9.2 and 18.9.4.

matttheotter, 2025-05-04:
Just updated to 8.19.4 same error.

Set to Control SoC Only.

Downgrading to 8.19.1 fixed it.

johnleeshore, 2025-05-04:
8.19.5 seems to sorted the issue for me

springfall2008, 2025-05-11:
This has been fixed in the latest release

closed 2025-05-11 by springfall2008