springfall2008/batpred #2322: Delayed Charging At Start Of Off-Peak Period

nbullus, 2025-05-03:
**Describe the bug**

[image]

**Expected behaviour**

Should start charging from 00:30

**Predbat version**

8.19.3

**Environment details**

- GE AIO
- Standard HAOS installer

**Screenshots**

See above

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**

[predbat_debug_22_10_00.yaml.txt](https://github.com/user-attachments/files/20025089/predbat_debug_22_10_00.yaml.txt)

nbullus, 2025-05-05:
Installed v8.19.7 and charge/export in off-peak period is lot better now, but we seem to have lost the Freeze Export and battery now charging to 100% using excess solar.

[image]

springfall2008, 2025-05-11:
Fixed on latest from what I can tell

[image]

nbullus, 2025-05-11:
Yes looks fine for me as well. You might as well close this one down.

closed 2025-05-11 by gcoan