springfall2008/batpred #2324: PV Forcast Warning in log file

alz41, 2025-05-03:
**Describe the bug**
2025-05-04 00:15:48.384358: Warn: PV Forecast data adds up to 77.02000000000001 kWh but total sensors add up to 59.97 kWh, this is unexpected and hence data maybe misleading (factor 1.3)

Wondering what total sensors is referring to ?

**Predbat version**

8.19.4

springfall2008, 2025-05-11:
Is this with Solcast? It's talking about the forecast value in the sensor vs the half hourly data within that sensor

alz41, 2025-05-11:
> Is this with Solcast? It's talking about the forecast value in the sensor vs the half hourly data within that sensor

Yes with Solcast HA Integration

springfall2008, 2025-05-19:
If your solar forecasts are okay you can ignore this, but I'd be interested to know why they don't add up