springfall2008/batpred #2326: apps.yaml has 15 errors

zahidsafdar, 2025-05-04:
Getting the following exception since moving to 19.4
	apps.yaml has 15 errors

 On version 19.4

Are these genuine exceptions or issues? 

[image]

[image]

gcoan, 2025-05-04:
@zahidsafdar the givtcp2 lines can all be deleted/commented out of your apps.yaml if you only have a single inverter

have a look in the predbat logfile if there is any more detail on the other issues, they look to be false though

raldred, 2025-05-04:
I also find that often after a HA reboot, batpred loads before the add-ons or components providing sensors for my inverter, a reload of batpred fixes things for me.

zahidsafdar, 2025-05-04:
Thanks Guys, Seemed to have cleared all except 2 validations 

[image]
[image]

gcoan, 2025-05-05:
> Thanks Guys, Seemed to have cleared all except 2 validations

I wonder if both of those validations are being flagged because there is no Octopus EV charging planned?

Might be worth seeing if they clear when EV charging is scheduled by Octopus.  If this is the case then they shouldn't be flagged because no planned charging is of course valid...

cyberkryten, 2025-05-14:
After updating to latest and doing a restart of HA today I had a TON of errors - not sure why as all values were as normal.

As @raldred noted, it seems that these never clear from the Predbat Dash (I'm using the AddOn if that makes a difference) - the "X errors" text remains even after the next pass when everything is normal and no errors are shown in logs.

From what I can see, the only way to clear the message seems to be to do a Predbat restart a few minutes after the HA restart.

gcoan, 2025-05-14:
> After updating to latest and doing a restart of HA today I had a TON of errors - not sure why as all values were as normal.
>
> As [@raldred](https://github.com/raldred) noted, it seems that these [...]

Have you tried refreshing the browser window?

I've seen quite a few examples where the HA front end caching behaves oddly, showing stale data, and refreshing the browser window/companion app display forces a redownload and its all fine

I am running the predbat addon as well and I'm not seeing the same problem with persistent validation errors in the web console

cyberkryten, 2025-05-14:
Yes, I've refreshed - in Safari on MacOS.  Not tested on anything else but will look a bit further when I notice it again.

raldred, 2025-05-14:
I find this happens every time I restart HA.   
I go to the "Predbat" addon screen and manually restart, then the errors clear after it comes back up.
Maybe there's a way to delay the validation check batpred is doing, until HA is completely finished loading?

[image]

gcoan, 2025-05-14:
It definitely doesn't happen all the time for me, because day before yesterday I upgraded HA to 2025.5.1 and whilst predbat complained about web socket errors, it sorted itself out on the next  run.

But I do agree, some form of delay in the predbat bootstrap to not consider an error fatal would be advantageous.  Almost every time I reboot HA and the addons completely I see predbat erroring whilst givtcp is still starting up, usually resulting in predbat restarting givtcp due to REST failure, and this can happen a couple of times before givtcp gets going to predbat's satisfaction.

Givtcp 3 is definitely slower to start than v2 and predbat isn't patient!

@springfall2008 suggest this be added as a FR

chalfontchubby, 2026-10-19:
_Posted by Claude on behalf of @chalfontchubby._

Closing as done by #4394, the same change as #2248: Predbat retries validation after start-up and treats briefly unavailable/unknown entities as transient, so it no longer needs Home Assistant and GivTCP to be fully up first.

closed 2026-10-19 by chalfontchubby