springfall2008/batpred #2328: Unneccessay updates to inverter

rszemeti, 2025-05-04:
**Describe the bug**
Looking through the logs I see:

<html><body>

6140 | 2025-05-04 11:05:15.770579:  Inverter 0 Updated start and end charge window to 23:59:00 - 12:30:00 (old 23:59:00 - 12:30:00)
-- | --
6139 | 2025-05-04 11:05:15.770421:  Successfully pressed button button.solis8_update_discharge_times on Inverter 0
6138 | 2025-05-04 11:05:11.671597:  Successfully pressed button button.solis8_update_charge_times on Inverter 0
6137 | 2025-05-04 11:05:07.483607:  Inverter 0 write_and_poll_value: No write needed for charge_end_minute: 30 == 30 fuzzy 0
6136 | 2025-05-04 11:05:07.483543:  Inverter 0 write_and_poll_value: No write needed for charge_end_hour: 12 == 12 fuzzy 0
6135 | 2025-05-04 11:05:07.483472:  Inverter 0 write_and_poll_value: No write needed for charge_end_time: 12:30:00 == 12:30:00 fuzzy 0
6134 | 2025-05-04 11:05:07.483413:  Inverter 0 write_and_poll_value: No write needed for charge_start_minute: 59 == 59 fuzzy 0
6133 | 2025-05-04 11:05:07.483344:  Inverter 0 write_and_poll_value: No write needed for charge_start_hour: 23 == 23 fuzzy 0
6132 | 2025-05-04 11:05:07.483268:  Inverter 0 write_and_poll_value: No write needed for charge_start_time: 23:59:00 == 23:59:00 fuzzy 0
6131 | 2025-05-04 11:05:07.483128:  Adjust demand (idle) time computed is 12:30:00-23:59:00
6130 | 2025-05-04 11:05:07.482717:  Adjust idle time, charge 23:59:00-12:30:00 discharge 00:00:00-00:00:00

</body>
</html>

So ... it has written and polled some values ... and all of them got "No Write Needed" ... so at the end of the process, no button press was needed, yet, it still triggered a button press.

**Expected behaviour**
If no values are written, dont trigger a button press.

**Predbat version**

current, atest

**Environment details**

- Solis, but whatever
- 
**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

chalfontchubby, 2026-08-20:
Thanks for the report. This issue has been open for over a year with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-08-28).

rszemeti, 2026-08-20:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

I assume it is still an issue as Predbat still does not use the Solis
inverter modes correctly.  My inverter got to ~75k writes in a year so
sadly I had to disable Predbat before it wore out the eeprom in the
inverter.  I strongly suspect you will eventually get inverter fails
through flash wear.

chalfontchubby, 2026-08-21:
Thanks @rszemeti - keeping this open, and given the flash-wear/hardware-damage angle this deserves more than a routine backlog pass. Solis inverter mode handling causing ~75k writes/year is worth prioritising a proper look at.

chalfontchubby, 2026-08-21:
When you file issues - please do include full logs, apps.yaml, and a debug yaml - it makes them much easier to autotriage.

chalfontchubby, 2026-08-21:
Dug into this properly with the log you posted. There's a real, narrow bug here: `press_and_poll_button()` presses both the charge and discharge update buttons whenever either one has something to send, on setups using separate `charge_update_button`/`discharge_update_button` entities rather than a single combined button. That's a genuine unnecessary write on the unrelated side, and we could gate it to only press the side that actually changed.

But we don't think that's the real issue behind what you were seeing. At most it halves the write count on cycles where a press was already going to happen - it doesn't explain a press firing on a cycle where nothing changed at all, which is what your log actually showed (every underlying `write_and_poll_value` check, including the charge start/end times themselves, logged "No write needed"). Something else is triggering the button press even when there's genuinely nothing to write, and halving an already-wrong number isn't the fix.

We can't take this further without a fresh full log, your `apps.yaml`, and a debug yaml captured against the current codebase - the original report is over a year old and a fair bit has changed since. Parking this for now rather than shipping a fix that wouldn't actually solve what you ran into. If you (or anyone else on a similar Solis separate-button setup) ever pick Predbat back up and can grab those, we'll take another look.

rszemeti, 2026-08-21:
The problem is caused by Predbat trying to manage SOC in the battery itself
instead of letting the inverter take care of it ...

The Solis inverters have a flexible set of modes including "grid first"
"self use" and timed charge and discharge to a specified battery level.
Predbat directly writes "magic mode numbers"  instead of using the
Solax_modbus mode names and tries to fake some of the inverter modes by
directly controlling charge and discharge itself from memory ... this can
lead to *hundreds* of inverter writes in a couple of hours.

As mentioned, I no longer run Predbat as I switched to my own solution, so
for me, the problem is not an issue, feel free to park it.

chalfontchubby, 2026-08-21:
Thanks for the extra detail - we've done a proper triage pass on this rather than just leaving it parked.

On the "modes vs micromanagement" framing: Predbat does write the high-level solax_modbus modes (Self-Use, Timed Charge/Discharge, Feed-in priority, etc.) by name, with change-detection against the entity's current state first - it's not bypassing the mode system to poke raw numbers at the inverter. Timed Charge/Discharge is one of those modes though, and it needs its own schedule (start/end time) written alongside it by design - that's not Predbat faking a mode, it's how that mode is meant to be driven. A Predbat plan does need to move that schedule around through the day to track variable rates, which a fixed mode alone genuinely can't express.

Two separate things came out of the triage:

1. **A real, understood ~2x-writes issue.** On a Solis setup using separate charge/discharge update buttons rather than one combined button, both get pressed whenever either side has something to send, rather than only the side that actually changed. Straightforward, low-risk fix - we'll get that in regardless of anything else here.

2. **A hypothesis for the actual mystery** - the button firing on a cycle where every value logged "no write needed," including the charge window itself. The schedule-enable check that gates the button press falls back to treating the schedule as "off" whenever its configured entity's *live* state doesn't read back as "on" that cycle. If that's flapping - the underlying inverter entity itself, or something in how it gets read - Predbat would re-enable and press the schedule every single cycle regardless of whether the window actually changed, which would produce exactly the pattern in your original log and plausibly explain the volume you saw. This lives in an older, separate part of the charge-window code, unrelated to the split-button handling.

We can't confirm that without real data, and don't want to guess and ship something that doesn't actually fix what you hit. If you (or anyone else on a similar setup) ever come back to this, the useful things to capture would be your `apps.yaml` (specifically whether `scheduled_charge_enable` is configured, and what entity it points at) alongside a log spanning a few consecutive cycles at the same verbosity as your original report. No pressure either way - you've said you've moved on, which is completely fair.

i-jk, 2026-09-17:
I made a temporary fix for the button press inverter timed charge issue that protects the inverter EPROM while keeping PB working as normal - https://github.com/springfall2008/batpred/issues/4712#issuecomment-5705817460 - incase it helps anyone until we get a full fix.

chalfontchubby, 2026-09-17:
*(Written by Claude, posted by @chalfontchubby.)*

Closing this out. The original report's log signature — every register logging "No write needed", the button pressed anyway — is the exact mechanism fixed in #5126: the `Inverter` objects were being rebuilt every 5-minute cycle, so no commit-once guard could ever latch. That PR persists the objects across cycles and adds the missing commit-once protection on the charge side to match the export side.

@rszemeti's later report in this thread (GS_fb00 + `charge_limit` set, entering what looked like the legacy micromanagement path) is a separate, unconfirmed symptom - no log against current head was ever supplied to pin it down, and it may or may not share this root cause. Rather than let that keep this issue open indefinitely, closing here: if it's still reproducible once #5126 lands, please open a fresh issue with a `predbat.log`/`apps.yaml`/`predbat_debug.yaml` set from current main, and we'll pick it up properly with real evidence rather than guessing at a two-year-old thread.

Thanks both for the logs and patience on this one.

chalfontchubby, 2026-09-17:
*(Written by Claude, posted by @chalfontchubby.)*

Split the two known-outstanding threads from this issue into their own tickets, so they survive this issue closing on #5126's merge rather than getting buried:

- #5127 - the unconfirmed GS_fb00 + `charge_limit` micromanagement report (still needs fresh logs against current main)
- #5128 - Feed-in priority modes (64/96/98) never reachable
- #5129 - `charge_limit` should default on for GS_fb00 rather than requiring manual apps.yaml edit

This issue will auto-close when #5126 merges.

closed 2026-09-26 by springfall2008