springfall2008/batpred #2334: Larger number of inverter writes than expected to maintain a SOC

dpe926, 2025-05-05:
When holding the battery charge level predbat seems to require a larger than expected number of inverter writes to achieve this. 

[533ea71a_givtcp_2025-05-05T08-57-38.457Z.log](https://github.com/user-attachments/files/20034050/533ea71a_givtcp_2025-05-05T08-57-38.457Z.log)

I am not sure if this is really a bug or a facet of the way that the system works. I would have thought that one command to pause and a wide band of hysteresis for the set SOC could reduce the number of writes. If this is not achievable would me increasing the time between calculations in predbat reduce the number of writes?

**Expected behaviour**
That the number of commands sent to the inverter is minimised as far as possible

**Predbat version
8.19.5 (although the previous versions have performed the same)

**Environment details**
- Standard HAOS installer

[predbat.log](https://github.com/user-attachments/files/20034054/predbat.log)

gcoan, 2025-05-05:
I think your problem is different from how you have described it.

It appears from the givtcp log that predbat is oscillating between pause charging and demand mode between 3-5am on the 5th May.

For the 4th, it seems very few state changes at all so it's probably something about the plan for the 5th.

Can you provide a copy of your predbat debug file and logfile so can see what predbat was doing.

Also the history of predbat.status over this time period would confirm that the status was oscillating

dpe926, 2025-05-05:
Thanks for the reply,
The predbat log is attached above, and you are correct that predbat enters this oscillating charging/hold charging status as soon as the set SOC is reached. this did not occur on the 4th because the battery did not discharge down to the set SOC level. I have attached some screen shots which may help. I will try to sort out the debug file

[image]

[image]

dpe926, 2025-05-05:
Its an Givenergy AC3.0 Gen 1. I have the pause mode start and end times commented out as I do not believe these are supported by a Gen 1. (when they were left in I got the spurious warnings):

[image]

I am happy to put these back or will try setting battery_rate_min to 200 in apps.yaml tonight

gcoan, 2025-05-05:
If you have a gen 1 AC inverter your setup is correct @dpe926, it only supports battery pause mode (in newer firmware), doesn't support pause start and end times

chalfontchubby, 2026-08-20:
Thanks for the report. This issue has been open for over a year with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-08-28).

springfall2008, 2026-09-09:
Automated first-pass triage (a maintainer will review before any action is taken).

**Classification:** question — not a bug report on current behaviour. The reporter themselves asked "I am not sure if this is really a bug or a facet of the way the system works", and the maintainer diagnosis in the thread settles it as expected behaviour given the hardware.

**Status:** The reporter [confirmed on 2026-08-24](https://github.com/springfall2008/batpred/issues/2334#issuecomment-5393017627) that this is no longer an issue for them (they switched Predbat off during the overnight hours concerned). This was flagged as an AUTOCLOSE_CANDIDATE on 2026-08-20 with closure promised around 2026-08-28 — the reporter's reply satisfies that condition, so this looks ready for a maintainer to close. No duplicate was found (open #1690 covers a related "hold not holding" symptom but is a distinct report).

**What I checked against current main:** The oscillation the GivTCP log shows (Charging ↔ Hold charging, 3-5am) matches the hold logic in `execute.py`. On the "Hold charging" path (`apps/predbat/execute.py:329-352`) the hold only sticks while SoC stays within ~1% of the target: at or above `target - 1%` Predbat disables the charge window and holds on reserve/pause, but once house load pulls SoC more than 1% below target the next plan cycle takes the other branch (`execute.py:357`, status back to "Charging") and rewrites the window and charge rate. A GivEnergy AC 3.0 **Gen 1** has no timed-pause registers (as @gcoan confirmed in-thread), so the hold falls back to reserve/discharge-rate writes rather than a single pause command, and with the battery discharging overnight it repeatedly falls out of the 1% band. This also explains why the `battery_rate_min: 200` suggestion made little difference — that setting feeds the charge-rate *model* (it is passed to `find_charge_rate` at `execute.py:260`), it does not reduce the number of register writes.

To answer the two questions raised for posterity: (a) a single pause command with wide hysteresis is only available on inverters with timed pause support (`inv_has_timed_pause`), which the Gen 1 lacks; (b) yes, raising `calculate_plan_every` in apps.yaml reduces how often the plan is recomputed and therefore how often the flip-flop issues writes, at the cost of plan responsiveness.

No test was run — there is no clean mapping to a test module for a hardware write-churn report, and the reporter has confirmed they are no longer affected.

No priority assigned (question / no longer reproducible for the reporter).

chalfontchubby, 2026-09-09:
Closing. The above thread covers what is think needed.

closed 2026-09-09 by chalfontchubby