springfall2008/batpred #2339: iBoost grid usage shown when house load and iboost usage greater than solar production

tmh88, 2025-05-05:
**Describe the bug**
When house load and iboost are greater than solar production load from grid is shown in plan

**Expected behaviour**
I would expect battery to discharge or iboost load to be lower for that period and not show grid usage

**Predbat version**

V8-19-7

**Environment details**

- Sunsynk Inverter and Batteries
- Standard HAOS

**Screenshots**

[image]

**Log file**
[predbat(1).log](https://github.com/user-attachments/files/20044925/predbat.1.log)

**Predbat debug yaml file**
[predbat_debug (2).txt](https://github.com/user-attachments/files/20045573/predbat_debug.2.txt)

mbuhansen, 2025-05-06:
I see the same behavior on my setup.
Another thing is how predbat want to control iboost power. in my situation solar power needs to be above battery max charge power, before iboost start use surplus solar. Another way could be a sensor from predbat with the predicted power to iboost, this can be used to control iboost device earlyer on the day.

[image]

[predbat (5).log](https://github.com/user-attachments/files/20054497/predbat.5.log)

[predbat_debug (2).txt](https://github.com/user-attachments/files/20054532/predbat_debug.2.txt)

gcoan, 2025-05-06:
> Another way could be a sensor from predbat with the predicted power to iboost, this can be used to control iboost device earlyer on the day.

there are several iboost sensors that predbat creates https://springfall2008.github.io/batpred/output-data/#iboost-solar-diverter-data that should give you what you want

tmh88, 2025-05-07:
@springfall2008 I've let it run the past 2days, it's forcasts grid usage but dosent set the inverter to charge or change the battery soc. So guess it's just in the forcast. Hope this helps

tmh88, 2025-05-30:
@springfall2008 any updates on this one?

springfall2008, 2025-06-21:
Fix released

https://github.com/springfall2008/batpred/releases/tag/v8.21.7

IBoost solar had a problem where boost was calculated after battery controls leading to predicted imports.

Added a new setting switch.predbat_iboost_solar_excess which when turned off (default) iBoost will operate on any generated solar, then turned on it will only operate if excess solar is being exported.