springfall2008/batpred #2340: Latest Update 18.19.7   in Monitor mode below error on startup

BondBug, 2025-05-05:
Homeassistant Raspeberry pi 5 

Bootstrap Predbat
Startup
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Predbat files are installed correctly for version v8.19.7
**** Starting Standalone Predbat ****
2025-05-05 21:13:13.729076: Loading apps.yaml
2025-05-05 21:13:13.778768: Info: Add-on slug is 6adb4f0d_predbat
2025-05-05 21:13:13.796249: Info: Connected to Home Assistant at http://supervisor/core
2025-05-05 21:13:13.796377: Creating task: <coroutine object HAInterface.socketLoop at 0x7f8cd49300>
2025-05-05 21:13:13.796839: Info: Web Socket task started
2025-05-05 21:13:13.797348: Creating task: <coroutine object WebInterface.start at 0x7f8cd86890>
2025-05-05 21:13:13.797465: Info: Start socket for url http://supervisor/core/api/websocket
Web interface started
2025-05-05 21:13:13.808498: Info: Web Socket active
Watching ['./futurerate.py', './prediction.py', './config.py', './ha.py', './unit_test.py', './predheat.py', './octopus.py', './db_manager.py', './hass.py', './userinterface.py', './utils.py', './solcast.py', './download.py', './alertfeed.py', './apps.yaml', './inverter.py', './plan.py', './execute.py', './gecloud.py', './compare.py', './fetch.py', './web.py', './output.py', './predbat.py', './energydataservice.py'] for changes
2025-05-05 21:13:18.932936: Error: Exception raised list index out of range
2025-05-05 21:13:18.936371: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 1334, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 628, in update_pred
    recompute = self.calculate_plan(recompute=recompute)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/plan.py", line 839, in calculate_plan
    self.optimise_all_windows(metric, metric_keep, debug_mode)
  File "/config/plan.py", line 2780, in optimise_all_windows
    best_price_charge, best_price_export, best_price_charge_level, best_price_export_level, best_metric, best_cost, best_keep, best_soc_min, best_cycle, best_carbon, best_import, best_battery_value = self.optimise_levels_pass(
                                                                                                                                                                                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/plan.py", line 2759, in optimise_levels_pass
    best_price_charge, best_price_export, best_price_charge_level, best_price_export_level = self.find_price_levels(price_set, price_links, window_index, self.charge_limit_best, self.charge_window_best, self.export_window_best, self.export_limits_best)
                                                                                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/plan.py", line 54, in find_price_levels
    highest_price_charge_level = price_set[-1]
                                 ~~~~~~~~~^^^^
IndexError: list index out of range

2025-05-05 21:13:18.951374: Info: record_status Error: Exception raised list index out of range
2025-05-05 21:13:18.951543: Error: list index out of range
Traceback (most recent call last):
  File "/config/hass.py", line 191, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 1340, in update_time_loop
    raise e
  File "/config/predbat.py", line 1334, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 628, in update_pred
    recompute = self.calculate_plan(recompute=recompute)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/plan.py", line 839, in calculate_plan
    self.optimise_all_windows(metric, metric_keep, debug_mode)
  File "/config/plan.py", line 2780, in optimise_all_windows
    best_price_charge, best_price_export, best_price_charge_level, best_price_export_level, best_metric, best_cost, best_keep, best_soc_min, best_cycle, best_carbon, best_import, best_battery_value = self.optimise_levels_pass(
                                                                                                                                                                                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/plan.py", line 2759, in optimise_levels_pass
    best_price_charge, best_price_export, best_price_charge_level, best_price_export_level = self.find_price_levels(price_set, price_links, window_index, self.charge_limit_best, self.charge_window_best, self.export_window_best, self.export_limits_best)
                                                                                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/plan.py", line 54, in find_price_levels
    highest_price_charge_level = price_set[-1]
                                 ~~~~~~~~~^^^^
IndexError: list index out of range

2025-05-05 21:15:02.172247: Error: Exception raised list index out of range
2025-05-05 21:15:02.173565: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 1395, in run_time_loop
[... 16 lines cut ...]

2025-05-05 21:15:02.184484: Info: record_status Error: Exception raised list index out of range
2025-05-05 21:15:02.184566: Error: list index out of range
Traceback (most recent call last):
  File "/config/hass.py", line 191, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 1400, in run_time_loop
    raise e
  File "/config/predbat.py", line 1395, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 628, in update_pred
    recompute = self.calculate_plan(recompute=recompute)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/plan.py", line 839, in calculate_plan
    self.optimise_all_windows(metric, metric_keep, debug_mode)
  File "/config/plan.py", line 2780, in optimise_all_windows
    best_price_charge, best_price_export, best_price_charge_level, best_price_export_level, best_metric, best_cost, best_keep, best_soc_min, best_cycle, best_carbon, best_import, best_battery_value = self.optimise_levels_pass(
                                                                                                                                                                                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/plan.py", line 2759, in optimise_levels_pass
    best_price_charge, best_price_export, best_price_charge_level, best_price_export_level = self.find_price_levels(price_set, price_links, window_index, self.charge_limit_best, self.charge_window_best, self.export_window_best, self.export_limits_best)
                                                                                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/plan.py", line 54, in find_price_levels
    highest_price_charge_level = price_set[-1]
                                 ~~~~~~~~~^^^^
IndexError: list index out of range

Amigos2222, 2025-05-06:
Same issue for me cannot get predbat to generate a plan

Amigos2222, 2025-05-06:
I've tried downgrading as this was working on earlier versions but predbat seems to immediately switch back to the most recent version. Auto update i've set to false but this still occurs.

pgit2345, 2025-05-06:
Same issue here. Happened after updating to latest octopus integration (from v14 something to v15.0.1), although might just be a coincidence e.g. post restart. 

Running:
Core 2025.4.4
Supervisor 2025.04.1
Operating System 15.2
Frontend 20250411.0

pgit2345, 2025-05-07:
see also [2343](https://github.com/springfall2008/batpred/issues/2343)

springfall2008, 2025-05-08:
Fixed in the latest .8 release

pgit2345, 2025-05-08:
Upgraded. Errors have now gone. 

The compare tariffs feature was also broken. Perhaps the same reason. Hopefully tomorrow it'll run okay.

gcoan, 2025-05-11:
> Upgraded. Errors have now gone. 
> 
> The compare tariffs feature was also broken. Perhaps the same reason. Hopefully tomorrow it'll run okay. 

the compare feature runs the predbat engine repeatedly for different import/export tariffs so if the engine was broken then this would explain why the compare was broken