springfall2008/batpred #2341: PredBat plan allowing unnecessary clipping

XoneDB4, 2025-05-06:
On very sunny days, my oversized array can produce more than my inverter limit for a good chunk of the morning. PredBat accounts for this by freeze exporting early in the morning and often later, once the battery is 80-90% charged. However, this rapid charge period in the middle sometimes still allows clipping. There are several hours later in the day when the PV generation is below the inverter limit but well above the predicted load, so it is not necessary to charge the battery fully in the morning peak. I would expect PredBat to plan a much longer export window in the morning, where the battery only charges from excess over the inverter limit. 

[predbat_debug_yaml.txt](https://github.com/user-attachments/files/20073229/predbat_debug_yaml.txt)

XoneDB4, 2025-05-07:
This morning, I also found that Predbat was using freeze export rather than export, causing the PV production to be limited by the inverter limit for no known reason.

springfall2008, 2025-05-09:
If you set 'metric_min_improvement_export' to a small value e.g 0.1 then you get the plan you might expect:

[image]

XoneDB4, 2025-05-09:
Thanks. It was set to zero and is now set to 0.1, but I can still improve the predicted cost by manually setting export slots for the whole morning:

- predbat.best_metric when manually forcing export until 1400: -336.3p
- predbat.best_metric with standard plan: -322.6p

springfall2008, 2025-05-10:
Please post the new debug yaml plan

springfall2008, 2025-05-10:
Also note you have best_soc_keep as non-zero which will increase costs

XoneDB4, 2025-05-13:
[predbat_debug_yaml_3.txt](https://github.com/user-attachments/files/20183739/predbat_debug_yaml_3.txt)

Waited for another very sunny morning. I'm 19p better off for the day if I force export all morning

Plan without intervention:
[image]

Plan with forced export
[image]

I will set best_soc_keep to zero but would that have any impact on this scenario?

XoneDB4, 2025-05-14:
Today's plan suggested freeze export slots mid-morning again. That _surely_ doesn't make sense in the peak production part of the day as it prevents the battery charging from excess solar, ensuring clipping happens.

The PredBat plan today was 22p worse by end of day than the manual forced export plan.

cyberkryten, 2025-05-22:
I think that the Clip kWh values themselves are at present informational only, derived from other data.

From going through the code, they are calculated in run_predicition (I'm still working on why they're out in some cases, like overnight), but they aren't actually used in plan.py to make any decisions, so the fact that column shows clipping doesn't trigger any response in the code.  

That makes some sense when you consider that in theory the data is already there in things like the Inverter Limit and Export Limit.  However, clearly there's a difference in either the way they are calculated or the usage/thresholds used to trigger Predbat behaviour.  I'm still looking for those!

closed 2026-08-28 by chalfontchubby