springfall2008/batpred #2347: Numerous Errors - Please help, need pushing in the right direction.

robomilo, 2025-05-08:
Hi,

This is my log, been trying to work this the last couple of days, but i've come against a road block. For information ive reconfigured my solar ct clamps etc recently, so i only have a day or two of solar data. Could I possibly send my apps.yaml to someone to check over. 

Much appreciated 

72476 | 2025-05-08 13:19:31.383746: Error: list index out of range
-- | --
72475 | 2025-05-08 13:19:31.383670: Info: record_status Error: Exception raised list index out of range
72473 | IndexError: list index out of range
72456 | 2025-05-08 13:19:31.370962: Error: Traceback (most recent call last):
72455 | 2025-05-08 13:19:31.368061: Error: Exception raised list index out of range
72412 | 2025-05-08 13:19:31.085276: Warn: Out of range index 1 within item inverter_time value None
72378 | 2025-05-08 13:19:30.906029: Warn: No solar data has been configured.

robomilo, 2025-05-08:
# ------------------------------------------------------------------
# This is an example configuration, please modify it
# ------------------------------------------------------------------
---
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
   - '£'
   - 'p'

  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto

  # If you are using Predbat outside of HA then set the HA URL and Key (long lived access token here)
  #ha_url: 'http://homeassistant.local:8123'
  #ha_key: 'xxx'

  # Sets the maximum period of zero load before the gap is filled, default 30 minutes
  # To disable set it to 1440
  load_filter_threshold: 30

  #
  # Sensors, more than one can be specified and they will be summed up automatically
  #
  # For two inverters the load today would normally be the master load sensor only (to cover the entire house)
  # If you have three phase and one inverter per phase then you would need three load sensors
  #
  # For pv_today if you have multiple solar inverter inputs then you should include one entry for each inverter
  #
  load_today:
    - sensor.calculated_house_load_today
  import_today:
    - sensor.myenergi_home_grid_import_today
  export_today:
    - sensor.myenergi_home_grid_export_today
  pv_today:
    - sensor.shed_solar_daily_generation_kwh
    - sensor.new_fox_daily_solar_generation_kwh
    - sensor.old_fox_daily_solar_generation_kwh

  # Load forecast can be used to add to the historical load data (heat-pump)
  # To link to Predheat
  # Data must be in the format of 'last_updated' timestamp and 'energy' for incrementing kWh
  #load_forecast:
  #  - predheat.heat_energy$external
  #
  # Controls/status - must by 1 per inverter
  #
  num_inverters: 2
  inverter_type:
    - LuxPower
    - LuxPower
  inverter:
    name: "LuxPower"
    has_rest_api: False
    has_mqtt_api: False
    has_service_api: True
    output_charge_control: "current"
    current_dp: 0
    has_charge_enable_time: True
[... 364 lines cut ...]
  # metric_octopus_gas: 're:(sensor.(octopus_energy_|)gas_[0-9a-z]+_[0-9a-z]+_current_rate)'

  # Nordpool market energy rates
  #futurerate_url: 'https://www.nordpoolgroup.com/api/marketdata/page/325?currency=GBP'
  #futurerate_adjust_import: True
  #futurerate_adjust_export: False
  #futurerate_peak_start: "16:00:00"
  #futurerate_peak_end: "19:00:00"
  #futurerate_peak_premium_import: 14
  #futurerate_peak_premium_export: 6.5

  # Watch list, a list of sensors to watch for changes and then update the plan if they change
  # This is useful for things like the Octopus Intelligent Slot sensor so that the plan update as soon as you plugin in
  # Only uncomment the items you actually have set up above in apps.yaml, of course you can add your own as well
  # Note those using +[] are lists that are appended to this list, whereas {} items are single items only
  #watch_list:
  #  - '{octopus_intelligent_slot}'
  #  - '{octopus_ready_time}'
  #  - '{octopus_charge_limit}'
  #  - '{octopus_saving_session}'
  #  - '+[car_charging_planned]'
  #  - '+[car_charging_soc]'
  #  - '{car_charging_now}'
  octopus_api_account: 'Removed'
  octopus_api_key: 'Removed'

robomilo, 2025-05-08:
Hi Trefor,

Yes here it is:

[predbat.log](https://github.com/user-attachments/files/20103233/predbat.log)

robomilo, 2025-05-08:
or is this more what you wanted?

74376 | 2025-05-08 14:50:03.326655: Error: list index out of range
-- | --
74375 | 2025-05-08 14:50:03.326573: Info: record_status Error: Exception raised list index out of range
74373 | IndexError: list index out of range
74356 | 2025-05-08 14:50:03.315784: Error: Traceback (most recent call last):
74355 | 2025-05-08 14:50:03.314997: Error: Exception raised list index out of range
74323 | 2025-05-08 14:50:03.184849: Warn: Out of range index 1 within item inverter_time value None
74301 | 2025-05-08 14:50:03.134322: Warn: No solar data has been configured.
74276 | 2025-05-08 14:45:03.396070: Error: list index out of range
74275 | 2025-05-08 14:45:03.395996: Info: record_status Error: Exception raised list index out of range
74273 | IndexError: list index out of range
74256 | 2025-05-08 14:45:03.385642: Error: Traceback (most recent call last):
74255 | 2025-05-08 14:45:03.384757: Error: Exception raised list index out of range
74223 | 2025-05-08 14:45:03.243556: Warn: Out of range index 1 within item inverter_time value None
74201 | 2025-05-08 14:45:03.178685: Warn: No solar data has been configured.
74176 | 2025-05-08 14:40:03.476209: Error: list index out of range
74175 | 2025-05-08 14:40:03.476132: Info: record_status Error: Exception raised list index out of range
74173 | IndexError: list index out of range
74156 | 2025-05-08 14:40:03.465500: Error: Traceback (most recent call last):
74155 | 2025-05-08 14:40:03.464678: Error: Exception raised list index out of range
74123 | 2025-05-08 14:40:03.336810: Warn: Out of range index 1 within item inverter_time value None
74101 | 2025-05-08 14:40:03.286698: Warn: No solar data has been configured.
74076 | 2025-05-08 14:35:03.232407: Error: list index out of range
74075 | 2025-05-08 14:35:03.232332: Info: record_status Error: Exception raised list index out of range
74073 | IndexError: list index out of range
74056 | 2025-05-08 14:35:03.222338: Error: Traceback (most recent call last):
74055 | 2025-05-08 14:35:03.221499: Error: Exception raised list index out of range
74023 | 2025-05-08 14:35:03.080831: Warn: Out of range index 1 within item inverter_time value None
74001 | 2025-05-08 14:35:03.020881: Warn: No solar data has been configured.
73976 | 2025-05-08 14:30:03.316207: Error: list index out of range
73975 | 2025-05-08 14:30:03.316136: Info: record_status Error: Exception raised list index out of range
73973 | IndexError: list index out of range
73956 | 2025-05-08 14:30:03.305571: Error: Traceback (most recent call last):
73955 | 2025-05-08 14:30:03.304726: Error: Exception raised list index out of range
73923 | 2025-05-08 14:30:03.169297: Warn: Out of range index 1 within item inverter_time value None
73901 | 2025-05-08 14:30:03.115046: Warn: No solar data has been configured.
73876 | 2025-05-08 14:25:02.980624: Error: list index out of range
73875 | 2025-05-08 14:25:02.980542: Info: record_status Error: Exception raised list index out of range
73873 | IndexError: list index out of range
73856 | 2025-05-08 14:25:02.969464: Error: Traceback (most recent call last):
73855 | 2025-05-08 14:25:02.968543: Error: Exception raised list index out of range
73823 | 2025-05-08 14:25:02.837403: Warn: Out of range index 1 within item inverter_time value None
73801 | 2025-05-08 14:25:02.785676: Warn: No solar data has been configured.
73776 | 2025-05-08 14:20:07.091108: Error: list index out of range
73775 | 2025-05-08 14:20:07.091033: Info: record_status Error: Exception raised list index out of range
73773 | IndexError: list index out of range
73756 | 2025-05-08 14:20:07.080261: Error: Traceback (most recent call last):
73755 | 2025-05-08 14:20:07.079432: Error: Exception raised list index out of range
73723 | 2025-05-08 14:20:06.947977: Warn: Out of range index 1 within item inverter_time value None
73701 | 2025-05-08 14:20:06.896893: Warn: No solar data has been configured.
73676 | 2025-05-08 14:15:03.979421: Error: list index out of range
73675 | 2025-05-08 14:15:03.979342: Info: record_status Error: Exception raised list index out of range
73673 | IndexError: list index out of range
73656 | 2025-05-08 14:15:03.968869: Error: Traceback (most recent call last):
73655 | 2025-05-08 14:15:03.968072: Error: Exception raised list index out of range
73623 | 2025-05-08 14:15:03.838787: Warn: Out of range index 1 within item inverter_time value None
73601 | 2025-05-08 14:15:03.781184: Warn: No solar data has been configured.
73576 | 2025-05-08 14:10:03.204816: Error: list index out of range
73575 | 2025-05-08 14:10:03.204742: Info: record_status Error: Exception raised list index out of range
73573 | IndexError: list index out of range
73556 | 2025-05-08 14:10:03.193539: Error: Traceback (most recent call last):
73555 | 2025-05-08 14:10:03.192675: Error: Exception raised list index out of range
73523 | 2025-05-08 14:10:03.065163: Warn: Out of range index 1 within item inverter_time value None
73501 | 2025-05-08 14:10:03.017092: Warn: No solar data has been configured.
73476 | 2025-05-08 14:05:03.345964: Error: list index out of range
73475 | 2025-05-08 14:05:03.345887: Info: record_status Error: Exception raised list index out of range
73473 | IndexError: list index out of range
73456 | 2025-05-08 14:05:03.334889: Error: Traceback (most recent call last):
[... 932 lines cut ...]
63116 | 2025-05-08 06:30:04.808714: Error: Exception raised list index out of range
[... cut ...]
62861 | 2025-05-08 06:20:02.936111: Warn: Historical day 7 has 250 minutes of gap in the data, filled from 24.42 kWh to make new average 29.55 kWh (percent 83%)
62859 | 2025-05-08 06:20:02.934952: Warn: No solar data has been configured.
62834 | 2025-05-08 06:15:02.546770: Error: list index out of range
62833 | 2025-05-08 06:15:02.546700: Info: record_status Error: Exception raised list index out of range
62831 | IndexError: list index out of range
62814 | 2025-05-08 06:15:02.537197: Error: Traceback (most recent call last):
62813 | 2025-05-08 06:15:02.536374: Error: Exception raised list index out of range

springfall2008, 2025-05-11:
```
2025-05-08 13:50:02.875221: Error: Exception raised list index out of range
2025-05-08 13:50:02.875998: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 1395, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 628, in update_pred
    recompute = self.calculate_plan(recompute=recompute)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/plan.py", line 839, in calculate_plan
    self.optimise_all_windows(metric, metric_keep, debug_mode)
  File "/config/plan.py", line 2780, in optimise_all_windows
    best_price_charge, best_price_export, best_price_charge_level, best_price_export_level, best_metric, best_cost, best_keep, best_soc_min, best_cycle, best_carbon, best_import, best_battery_value = self.optimise_levels_pass(
                                                                                                                                                                                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/plan.py", line 2759, in optimise_levels_pass
    best_price_charge, best_price_export, best_price_charge_level, best_price_export_level = self.find_price_levels(price_set, price_links, window_index, self.charge_limit_best, self.charge_window_best, self.export_window_best, self.export_limits_best)
                                                                                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/plan.py", line 54, in find_price_levels
    highest_price_charge_level = price_set[-1]
                                 ~~~~~~~~~^^^^
IndexError: list index out of range
```
Your last error relates to an older version of Predbat, if you update to the latest it should get fixed

gcoan, 2025-05-11:
there are also some errors with your yaml indenting

>   inverter_clock_skew_start: 0
>   inverter_clock_skew_end: 0
>   inverter_clock_skew_discharge_start: 0
>   inverter_clock_skew_discharge_end: 0
>   inverter_can_charge_during_export: False
> 
>   # Clock skew adjusts the Predbat computer

all of these look to be indented too far so they are sub-children of battery discharge curve, not children of the predbat module

and your currency symbol near the beginning the £ and p should be indented a further 1 space

robomilo, 2025-05-11:
Yeh exactly it was the ident were all messed up. I'm pretty much sorted now thanks. This thread should probably be deleted.

closed 2025-05-11 by robomilo