springfall2008/batpred #2352: no errors shown in apps.yaml, but predbat won't start (Error: Template Configuration, please edit apps.yaml)

zarch1972, 2025-05-10:
I'm starting from a fresh Home Assistant 2025.5 install along with new Octopus and Luxpower_Dev installs.
Predbat is installed as an add-on.

I'm using the luxpower yaml file and have painstakingly got rid of all the highlighted errors in red under the apps.yaml section.

But despite this, predbat won't start.  I get the following errors in the log

```
-----------------------------------------------------------
 Add-on: Predbat
 Home Battery Prediction and Control
-----------------------------------------------------------
 Add-on version: 1.2.7
 You are running the latest version of this add-on.
 System: Home Assistant OS 15.2  (amd64 / generic-x86-64)
 Home Assistant Core: 2025.5.0
 Home Assistant Supervisor: 2025.04.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[15:29:28] INFO: Configuring timezone (Europe/London)...
s6-rc: info: service base-addon-log-level successfully started
Bootstrap Predbat
Startup
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Predbat files are installed correctly for version v8.19.9
**** Starting Standalone Predbat ****
2025-05-10 15:29:29.207094: Loading apps.yaml
2025-05-10 15:29:29.240282: Info: Add-on slug is 6adb4f0d_predbat
2025-05-10 15:29:29.246548: Info: Connected to Home Assistant at http://supervisor/core
2025-05-10 15:29:29.246632: Creating task: <coroutine object HAInterface.socketLoop at 0x7efd75a797b0>
2025-05-10 15:29:29.246905: Info: Web Socket task started
2025-05-10 15:29:29.247173: Creating task: <coroutine object WebInterface.start at 0x7efd7580aa70>
2025-05-10 15:29:29.247414: Info: Start socket for url http://supervisor/core/api/websocket
Web interface started
2025-05-10 15:29:29.264333: Info: Web Socket active
Watching ['./alertfeed.py', './compare.py', './predheat.py', './apps.yaml', './web.py', './solcast.py', './execute.py', './energydataservice.py', './config.py', './prediction.py', './futurerate.py', './userinterface.py', './gecloud.py', './db_manager.py', './utils.py', './ha.py', './inverter.py', './hass.py', './plan.py', './unit_test.py', './fetch.py', './download.py', './predbat.py', './output.py', './octopus.py'] for changes
2025-05-10 15:29:31.377294: Error: Template Configuration, please edit apps.yaml
2025-05-10 15:29:31.401566: Warn: record_status Error: Template Configuration, please edit apps.yaml
```

The logs aren't giving me any pointers to what could be wrong and there are no 'red' issues in the apps.yaml page.

I've looked in the debug.html file, but nothing shouts out to me in there either.

Any ideas where to look next please?

zarch1972, 2025-05-10:
Think I might have solved it.

I hadn't commented out this line about the template

  ```
# XXX: Template - You must delete the following line once you have configured your inverter
  # template: True
```
Seems to be doing things now that I have added the #.

```
2025-05-10 18:11:09.187368: Warn: Historical day 7 has 840 minutes of gap in the data, filled from 6.74 kWh to make new average 16.17 kWh (percent 42%)
2025-05-10 18:11:09.367815: Warn: No cost_today data for yesterday
2025-05-10 18:11:13.905675: Info: record_status Demand
```

Perhaps there could be some better error checking for the status of the template indicator?  And a more helpful error message in the log? 👍 

Thank you.

raldred, 2025-05-10:
This is the key line:

```
2025-05-10 15:29:31.377294: Error: Template Configuration, please edit apps.yaml
```

Glad you're up and running.

gcoan, 2025-05-11:
> Perhaps there could be some better error checking for the status of the template indicator? And a more helpful error message in the log? 👍

It is clearly documented as a step you need to do in the installation instructions https://springfall2008.github.io/batpred/install/#ready-to-light-the-touch-paper

Sorry it caught you out but RTFM !!

closed 2025-05-11 by gcoan

zarch1972, 2025-05-11:
Thanks for the replies, appreciate it.

Yes, glad i'm working now and yes, RTFM.  I did read lots of it!! LOL

Bu I still think the 'error' is too cryptic for a newbie.
`2025-05-10 15:29:31.377294: Error: Template Configuration, please edit apps.yaml`

You guys know what it means as you've been around this for a while, but I don't think it was precise enough.

How about "Error: Template is set to True, predbat is in preview mode.  Set Template: False to run configuration and generate your plan"

Or something similar.  just my two-penneth.
Thanks for all your hard work.

gcoan, 2025-05-13:
@zarch1972 thanks for your suggestion, I have changed the error message logged from

Error: Template Configuration, please edit apps.yaml

to

Error: You have not completed editing the apps.yaml template, Predbat cannot run. Please comment out 'Template: True' line in apps.yaml to start Predbat running

and changed the predbat.status value set from

Error: Template Configuration, please edit apps.yaml

to

Error: Template Configuration, remove 'Template: True' line in apps.yaml to start predbat running

zarch1972, 2025-05-14:
That sounds perfect.