springfall2008/batpred #2355: Predbat crashing during tariff comparison

hnb2907, 2025-05-12:
**Describe the bug**
Works perfectly when tariff comparison is not enabled.
When adding in any number of comparisons (based on the examples from [here](https://springfall2008.github.io/batpred/compare/), then receive repeating errors in log:

> `6879	2025-05-12 18:47:40.698051: Error: Too many API errors, stopping
6837	2025-05-12 18:47:15.346995: Error: 0
6836	2025-05-12 18:47:15.346852: Info: record_status Error: Exception raised 0
6834	KeyError: 0
6820	2025-05-12 18:47:15.326719: Error: Traceback (most recent call last):
6819	2025-05-12 18:47:15.321691: Error: Exception raised 0
6542	2025-05-12 18:44:59.584673: Error: Too many API errors, stopping
6500	2025-05-12 18:44:34.152298: Error: 0
6499	2025-05-12 18:44:34.152153: Info: record_status Error: Exception raised 0
6497	KeyError: 0
6483	2025-05-12 18:44:34.133901: Error: Traceback (most recent call last):
6482	2025-05-12 18:44:34.128822: Error: Exception raised 0
6193	2025-05-12 18:42:15.538098: Error: Too many API errors, stopping
6151	2025-05-12 18:41:50.140287: Error: 0
6150	2025-05-12 18:41:50.140132: Info: record_status Error: Exception raised 0
6148	KeyError: 0
6134	2025-05-12 18:41:50.114381: Error: Traceback (most recent call last):
6133	2025-05-12 18:41:50.108985: Error: Exception raised 0`

This causes predbat to crash/restart.
Then predbat status rarely gets out of "None" before restarting again.
Accessing any predbat status page returns "502: Bad Gateway" while predbat restarts.

Sometimes the comparison does complete.

Removing the comparisons, and restarting predbat, everything works as expected, without crashing/restarting.

I'm currently not registered for export, so "#metric_octopus_export:" is commented out.

I thought maybe the comparisons with fixed export rate = 0.0 was causing the crash, but setting the fixed rate to >0.0 also crashes.

**Expected behaviour**

**Predbat version**
Definitely present on v8.19.9 and v8.19.2

**Environment details**

- Foxess H1-3.7-E-G2 & 3x EP11H batteries.  Connected by RS485 using "FoxESS - Modbus".
- Standard HAOS, installed on x86 PC, version 2025.5.1

[apps.yaml.txt](https://github.com/user-attachments/files/20171915/apps.yaml.txt)

**Screenshots**

[image]
[image]
[image]
[image]

**Log file**
[predbat(8).log](https://github.com/user-attachments/files/20169006/predbat.8.log)

**Predbat debug yaml file**

[predbat_debug(1).yaml.txt](https://github.com/user-attachments/files/20170638/predbat_debug.1.yaml.txt)

springfall2008, 2025-05-12:
2025-05-12 18:53:01.789043: Warn: Failed to decode response <Response [404]> from http://supervisor/core/api/states/predbat.compare_tariff_agile_import/agile_export
2025-05-12 18:53:01.796587: Warn: Failed to decode response <Response [404]> from http://supervisor/core/api/states/predbat.compare_tariff_agile_import/agile_export
2025-05-12 18:53:01.807490: Warn: Failed to decode response <Response [404]> from http://supervisor/core/api/states/predbat.compare_tariff_agile_import/fixed_export
2025-05-12 18:53:01.814867: Warn: Failed to decode response <Response [404]> from http://supervisor/core/api/states/predbat.compare_tariff_agile_import/fixed_export
2025-05-12 18:53:01.822571: Warn: Failed to decode response <Response [404]> from http://supervisor/core/api/states/predbat.compare_tariff_agile_import/no_export
2025-05-12 18:53:01.829945: Warn: Failed to decode response <Response [404]> from http://supervisor/core/api/states/predbat.compare_tariff_agile_import/no_export
2025-05-12 18:53:01.837618: Warn: Failed to decode response <Response [404]> from http://supervisor/core/api/states/predbat.compare_tariff_fixed_import/fixed_export
2025-05-12 18:53:01.845051: Warn: Failed to decode response <Response [404]> from http://supervisor/core/api/states/predbat.compare_tariff_fixed_import/fixed_export
2025-05-12 18:53:01.852719: Warn: Failed to decode response <Response [404]> from http://supervisor/core/api/states/predbat.compare_tariff_fixed_import/no_export
2025-05-12 18:53:01.860004: Warn: Failed to decode response <Response [404]> from http://supervisor/core/api/states/predbat.compare_tariff_fixed_import/no_export

springfall2008, 2025-05-12:
The problem is the slash in the 'id', please replace it with an underscore or similar

```yaml
- id: "fixed_import/no_export"
      name: "12m Fix import 20250511/No export"
      rates_import:
        - rate: 26.38
      rates_export:
        - rate: 0.0
    - id: "fixed_import/fixed_export"
      name: "12m Fix import 20250511/Fixed export"
      rates_import:
        - rate: 26.38
      rates_export:
        - rate: 0.0
    - id: "agile_import/no_export"
      name: "Agile import/No export"
      rates_import_octopus_url: "https://api.octopus.energy/v1/products/AGILE-24-10-01/electricity-tariffs/E-1R-AGILE-24-10-01-{octopus_region}/standard-unit-rates/"
      rates_export:
        - rate: 0.01
    - id: "agile_import/fixed_export"
      name: "Agile import/Fixed export"
      rates_import_octopus_url: "https://api.octopus.energy/v1/products/AGILE-24-10-01/electricity-tariffs/E-1R-AGILE-24-10-01-{octopus_region}/standard-unit-rates/"
      rates_export_octopus_url: "https://api.octopus.energy/v1/products/OUTGOING-VAR-BB-24-10-26/electricity-tariffs/E-1R-OUTGOING-VAR-BB-24-10-26-{octopus_region}/standard-unit-rates/"
    - id: "agile_import/agile_export"
      name: "Agile import/Agile export"
      rates_import_octopus_url: "https://api.octopus.energy/v1/products/AGILE-24-10-01/electricity-tariffs/E-1R-AGILE-24-10-01-{octopus_region}/standard-unit-rates/"
      rates_export_octopus_url: "https://api.octopus.energy/v1/products/AGILE-OUTGOING-BB-23-02-28/electricity-tariffs/E-1R-AGILE-OUTGOING-BB-23-02-28-{octopus_region}/standard-unit-rates/"
    - id: "flux"
      name: "Flux import/export"
      rates_import_octopus_url: "https://api.octopus.energy/v1/products/FLUX-IMPORT-23-02-14/electricity-tariffs/E-1R-FLUX-IMPORT-23-02-14-{octopus_region}/standard-unit-rates"
      rates_export_octopus_url: "https://api.octopus.energy/v1/products/FLUX-EXPORT-BB-23-02-14/electricity-tariffs/E-1R-FLUX-EXPORT-BB-23-02-14-{octopus_region}/standard-unit-rates"#

```

hnb2907, 2025-05-12:
Hi, thanks for the really quick reply.  Now you've pointed out my error, it's obvious why the API calls were failing!  All working now the / are replaced with _ in the ID's. 👍 

Would it be possible to put a note [here](https://springfall2008.github.io/batpred/compare/) listing invalid ID characters?

Cheers,
Chris.

closed 2025-05-12 by hnb2907

gcoan, 2025-05-13:
> Would it be possible to put a note [here](https://springfall2008.github.io/batpred/compare/) listing invalid ID characters?

Ill put this in the docs