springfall2008/batpred #2357: Install exception errors

weevil23, 2025-05-13:
I am running HA/GivTCP/Solcast and have been doing so for years.
1 x Gen 1 5.0 2 x 9.5kWH batts

I cannot get this to install - it throws up exception error so need some help please.

[image]

gcoan, 2025-05-13:
please share your logfile and apps.yaml

weevil23, 2025-05-13:
-----------------------------------------------------------
 Add-on: Predbat
 Home Battery Prediction and Control
-----------------------------------------------------------
 Add-on version: 1.2.7
 You are running the latest version of this add-on.
 System: Home Assistant OS 15.2  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2025.5.0
 Home Assistant Supervisor: 2025.05.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[07:25:10] INFO: Configuring timezone (Europe/London)...
s6-rc: info: service base-addon-log-level successfully started
Bootstrap Predbat
Startup
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Predbat files are installed correctly for version v8.19.9
**** Starting Standalone Predbat ****
2025-05-13 07:25:12.552964: Loading apps.yaml
2025-05-13 07:25:12.690970: Info: Add-on slug is 6adb4f0d_predbat
2025-05-13 07:25:12.712262: Info: Connected to Home Assistant at http://supervisor/core
2025-05-13 07:25:12.712620: Creating task: <coroutine object HAInterface.socketLoop at 0x7f9cf297b0>
2025-05-13 07:25:12.713669: Info: Web Socket task started
2025-05-13 07:25:12.714288: Info: Start socket for url http://supervisor/core/api/websocket
2025-05-13 07:25:12.715306: Creating task: <coroutine object WebInterface.start at 0x7f9cf5ea70>
Web interface started
2025-05-13 07:25:12.742502: Info: Web Socket active
2025-05-13 07:25:12.773437: Warn: Regular expression argument: pv_forecast_today unable to match re:(sensor.(solcast_|)(pv_forecast_|)forecast_today), now will disable
2025-05-13 07:25:12.776085: Warn: Regular expression argument: pv_forecast_tomorrow unable to match re:(sensor.(solcast_|)(pv_forecast_|)forecast_tomorrow), now will disable
2025-05-13 07:25:12.778310: Warn: Regular expression argument: pv_forecast_d3 unable to match re:(sensor.(solcast_|)(pv_forecast_|)forecast_(day_3|d3)), now will disable
2025-05-13 07:25:12.780305: Warn: Regular expression argument: pv_forecast_d4 unable to match re:(sensor.(solcast_|)(pv_forecast_|)forecast_(day_4|d4)), now will disable
2025-05-13 07:25:12.782191: Warn: Regular argument car_charging_planned expression re:(sensor.wallbox_portal_status_description|sensor.myenergi_zappi_[0-9a-z]+_plug_status) failed to match - disabling this item
2025-05-13 07:25:12.783803: Warn: Regular expression argument: carbon_intensity unable to match re:(sensor.carbon_intensity_uk), now will disable
2025-05-13 07:25:12.786642: Warn: Regular expression argument: octopus_free_session unable to match re:(event.octopus_energy_([0-9a-z_]+|)_octoplus_free_electricity_session_events), now will disable
Watching ['./userinterface.py', './unit_test.py', './db_manager.py', './inverter.py', './web.py', './output.py', './ha.py', './fetch.py', './utils.py', './predheat.py', './hass.py', './solcast.py', './alertfeed.py', './compare.py', './download.py', './predbat.py', './config.py', './energydataservice.py', './apps.yaml', './plan.py', './execute.py', './prediction.py', './gecloud.py', './octopus.py', './futurerate.py'] for changes
2025-05-13 07:25:18.384913: Warn: No Octopus data in sensor sensor.octopus_energy_electricity_21l4259002_1417650400005_previous_rate attribute 'all_rates'
2025-05-13 07:25:18.384990: Warn: No Octopus data in sensor event.octopus_energy_electricity_21l4259002_1417650400005_current_day_rates attribute 'all_rates' / 'rates' / 'raw_today'
2025-05-13 07:25:18.385048: Error: metric_octopus_import is not set correctly or no energy rates can be read
2025-05-13 07:25:18.412223: Warn: record_status Error: metric_octopus_import not set correctly or no energy rates can be read
2025-05-13 07:25:18.412449: Error: Exception raised 
2025-05-13 07:25:18.416653: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 1334, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 615, in update_pred
    self.fetch_sensor_data()
  File "/config/fetch.py", line 884, in fetch_sensor_data
    raise ValueError
ValueError
2025-05-13 07:25:18.448825: Info: record_status Error: Exception raised 
2025-05-13 07:25:18.449019: Error: 
Traceback (most recent call last):
  File "/config/hass.py", line 191, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 1340, in update_time_loop
    raise e
  File "/config/predbat.py", line 1334, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 615, in update_pred
    self.fetch_sensor_data()
[... 1062 lines cut ...]
  #    name: 'Snug import/Fixed export'
  #    rates_import_octopus_url: 'https://api.octopus.energy/v1/products/SNUG-24-11-07/electricity-tariffs/E-1R-SNUG-24-11-07-{octopus_region}/standard-unit-rates/'
[... cut ...]
  # This is useful for things like the Octopus Intelligent Slot sensor so that the plan update as soon as you plugin in
  # Only uncomment the items you actually have set up above in apps.yaml, of course you can add your own as well
  # Note those using +[] are lists that are appended to this list, whereas {} items are single items only
  #watch_list:
  #  - '{octopus_intelligent_slot}'
  #  - '{octopus_ready_time}'
  #  - '{octopus_charge_limit}'
  #  - '{octopus_saving_session}'
  #  - '+[car_charging_planned]'
  #  - '+[car_charging_soc]'
  #  - '{car_charging_now}'
```

gcoan, 2025-05-13:
If you look in the logfile there are two sets of errors, on Solcast and on Octopus rates.

Double check you have followed the installation instructions thoroughly https://springfall2008.github.io/batpred/install/

Is Solcast configured and working and do the entity names match the regular expressions Predbat is trying to use (as configured in apps.yaml)?

Likewise the Octopus integration setup https://springfall2008.github.io/batpred/energy-rates/#octopus-energy-home-assistant-integration

Have you enabled the events predbat requires, are the entities named correctly to match the regular expressions

weevil23, 2025-05-13:
the problem, as I see it, is that there are fields Predbat is looking for that are not on HA (everythign is standard hree as far as I am aware.) So the attributes for sensor.octopus_energy_electricity_21l4259002_1417650400005_previous_rate does note appear to have an 'all rates' attribute

[image]

And there is NO sensor event.octopus_energy_electricity_21l4259002_1417650400005_current_day_rates on HA - maybe because I run agile??

[image]

weevil23, 2025-05-13:
if I pull the agile rates externally it runs so it must be down to where the app is calling them locally

``` # Can be used instead of the plugin to get import rates directly online
  # Overrides metric_octopus_import and rates_import
  # rates_import_octopus_url : "https://api.octopus.energy/v1/products/FLUX-IMPORT-23-02-14/electricity-tariffs/E-1R-FLUX-IMPORT-23-02-14-A/standard-unit-rates"
  rates_import_octopus_url : "https://api.octopus.energy/v1/products/AGILE-BB-24-10-01/electricity-tariffs/E-1R-AGILE-BB-24-10-01-A/standard-unit-rates/"

  # Overrides metric_octopus_export and rates_export
  # rates_export_octopus_url: "https://api.octopus.energy/v1/products/FLUX-EXPORT-BB-23-02-14/electricity-tariffs/E-1R-FLUX-EXPORT-BB-23-02-14-A/standard-unit-rates"
  rates_export_octopus_url: "https://api.octopus.energy/v1/products/AGILE-OUTGOING-BB-23-02-28/electricity-tariffs/E-1R-AGILE-OUTGOING-BB-23-02-28-A/standard-unit-rates/"
  # rates_export_octopus_url: "https://api.octopus.energy/v1/products/OUTGOING-FIX-12M-BB-23-02-09/electricity-tariffs/E-1R-OUTGOING-FIX-12M-BB-23-02-09-A/standard-unit-rates/"```

weevil23, 2025-05-13:
Sorry guys - all down to this :
https://springfall2008.github.io/batpred/energy-rates/

Working now - apologies

gcoan, 2025-05-13:
there isn't an 'all rates' attribute on 'sensor.octopus....previous_rate'

[image]

and there is an event.octopus.....current_day_rates sensor, and am on Octopus Agile, so that's not your problem

Have you enabled this event per the installation instructions - it is disabled by default.

If you installed the Octopus integration a long while ago it may have some of the entity names wrong - I have seen this happen.  It might be worth uninstalling it, rebooting HA, and reinstalling the integration to get the names right,  or figuring out what the names should be	that predbat is looking for

gcoan, 2025-05-13:
> if I pull the agile rates externally it runs so it must be down to where the app is calling them locally
> 
>   # Overrides metric_octopus_import and rates_import
>   # rates_import_octopus_url : "https://api.octopus.energy/v1/products/FLUX-IMPORT-23-02-14/electricity-tariffs/E-1R-FLUX-IMPORT-23-02-14-A/standard-unit-rates"
>   rates_import_octopus_url : "https://api.octopus.ener

you're sorted now, but just to clarify,  there are 3 different ways predbat can find the import/export rates, either manual rate configuration, the Octopus integration events/sensors or octopus URL's. If you are using the Octopus integration (which is easiest as it will automatically update as tariff rates change or you change tariff) then the URL's are not used and can be commented out in apps.yaml

weevil23, 2025-05-13:
Yeah running Octopus integration too - all seems to be working fine.

One question - slightly OT - I have two x 9.5kWh Givbats here -  does the batt charging curve need to alter when there are two...??
 battery_charge_power_curve:
    91 : 0.91
    92 : 0.81
    93 : 0.71
    94 : 0.62
    95 : 0.52
    96 : 0.43
    97 : 0.33
    98 : 0.24
    99 : 0.24
    100 : 0.24

closed 2025-05-13 by weevil23

gcoan, 2025-05-13:
> One question - slightly OT - I have two x 9.5kWh Givbats here - does the batt charging curve need to alter when there are two...??
> battery_charge_power_curve:
> 91 : 0.91

I believe both batteries will charge and discharge at the same rate so no is the answer.

But that curve in the template is really a starter for 10 and different BMS's are behaving diferently  I'd suggest you don't define a battery charge or discharge curve and let predbat generate a curve for you from historical data and then copy the resultant curve into apps.yaml - however to do this you do need to be doing full rate charge/discharge or else the curve will be wrong

My curves on BMS 3015 are different:

```
  battery_charge_power_curve:
  #  auto
    100 : 0.14
    99 : 0.14
    98 : 0.42
    97 : 0.42
    96 : 0.42
    95 : 0.71
    94 : 0.71
    93 : 0.71
    92 : 0.71
    91 : 0.88
    90 : 0.9

  battery_discharge_power_curve:
  #  auto
    20 : 1.0
```