springfall2008/batpred #2359: Solcast solar forecast is one hour early? timezone issue?

zarch1972, 2025-05-13:
Does anyone have any idea why my solcast forecast is one hour out?

I'm assuming it's a simple timezone / daylights savings setting, but I can't see where.

My Home Assistant timezone is settings is (GMT+00:00) London
(time is showing correct on the clock/time sensor)

Solcast account says Local time (Europe/London)

Timezone in apps.yaml
```
# Timezone to work in
  timezone: Europe/London
```

i'm using the internal solcast cloud interface
```
 # Solcast cloud interface, set this or the local interface below
  solcast_host: 'https://api.solcast.com.au/'
  solcast_api_key: 'NhjjXXXXXXXXwCv'
  solcast_poll_hours: 8
```
I've checked closed issues and looked in the documentation but can't see anything similar.

Any pointers or thoughts would be appreciated, thanks.

[image]

[image]

zarch1972, 2025-05-13:
Okay, I might have worked this out.

I didn't realise that if you had two PV sites within solcast it aggregated the forecast from both into a single figure (that predbat then uses).

I'd put in a second PV site as test a long time ago as I was wondering about adding some extra panels, so wanted to see what benefit they bring  So it would appear they would give me more early day generation. :-)

I've now removed that second site from within solcast, so will see how the forecast looks tomorrow.

gcoan, 2025-05-13:
Yes solcast aggregates both sites to give a single forecast in Predbat

If you are using the Solcast integration you can access the individual site breakdowns and could have excluded one site,  so left Predbat running with the 'live' site and kept the 'potential' site in there for graphing etc

zarch1972, 2025-05-17:
Happy to close this one, thank you.  Removing the second site sorted things out.

[image]

closed 2025-05-17 by gcoan