springfall2008/batpred #2361: Predbat setting charging during expensive times

SamKua, 2025-05-13:
**Describe the bug**
Predbat keeps deciding to charge my battery during expensive times. For instance, now, when my unit price is 28.96p, its decided its a good time to charge the battery. Looking at the plan, based of the colouring, it thinks that 28.96p and 8.5p are both equally good (they are both green) so its charging from the grid even though iin a few hours it will be much cheaper and the battery is almost full.
It also wants to charge my car, luckily its not the one that manages that.

**Expected behaviour**
Charge when the unit price is lower

**Predbat version**

I've seen this happen with all the 19.x versions. I had previously locked to the last v18 build and then its fine, but decided to try v19 again and now I don't even know how to force it to use the old one any more as its not in the list.

**Environment details**

- GivEnergy setup 9.8kWh battery
- I think its a docker install on a raspberry pi.

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

Sorry, I don't quite follow how to do these two.

gcoan, 2025-05-13:
> Sorry, I don't quite follow how to do these two.

If you are not able to provide a logfile or a debug file then its going to be almost impossible to identify what your problem is,  which is quite possibly a configuration issue.

If you go into the predbat web interface https://springfall2008.github.io/batpred/output-data/#web-interface click on the Dash tab and then there's an option to create a predbat_debug.yaml in the second block 

The location of the logfile is described in the documentation https://springfall2008.github.io/batpred/output-data/#predbat-logfile

Option to download it is in the same debug block of the web interface

SamKua, 2025-05-14:
Ah, I see now, thank you for the clearer explanation - I rately go to the web UI and had forgotten it even existed. I've been able to generate them, but at the moment its behaving so these ones presumably won't be much use (its done this before though, looked like its made a good plan then in the evening starts to import for no apparent reason - I'll keep monitoring it today). 

Interestingly I did manage to get it to make a good plan last night by taking away its awareness of the car. While its been a useful place to check if I plugged in the car, I don't think it should cause any issues since I manage the car charge with a different automation. I've tried undoing those changes again this morning to see if I can prompt the strange behaviour again.

cyberkryten, 2025-05-14:
I appear to have a good plan with charging only during low rate times, but also have the same 'all green' colouring for peak/off-peak values shown on the screenshot above when it used to show peak as red.  

I upgraded from v8.18.10 this morning, so probably not much help for narrowing it down

[predbat_debug-2.yaml.txt](https://github.com/user-attachments/files/20206463/predbat_debug-2.yaml.txt)

SamKua, 2025-05-14:
Yep, evening has rolled around and its pulling from the grid again. It seems it is something to do with the car, as it started when I plugged the car in ready to charge overnight. 

Having fiddled with a few settings and tried a few things, it seems convinced the car is charging. This was resulting in the house running from the grid as I had the Allow car charge from battery set to off. Turning this on has meant it is now running the house off the battery as it should, but I don't understand why it thinks the car is charging. Looking at the apps.yaml tab in the ui, I can see it has

[image]

and yet in the plan I have

[image]

Any suggestions?

[predbat (2).log](https://github.com/user-attachments/files/20213870/predbat.2.log)

[predbat_debug (1).yaml.txt](https://github.com/user-attachments/files/20213921/predbat_debug.1.yaml.txt)

SamKua, 2025-05-14:
Since there is a specific part of the docs that reference Hypervolt chargers and I was using a different sensor to identify if the car is charging, I updated my apps.yaml to match the example, and its still doing the same thing. 

[image]

It is also still doing the thing where all the prices are green (aside from one slot where I get free electric, thats blue).

johnwb87, 2025-05-16:
How are you controlling the car charging?

SamKua, 2025-05-16:
I have an automation I set up prior to predbat that gives me the control I want over the car charging. It doesn't do anything that fancy, just sets the charge mode and speed and locks or unlocks the charger dependent on the option I've selected from a dropdown. At the time I took that screenshot the car was plugged in, but charging was stopped and the charger was locked still, ready to charge overnight when the change of elextricity rate triggers the automation to let the charging start.

cyberkryten, 2025-05-16:
There's definitely something wrong with the way Predbat is ingesting pricing data, or storing it in the latest version.

Here's my text summary, which is correct:

```
Import rates are very expensive (28.8p) for the next 13 hours and then cheap (7.0p) for the next 6 hours.
The battery is currently at 59% and is in eco mode for the next 2 hours.
For the next 10 hours there will be a lot of solar generation and then it will decrease.
You will run out of battery in 13 hours.
Your next charge slot will be in 16 hours where import rates will be cheap (7.0p).
Your next export slot will be in 2 hours where export rates will be fixed (15.0p).
```

However, all the daytime high-rate slots are coloured green in the plan and looking at the individual values it shows as currently being in a low rate period @ 28p!

[image]

johnwb87, 2025-05-16:
> I have an automation I set up prior to predbat that gives me the control I want over the car charging. It doesn't do anything that fancy, just sets the charge mode and speed and locks or unlocks the [...]

I asked about controlling the car because in your first screen shot of the plan predbat thinks the car is charging for several hours and the additional charges are in that window. I’m wondering where it is getting that from? What car charging switch do you have set in Config?

SamKua, 2025-05-16:
It thinks the car is charging, but it categorically was not. It shows the two entities in the screenshots that it should be using to know the car is charging, and it shows that they were both returning values that should not have classed as charging. Is there a different section in apps.yaml it might be looking at for this?

The only extra entity created for my automation is the drop down of options for me to choose what sort of charge I want the car to have (boosted (i.e. using grid power), draining battery, or just using solar). Other than it it just manipulates the standard entities provided by the car charger, which are the same ones Predbat is looking at.

johnwb87, 2025-05-16:
It looks to me like predbat is creating/getting a schedule. But the way you have described your set up predbat should only know when the car is charging and not into the future. That’s what I would explore.

Have you considered using predbat to plan the charging and retire the automations?

SamKua, 2025-05-16:
I like the flexibility of my automation, if I want to put a little extra into the car during the day, I can either top it up slowly and just take from the battery, or really boost it if for some reason I need it quickly. I've also had a couple of occasions recently where something has gone werid with the charger overnight and it hasn't listened to the instruction to charge, and being able to test if the message is getting through easily has been a huge help. Its not like the car charging needs a lot of complicated planning, its just 'charge when its cheap'. I value predbat for doing the more complicated planning of the house battery as it helps maximise the amount we can make from export based on how our solar will do and if we get any free electric slots.

I did notice that somewhere along the way the toggle for Octopus Intelligent Charging had got turned on. and I'm wondering if that might have done something strange. I shall investigate further.

For now, having predbat think we don't have a car seems to be working acceptably.

cyberkryten, 2025-05-16:
First image is after a restart - you can see that Predbat has coloured the charging slots correctly - maybe it does this 'first time through' and there's something being leaked or not reset causing issues on future runs?

[image]

This is after the next plan calc, this was only 30 seconds after the first one, as you can see it has got the colours very wrong and thinks it is within a low rate slot.

[image]

matttheotter, 2025-05-19:
> I like the flexibility of my automation, if I want to put a little extra into the car during the day, I can either top it up slowly and just take from the battery, or really boost it if for some [...]

Morning,

I have had my Hypervolt v3 work on Agile and IOG using Predbat, there maybe an error in your apps config or automation issue.

Few questions.

- What version of Hypervolt are you using a v2 or v3?
- How is your automation setup?
- Could also do with seeing more of you apps config for the car

From the looks of the sensor response you have the automation set to `switch.hypervolt_charging` be on or off? 

This isn't the correct method of controlling the charge process, and that sensor is going to be made read only in the future. You should use `select.hypervolt_activation_mode` and have it set to "Plug and Charge" when you want to charge and "Scheduled" when you don't otherwise the car is likely to go to sleep.

Worth noting that lock/unlock is a bit of a mare to work, mine unlocks to initiate charge and will only re-lock once the car has been unplugged for 5 minutes as the charger won't accept a lock command when a charge has started and is plugged in.

cyberkryten, 2025-05-19:
I posted a Debug a few days back - my plans seem Ok but the colours are wrong and the "Next low rate start" values are also incorrect.  It is intermittent though and I can't quite figure out what's causing it.

My system is unchanged from last week but every time I've checked this morning the values are all correct - will post a Debug when I spot it wrong again.

SamKua, 2025-05-19:
> Morning,
>
> I have had my Hypervolt v3 work on Agile and IOG using Predbat, there maybe an error in your apps config or automation issue.
>
> Few questions.
>
>     * What version of Hypervolt are you using [...]

I have a v3 Hypervolt.

The automation just sets the various Hypervolt entities to make the car charge at my desired max current and mode, and turns the lock on or off depending. There is no reason it should be interacting with predbats planning - its been working this way for the entire time I've had predbat and been fine. Predbat has been correctly indicating that the car should charge in the cheap overnight slot, though it has not been the one in control of it starting. Its only since upgrading to v8.19.x that its started thinking the car is charging/should charge when it isn't.

I previously had the sensor checking the hypervolt_charging switch, but when checking through the docs I saw that it recommended using the charging_readiness instead, and adjusted my apps.yaml accordingly. I showed this in the next comment. Changing this did not fix the issue - in the ui it is showing that the value for charging_readiness is 'Not Ready - Force Stopped', while the expected value to show it is actually charging is 'charging'.

SamKua, 2025-05-19:
> I'd really need a debug yaml captured when the plan is wrong, the screenshot above looks fine apart from colours. Can you grab one please?

I have not managed to induce it to start charging the house battery again since messing with the settings to try desperately to get it to stop before. 

This is one where it is thinking that my car is charging when it isn't. So long as I have predbat set to allow the car to charge from the battery this isn't a massive issue, but it does mess with the short term predictions so its easier to just tell predbat the car doesn't exist, at which point everything works fine.

[predbat_debug (2).yaml.txt](https://github.com/user-attachments/files/20301799/predbat_debug.2.yaml.txt)

matttheotter, 2025-05-19:
Would it not be easier to sign up to IOG?

Lower rate, longer off peak period and is compatible with the Hypervolt.

SamKua, 2025-05-20:
I've been considering it, I've been having an absoute mare with Octopus and am waiting on an Ombudsman complaint at the moment. Given I'll feel the need to monitor it really closely for a while before I'm happy with it, I've not had the energy to go through with it just yet. I'll probably do it in the summer holidays when its less of an issue if something goes wonky with the charge.

cyberkryten, 2025-05-20:
> I'd really need a debug yaml captured when the plan is wrong, the screenshot above looks fine apart from colours. Can you grab one please?

Well, it's taken a while but I've caught one again, though once again the plan itself looks correct now.  

However, hopefully this is useful as it demonstrates two things:

One is the colours being wrong - all slots are green low-rate - I suspected this was related to the original issue, as I'd also seen some strange plans and "Next Low Rate" values were totally wrong.  However, they seem to be correct (plans and colours) most of the time I've checked in the last 24 hours, so maybe these are different issues.

This debug also shows the Clip kWh values being incorrect - it is showing clipping overnight from 20:30 to 22:30 after sunset and also again at 07:30 when there isn't enough Solar to clip.

[predbat_plan.html.txt](https://github.com/user-attachments/files/20342991/predbat_plan.html.txt)

[predbat_debug.yaml.txt](https://github.com/user-attachments/files/20342997/predbat_debug.yaml.txt)

PhillWalker, 2025-06-27:
I believe I have the same issue, my battery has been charging at expensive rates overnight, when if predate had just let the battery go flat it could have had access to rates significantly cheaper.    (charge at 16p, could have accessed 4p)

Apollo1101, 2025-12-18:
Having the same issue 38 and 27p import is being treated as the same. seems to swap green red every now and again from what is shown here. 

[image]

[predbat (1).log](https://github.com/user-attachments/files/24242958/predbat.1.log)

gcoan, 2025-12-18:
> Having the same issue 38 and 27p import is being treated as the same. seems to swap green red every now and again from what is shown here.

its not obvious from your sample of the plan that this is the same issue,  other than perhaps the colour coding of the import rate.
The plan shows a period of charging in the cheap rate, but otherwise you are running in demand mode off the batteries with no charging at the wrong times.

As per earlier requests, Trefor needs to see the full plan, the logfile and the debug file captured when there is a problem to be able to investigate it

closed 2026-08-20 by chalfontchubby