springfall2008/batpred #2364: Not Charging To 100% - Stuck on 2.6kw charge/discharge rate

Evaske, 2025-05-13:
**Describe the bug**
I have just had a Sigenergy system fitted with a 6kw inverter and 24kw of batteries. I don't currently have any PV fitted so have set that to 0 and the export isn't set up yet so I the export rate is set at 0.

I am on Octopus Intelligent Go

For some reason Predbat is not charging the battery to 100% during the cheap period (2330 - 0530) and is setting a charge limit and I can't really work out why as it is evident from the plan that I will run out of battery power before the next cheap charge rate.

It is also charging at peak rate at random points during the day.

On looking in to the logs I can see that it says the following:

```
Inverter 0 SOC: 1.74kW 7% Current charge rate 8400W Current discharge rate 8400W Current power -1022.0W Current voltage 52.0V
Inverter 0 with soc_max 24.18 kWh nominal_capacity 24.18 kWh battery rate raw 2600.0 w charge rate 2.6 kW discharge rate 2.6 kW battery_rate_min 0.0 w ac limit 6.0 kW export limit 5999940.0 kW reserve 4.0 % current_reserve 4.0 % temperature 26 c
```

As you can see, the first line is picking up the charge rate and discharge rate from the `input_number.charge_rate` correctly but the line below is not and is setting it to 2.6kw which I'm guessing is the issue but I can't work out where it is getting this value from apart from it being the default if it can't find values elsewhere.

If I was to set the battery scaling to 2.5 to effectively get the 2.6k default up to 6.5kw then it indeed charges the battery to 100%

**Expected behaviour**
Charge the battery to 100% during the off peak period or at least to a % that will power the house for the entire day

**Predbat version**

8.19.10

**Environment details**

- Sigenergy Sigenstor 6kw inverter and 24kw batteries
- HAOS on Proxmox

**Screenshots**

[image]
[image]

**Log file**

[predbat.log](https://github.com/user-attachments/files/20195192/predbat.log)

**Predbat debug yaml file**

[predbat_debug_21_15_0.txt](https://github.com/user-attachments/files/20195262/predbat_debug_21_15_0.txt)

gcoan, 2025-05-13:
Looks like there is a fault in the Sig inverter definition that its not pulling in the charge and discharge rate correctly.  The 2600W is the default value for the GivEnergy inverters that Predbat was originally written for

Evaske, 2025-05-13:
> Looks like there is a fault in the Sig inverter definition that its not pulling in the charge and discharge rate correctly. The 2600W is the default value for the GivEnergy inverters that Predbat was originally written for

Any idea what I can edit to give it an attempt at fixing it? 😊

gcoan, 2025-05-13:
> Any idea what I can edit to give it an attempt at fixing it? 😊

no not really.  I have looked at the SIG definition in config.py and I don't see any obvious issues (other than the name is wrong).  
I'd need to dig through the code to work out where the different messages come from, what variables are being used and why there is a discrepancy

probably one for @springfall2008

gcoan, 2025-05-13:
you could try creating a custom inverter definition using the config in #2077 (call it something different to SIG) and edit apps.yaml to refer to the new name, and see if that is any better

that would point to whether it was an original inverter definition issue or whether it was when Sig was merged into predbat core

Evaske, 2025-05-14:
After doing some research I noticed it was picking the values up from this code:

```
if self.inverter_type in ["GE", "GEC", "GEE"]:
                self.battery_rate_max_raw = self.base.get_arg("charge_rate", attribute="max", index=self.id, default=2600.0)
            elif "battery_rate_max" in self.base.args:
                self.battery_rate_max_raw = self.base.get_arg("battery_rate_max", index=self.id, default=2600.0)
            else:
                self.battery_rate_max_raw = 2600.0
```

There was no `battery_rate_max` argument set in the sigenergy app.yaml. After adding that and matching it to the inverter max battery rate, it was showing the correct 6kw charge limit on the graphs and charging the battery properly to 100%.

springfall2008, 2025-05-19:
Resolved :)

closed 2025-05-19 by springfall2008

gcoan, 2025-05-19:
@Evaske I have added battery_rate_max for the SigStore inverter to the Sig apps.yaml template and updated the documentation to make clear the importance of this being set for non-GivEnergy inverters