springfall2008/batpred #2366: New Gen3 Single phase high voltage inverter and batteries

TheSargey, 2025-05-14:
Hi folks, hope you're well.

I mentioned a while back I was upgrading to the latest GE inverter and batteries, it's 10kW inverter (Single phase) paired with 20kW of HV batteries.

Seems like the kit is still in it's infancy, had big issues getting the battery firmware upgraded (ended up having to go via USB direct into the battery stack) but we're up and running today.  Pretty frightening to see it suddenly pulling 10kW from the grid to charge the battery!

Anyway, I digress.  One issue I've got (which I've raised with GE directly) is the local modbus connection seems to vanish after a couple of hours, however the cloud information continues to work.  A quick restart of the inverter allows GivTCP to get it's connection back and things work as expected.

Whilst GE looks into this, I'm just looking at the restart options I have available in apps.yaml - namely:

  # When enabled automatic restart will restart the add-on if communication fails
  # Example below is auto-restart for GivTCP add-on itself
  auto_restart:
    - shell: 'rm -rf /homeassistant/GivTCP/*.pkl'
    - service: hassio/addon_restart
      addon: 533ea71a_givtcp

  #  Example on how to restart the inverter via GivTCP
  #  - service: switch.turn_on
  #    entity_id: switch.givtcp_{geserial}_reboot_invertor

If I simply enable the inverter restart by removing the hashes on the last two lines, does this send a restart via the cloud or locally?  I'm guessing locally, however the problem I have is there is no local control when the inverter "disappears" however I can restart the inverter via the app/cloud.  Just trying to get it so that predbat can restart the inverter when the clock skews and REST becomes unavailable (which is basically the sign that the inverter has for some reason gone to sleep locally).

Just wanted to get your thoughts on this.

Thanks as always for your help.

Sarge

gcoan, 2025-05-14:
All such commands are made via GivTCP locally through modbus, so wouldn't work for you.

Might be worth looking at the GivEnergy API's to see if there is an API to poke the inverter to restart itself.

Alternatively why not use the GE cloud integration already built into predbat and get the data from the cloud rather than locally via GivTCP.  https://springfall2008.github.io/batpred/inverter-setup/#givenergy-with-ge_cloud

I have used the older method of the ge_cloud integration that Trefor wrote when I had an issue with accidentally zapping all the history of my givtcp import sensor, so predbat broke,  and using that cloud integration worked perfectly.

Also, are you sure it is an inverter issue?  Have you tried the latest beta or dev versions of givtcp in case its fixed there?

TheSargey, 2025-05-14:
Hi Geoff, thanks as always for your advice.

I was looking at the GE Cloud integration, I can give that a go for sure.

I think it's definitely a GE issue rather than a GivTCP as I lose the local connection for the GE app at the same time as losing GivTCP, which suggests to me it's the inverter "shutting shop" and disallowing local network connection somehow.  What do you think?

Also, I have an issue with PredBat updating the charge/discharge rate for the inverter to 10kW when I'm trying to keep this to around 7kW until I update the breaker I've got in the consumer unit.  I've set in apps.yaml the rate (7000) but for some reason it's forcing GivTCP to reset this to 10000 - I've ascertained it's PredBat as I can set it to read only and the update doesn't happen via GivTCP, but if I let PredBat take control it changes the charge/discharge rate that I've set directly in the inverter... Here's the line in GivTCP
Setting battery charge rate 8240 was a success
2025-05-14 22:53:36,394 - GivTCPHV - write       -  [INFO    ] - Setting battery charge rate 8240 was a success
2025-05-14 22:53
It says 8240 but when I look at the inverter it's set it to 10000.  I'm not sure what setting in PredBat would be forcing this to max, would you know off the top of your head?

Many thanks as always,

Sarge

TheSargey, 2025-05-14:
I think I found it - inverter_limit_charge and discharge - just testing this now

TheSargey, 2025-05-14:
> I think I found it - inverter_limit_charge and discharge - just testing this now

This worked, had to set it to 5000 in yaml and it worked out to 7170 in the inverter.

gcoan, 2025-05-14:
> > I think I found it - inverter_limit_charge and discharge - just testing this now
> 
> This worked, had to set it to 5000 in yaml and it worked out to 7170 in the inverter.

I was going to say that, by default Predbat will do all charge/discharge at the max inverter rate.  Setting these in apps.yaml overwrites this behaviour.

It does sound like it could be an inverter issue if the givenergy app stops working.  Have you tried the BBC basic or Rubikcube Android app, do they stop as well?

TheSargey, 2025-05-15:
Hi Geoffrey, no I've not tried those apps, hadn't heard of them until you
mentioned it.  I'll have a look into.

Cheers,

Luke

gcoan, 2025-05-15:
Here's a link to the ios version of the BBC app https://apps.apple.com/app/id6443976497

TheSargey, 2025-05-19:
Morning Geoff, hope you're well.

I'm setting up the ge_cloud as it seems that this is a more stable way of being able to monitor my inverter whilst I work on trying to get the local monitoring sorted (not sure how likely that will be in the near future).  

One thing I've noted is in the sensor setup, in the usual givtcp sensors there is inverter_mode which switches Eco on and off etc.  However this is missing from the givenergy_cloud.yaml - I think the most appropriate entity this would link to is switch.ge_inverter_{geserial}_enable_eco_mode but I assume it's been removed for a reason.  Should I just comment out inverter_mode whilst using the cloud?  Just wanted to make sure I'm setting it up correctly - potentially one for Trefor to clarify.

Cheers,

Sarge

TheSargey, 2025-05-19:
Hmm - getting the following in predbat log:
2025-05-19 10:25:00.732545: Warn: Error downloading GE data from URL https://api.givenergy.cloud/v1/inverter/{my inverter serial}/data-points/2025-05-04?pageSize=4096
2025-05-19 10:25:00.771645: Info: record_status Warn: Error downloading GE data from cloud
2025-05-19 10:25:02.487263: Warn: Load mean is zero, unable to calculate divergence!

However, the entities for ge_cloud are all populating as expected.  Any idea how can I establish what the problem is here?

TheSargey, 2025-05-19:
> Morning Geoff, hope you're well.
>
> I'm setting up the ge_cloud as it seems that this is a more stable way of being able to monitor my inverter whilst I work on trying to get the local monitoring [...]

I think I've found this - looks like the inverter mode is set by the inverter_type being 'GEC' under num_inverters.

gcoan, 2025-05-19:
> I think I've found this - looks like the inverter mode is set by the inverter_type being 'GEC' under num_inverters.

it wouldn't surprise me if it does special things for the GE inverter types.  So is the template OK or need changing?

> Hmm - getting the following in predbat log:
> 2025-05-19 10:25:00.732545: Warn: Error downloading GE data from URL https://api.givenergy.cloud/v1/inverter/{my inverter [...]

this sounds more serious.  If you look in the portal data, what do you see in the load column, is it populated properly?

if its not (and this is thus a givenergy issue) then a workaround would be to calculate the load yourself from a template sensor, example in the docs https://springfall2008.github.io/batpred/apps-yaml/#data-from-home-assistant but you'd have to wait for this to run for a while to build up history

TheSargey, 2025-05-19:
Evening,

I dont think any changes are required on the template.

All of the data is populated appropriately in the load. Looking at the link that Predbat is trying to retrieve, it's giving a date of 2025-05-04 (I. E. May 4th), which makes me think it's trying to look back a couple of weeks from today to get load data (which makes sense as my apps.yaml looks at 7 and 14 days for historical data) but obviously I've only had the new inverter for a week so I'd assume it's failing as the cloud doesn't have data for my specific inverter going back that far. 

I'm going to set my apps.yaml to only look back 7 days and see if that clears the error.

Cheers,

Sarge

gcoan, 2025-05-19:
yes predbat will try to load import data, load history, PV data, etc based upon days_previous

so if you don't have this history in either givtcp or the cloud then its not going to work well!

reducing days_previous should resolve this

TheSargey, 2025-05-19:
> yes predbat will try to load import data, load history, PV data, etc based upon days_previous
> 
> so if you don't have this history in either givtcp or the cloud then its not going to work well!
> 
> reducing days_previous should resolve this

Exactly, changed it to 7 and the errors are gone. Thanks for being my sound board sir!

TheSargey, 2025-05-20:
Hi @gcoan 

Im getting the cloud data error again, I've tried to shift the days previous around (even just setting it to 1) but it fires the following 

2025-05-21 00:27:47.032734: Info: record_status Warn: Error downloading GE data from cloud
2025-05-21 00:27:48.238438: Warn: Load mean is zero, unable to calculate divergence!

https://api.givenergy.cloud/v1/inverter/{geserial}/data-points/2025-05-21?pageSize=4096

Load data exists in the cloud, predbat was working happily until midnight which then triggered this error. Is it potentially to do with the link that predbat is trying to use to retrieve the historical data? Seems the switch from 2025-05-20 to -21 in the api call could be the culprit, what do you think? 

Inverter is online, ge cloud is up and populated. Not sure what's going on, I've tried the restart predbat etc but the load is not showing now in predbat. 

Curious!

gcoan, 2025-05-21:
Its curious that it started reporting the error just after midnight.  I would avoid days_previous of 1 because that can cause weird results

Let's see if it happens again.

One option is to set a base load figure https://springfall2008.github.io/batpred/apps-yaml/#base-load which predbat will use if the load data is zero.  Not sure if this would fix this issue

gcoan, 2025-05-22:
Interesting.  The inability to read load data early in the morning seems a givenergy portal/API bug, if the data isn't there for 8 hours then as you suspected, something must be delaying it on the portal end.

Was Predbat trying to retrieve today's data?

@springfall2008  Possibly predbat could be changed to only retrieve the historical data for the days_previous you require which could avoid the issue?

The failure to read charge date/time happens to me as well with givtcp.  Usually corresponds to a read error reported in the givtcp log. Predbat farts about the missing data, givtcp errors and restarts itself, and sometimes predbat has to intervene by restarting the givtcp addon.   But for me it's a transient issue and sorts itself out.  Unrelated I think

TheSargey, 2025-05-23:
> 
> Was Predbat trying to retrieve today's data?
> 

Morning Geoff, it seems that Predbat (when it turns midnight) is trying to retrieve a data file using that current days date (i.e. this morning at the turn of midnight the following error posted):

2025-05-23 00:00:02.704955: Warn: Error downloading GE data from URL https://api.givenergy.cloud/v1/inverter/{geserial}/data-points/**2025-05-23**?pageSize=4096
2025-05-23 00:00:02.770747: Info: record_status Warn: Error downloading GE data from cloud
2025-05-23 00:00:06.121975: Warn: Load mean is zero, unable to calculate divergence!
2025-05-23 00:00:37.023710: Info: record_status Read-Only

This error then continously posts every five minutes in line with PredBats plan update policy.  I've still got the previous_days calculations set to 5,6,7 (not using 1).  As soon as 8am comes the following occurs:

2025-05-23 08:00:48.900804: Info: record_status Read-Only
2025-05-23 08:00:50.359651: Info: Refresh config entities due to their age of 480.0 minutes
2025-05-23 08:01:20.347058: Info: record_status Read-Only

Then the load data comes back into the Predbat plan.  The issue here is during the eight hour down time, the plan loses all of the load_kWh data - regardless of what days_previous is set.

As you've suggested, if Predbat can stop trying to retrieve the file using the current days date at midnight and instead continues to use the previous days date (i.e. for today it would use https://api.givenergy.cloud/v1/inverter/{geserial}/data-points/**2025-05-22**?pageSize=4096 until 8am) then potentially that would fix it.  

Cheers,

Luke

gcoan, 2025-05-23:
> As you've suggested, if Predbat can stop trying to retrieve the file using the current days date at midnight and instead continues to use the previous days date (i.e. for today it would use [...]

I think the solution is simpler than special coding around midnight, Predbat should be retrieving load data from the GE Cloud integration using days_previous

So just after midnight this morning (23/5/25), with your days_previous of 5, 6 and 7, Predbat should be trying to retrieve load data from 23-5 =18/5/25, 17/5/25 and 16/5/25, not starting from today's date (23/5/25) and working backwards.

I've changed the title to reflect this specific issue @springfall2008

TheSargey, 2025-05-23:
Ok, thanks gentlemen.  Happy to help test when any updates are available.

TheSargey, 2025-05-31:
> Fix on main for testing

Thanks Trefor, I've installed will report back tomorrow.

TheSargey, 2025-06-01:
> Fix on main for testing

Morning - sadly after update still get the error:

2025-06-01 00:00:03.680334: Warn: Error downloading GE data from URL https://api.givenergy.cloud/v1/inverter/{ge_serial}/data-points/2025-06-01?pageSize=4096
2025-06-01 00:00:03.721956: Info: record_status Warn: Error downloading GE data from cloud
2025-06-01 00:00:08.464342: Warn: Load mean is zero, unable to calculate divergence!

As per last time, this goes until 8am when the file must become available.

springfall2008, 2025-06-08:
The reason it downloads all the data and not just the days previous is the load data is also used for cost calculations.

I've pushed another fix to 'main' for testing.

TheSargey, 2025-06-08:
Thanks Trefor, I've updated, will let you know how it goes in the morning.

TheSargey, 2025-06-10:
@springfall2008 Sadly issue still persists after main update.

springfall2008, 2025-06-16:
Do you have a logfile with the error?

springfall2008, 2025-06-16:
A copy of the logfile including the time around midnight would be good.

I don't see why missing data at midnight would impact 7 days ago

springfall2008, 2025-06-16:
New fix released in .2 which might help

TheSargey, 2025-06-16:
> New fix released in .2 which might help

Thanks Trefor, I've updated to main, will let you know how it goes this evening.  Here's a sample from predbat.log from the 15th at midnight when it tries to get the latest file from the cloud (note I'm running in read only at the moment whilst I'm working on the latest HV gen3 GE inverter):

----<snip>----
2025-06-15 00:00:00.458333: --------------- PredBat - update at 2025-06-15 00:00:00+01:00 with clock skew 0 minutes, minutes now 0
2025-06-15 00:00:00.462928: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json
2025-06-15 00:00:00.466889: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 45.0 minutes
2025-06-15 00:00:00.466988: Predbat /config/predbat.py version v8.20.15 currently running, latest version is v8.20.14 latest beta v8.20.14
2025-06-15 00:00:00.467156: Autoupdate: Currently on unknown version v8.20.15 (?)
2025-06-15 00:00:00.575146: Configuration: forecast_hours 48 num_cars 0 debug enable is False calculate_plan_every 10
2025-06-15 00:00:00.575550: Predbat mode is set to Control charge
2025-06-15 00:00:00.575622: NOTE: Read-only mode is enabled, the inverter controls will not be used!!
2025-06-15 00:00:01.212202: Fetching https://api.givenergy.cloud/v1/inverter/{ge_serial}/data-points/2025-06-07?pageSize=4096
2025-06-15 00:00:03.273064: Return cached GE data for https://api.givenergy.cloud/v1/inverter/{ge_serial}/data-points/2025-06-08?pageSize=4096 age 460.0 minutes
2025-06-15 00:00:03.283665: Return cached GE data for https://api.givenergy.cloud/v1/inverter/{ge_serial}/data-points/2025-06-09?pageSize=4096 age 460.0 minutes
2025-06-15 00:00:03.290086: Fetching https://api.givenergy.cloud/v1/inverter/{ge_serial}/data-points/2025-06-10?pageSize=4096
2025-06-15 00:00:04.481907: Fetching https://api.givenergy.cloud/v1/inverter/{ge_serial}/data-points/2025-06-11?pageSize=4096
2025-06-15 00:00:05.695994: Fetching https://api.givenergy.cloud/v1/inverter/{ge_serial}/data-points/2025-06-12?pageSize=4096
2025-06-15 00:00:06.333660: Fetching https://api.givenergy.cloud/v1/inverter/{ge_serial}/data-points/2025-06-13?pageSize=4096
2025-06-15 00:00:07.035372: Fetching https://api.givenergy.cloud/v1/inverter/{ge_serial}/data-points/2025-06-14?pageSize=4096
2025-06-15 00:00:07.490995: Fetching https://api.givenergy.cloud/v1/inverter/{ge_serial}/data-points/2025-06-15?pageSize=4096
2025-06-15 00:00:07.693975: Warn: Error downloading GE data from URL https://api.givenergy.cloud/v1/inverter/{ge_serial}/data-points/2025-06-15?pageSize=4096
2025-06-15 00:00:07.761993: Info: record_status Warn: Error downloading GE data from cloud
....
2025-06-15 00:00:09.974545: Warn: Load mean is zero, unable to calculate divergence!
2025-06-15 00:00:09.974666: Load divergence over 8.0 hours mean 0.0 W, min 0.0 W, max 0 W, std dev 0.0 W, divergence 0.0%
---<snip>---

If I go back five minutes before this to 23:55 this is what the log shows:

---<snip>---
2025-06-14 23:55:01.196598: --------------- PredBat - update at 2025-06-14 23:55:00+01:00 with clock skew 0 minutes, minutes now 1435
2025-06-14 23:55:01.197610: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json
2025-06-14 23:55:01.211973: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 40.0 minutes
2025-06-14 23:55:01.212164: Predbat /config/predbat.py version v8.20.15 currently running, latest version is v8.20.14 latest beta v8.20.14
2025-06-14 23:55:01.212300: Autoupdate: Currently on unknown version v8.20.15 (?)
2025-06-14 23:55:01.294191: Configuration: forecast_hours 48 num_cars 0 debug enable is False calculate_plan_every 10
2025-06-14 23:55:01.294616: Predbat mode is set to Control charge
2025-06-14 23:55:01.294685: NOTE: Read-only mode is enabled, the inverter controls will not be used!!
2025-06-14 23:55:01.660723: Return cached GE data for https://api.givenergy.cloud/v1/inverter/{ge_serial}/data-points/2025-06-06?pageSize=4096 age 25.0 minutes
2025-06-14 23:55:01.663355: Return cached GE data for https://api.givenergy.cloud/v1/inverter/{ge_serial}/data-points/2025-06-07?pageSize=4096 age 455.0 minutes
2025-06-14 23:55:01.665847: Return cached GE data for https://api.givenergy.cloud/v1/inverter/{ge_serial}/data-points/2025-06-08?pageSize=4096 age 455.0 minutes
2025-06-14 23:55:01.671823: Return cached GE data for https://api.givenergy.cloud/v1/inverter/{ge_serial}/data-points/2025-06-09?pageSize=4096 age 455.0 minutes
2025-06-14 23:55:01.675587: Return cached GE data for https://api.givenergy.cloud/v1/inverter/{ge_serial}/data-points/2025-06-10?pageSize=4096 age 475.0 minutes
2025-06-14 23:55:01.679371: Return cached GE data for https://api.givenergy.cloud/v1/inverter/{ge_serial}/data-points/2025-06-11?pageSize=4096 age 475.0 minutes
2025-06-14 23:55:01.683492: Return cached GE data for https://api.givenergy.cloud/v1/inverter/{ge_serial}/data-points/2025-06-12?pageSize=4096 age 475.0 minutes
2025-06-14 23:55:01.687115: Return cached GE data for https://api.givenergy.cloud/v1/inverter/{ge_serial}/data-points/2025-06-13?pageSize=4096 age 475.0 minutes
2025-06-14 23:55:01.689522: Return cached GE data for https://api.givenergy.cloud/v1/inverter/{ge_serial}/data-points/2025-06-14?pageSize=4096 age 475.0 minutes
2025-06-14 23:55:03.147283: Downloaded 12953 datapoints from GE going back 8 days
...
[... cut ...]
2025-06-15 00:00:09.626121: PV Forecast for day tomorrow is 22.3 (15.85 10% 26.44 90%) kWh
2025-06-15 00:00:09.668866: PV Forecast for day d2 is 24.97 (13.01 10% 26.48 90%) kWh
2025-06-15 00:00:09.704875: PV Forecast for day d3 is 21.57 (11.74 10% 26.8 90%) kWh
2025-06-15 00:00:09.821881: Inverter 0: Type GEC GivEnergy Cloud
---<snip>---

So it seems to ignore the historical load calculation completely when it errors on the file download.

Is this enough info from the logfile or do you want to see more?

Cheers,

Sarge

springfall2008, 2025-07-03:
New fix for the GE CLoud midnight issue: https://github.com/springfall2008/batpred/releases/tag/v8.22.0

TheSargey, 2025-07-03:
> New fix for the GE CLoud midnight issue: https://github.com/springfall2008/batpred/releases/tag/v8.22.0

Thanks as always @springfall2008 - will report back in the morning.

Cheers,

Sarge