springfall2008/batpred #2367: Expand function of Inverter "Set charge window before start"

johnwb87, 2025-05-15:
Problem:
I am increasingly seeing Predbat be very slow in changing status and therefore making inverter changes at rate change thresholds, needing to wait for a full run that may include a plan update as well. 
This can lead to important inverter changes being late, the plan not executing accurately and leading to more cost at the 30 minute boundaries where charge, discharge or hold status changes. 
For example a Hold Charging at the end the an intelligent slot was held until 6 minutes after the planned change to demand whilst predbat was running, the planned change was showing in the html plan with the final 5 minutes of the previous 30 minute block still showing. Through this 6 minutes I was pulling from the grid at peak rate rather than being in the planned demand/eco mode.

Enhancement:
Expand the "Inverter Set charge window before start" function to cover all time window based setting changes, making the inverter settings changes in the previous predbat run.

I've thought about an automation to catch the worst situation which is usually related to car charging. But given it is a speed of action problem I'm not sure what else I can do.

springfall2008, 2025-05-19:
I'm suprised as inverter actions are meant to run before the plan re-calculation. Are you seeing very long runtimes for the plan?

johnwb87, 2025-05-19:
I'd say that is certainly a contributing factor. recent log entries are as follows. Everything is bleeding together.

2025-05-19 13:02:19.196167: Warn: Historical day 7 has 10 minutes of gap in the data, filled from 17.12 kWh to make new average 17.24 kWh (percent 99%)
2025-05-19 13:11:06.605116: Info: record_status Demand
2025-05-19 13:11:29.303312: Warn: Historical day 7 has 10 minutes of gap in the data, filled from 17.03 kWh to make new average 17.15 kWh (percent 99%)
2025-05-19 13:20:25.101179: Info: record_status Demand
2025-05-19 13:20:45.093477: Warn: Historical day 7 has 10 minutes of gap in the data, filled from 17.03 kWh to make new average 17.15 kWh (percent 99%)
2025-05-19 13:29:41.598493: Info: record_status Demand
2025-05-19 13:30:01.514145: Warn: Historical day 7 has 10 minutes of gap in the data, filled from 17.19 kWh to make new average 17.31 kWh (percent 99%)
2025-05-19 13:30:12.283713: Info: record_status Demand
2025-05-19 13:30:34.927737: Warn: Historical day 7 has 10 minutes of gap in the data, filled from 17.24 kWh to make new average 17.36 kWh (percent 99%)

johnwb87, 2025-05-19:
And just for a bit more specificity I restarted predbat. First plan run appears to have taken about 9 minutes and 30 seconds.

2025-05-19 13:44:46.443215: Warn: Historical day 7 has 10 minutes of gap in the data, filled from 17.24 kWh to make new average 17.36 kWh (percent 99%)
2025-05-19 13:54:15.602368: Info: record_status Demand

I'm trying to use the web interface debug but it just seems to hang and not give me a file.

johnwb87, 2025-05-19:
I've tried both. Neither seems to get the instant file.

johnwb87, 2025-05-19:
[predbat_debug_15_10_00.yaml.txt](https://github.com/user-attachments/files/20298661/predbat_debug_15_10_00.yaml.txt)

Whilst this is named as the 15:10 run, it's modified date was 15:27.

I've also done some looking into the history of the predbat active switch. My active times visibly increase from a max of around 6 minutes to 9 minutes on the morning of 28 April. Looking at the releases that coincides with v8.19.1 being released as I have auto update turned on.

cyberkryten, 2025-05-19:
If a 15_10 named file was modified at 15:27 that suggests it took 17 minutes for Predbat to run - that's a major issue!

What's your system spec?  Your original stated six minutes is a long time for Predbat to be running constantly (even with dual pass I'm usually under a minute) - have you checked if there's something else on your HA system burning CPU?

It may be you've hit a 'hump' on memory or storage - sometimes a minor increase starts swapping and things just get slower and slower - have you tried SSH onto the system and using TOP to see what CPU usage is like?

johnwb87, 2025-05-19:
I’m running on a Home Assistant Green. I basically only have Predbat running, a few integrations to pull energy stats in and some lighting automations.

That jump from 6 mins to 9 is just what I can see in the log book history that’s retained.

I did a full system reboot today and still getting around 10 minutes for a predbat run. Seeing memory usage around 50% and predbat ticking away at about 30-40% usage with occasional spikes to 90+.

Any ideas anyone? Anything jumping out from the debug file?

cyberkryten, 2025-05-20:
Ah, I don't know much about the Green, I think your best hope is for someone else on Green to say how long theirs takes.

Benchmarking/Performance evaluation is a whole topic itself, but does it speed up much if you disable Debug?  That will reduce disk writes significantly as the Green has eMMC, so may be worth a try.  I don't think you can get much in terms of disk stats out of HA, so you'll need to do any disk IO stats/checks at the OS level.

It's worth checking the CPU/RAM graphs in Settings->Hardware with Predbat running but also with it disabled just to see if there is anything else taxing the system.  Predbat does a lot of calculations, so is very dependent on system performance - I'd expect to see CPU go to 100% whilst it is running and then fall back to very little <10% the rest of the time if there really isn't much else going on.

If you are stuck with 'slow' then maybe changing Calculate plan every N minutes to a larger value will help, maybe 15/20min so that most Predbat 'runs' are just doing checks rather than a plan recalculation.  That should at least make a big difference to the CPU graphs and give you time between runs to find your real baseline CPU usage.

Mine isn't a good comparison as it's a VM but my CPU usage baseline is 2% and peak to 100% when Predbat runs before dropping back (I have full second-pass enabled as well as a lot of debug).

[image]

johnwb87, 2025-05-20:
Looking at the CPU graphs shows that baseline is around 3% and always under 5% with a predbat run but no plan calculation peaking at 31%
[image]

When a plan is calculated it looks like this. 90% then bumps along mostly around 30% for the around 10 minute run. 

[image]
[image]

So the CPU isn’t being utilised. Memory looks OK. So either disk writes are a problem or something with predbat.

I did a rebuild yesterday and am considering uninstalling and a fresh install.

johnwb87, 2025-05-23:
Around 10 entries a day in the HA Core log. I don't knowingly have any debug turned on.

[predbat (5).log](https://github.com/user-attachments/files/20409252/predbat.5.log)

From this file;

2025-05-23 09:55:29.286263: >> Region optimisation pass width 60
2025-05-23 09:57:25.962213: Calculated end_record as 05-24 23:30:00 based on best_price 7.0 next_charge_start 05-23 23:30:00 max_windows 3

The Region optimisations are what are taking the time, getting steadily slower as they get more detailed, which makes sense. But is that duration normal/average for the HA Green hardware? Is threading working correctly if the CPU isn't being fully utilised?

In that run the coarsest pass starts at 2025-05-23 09:51:36.919997: >> Region optimisation pass width 480
and as above the final pass finishes at 2025-05-23 09:57:25.962213. So 6 minutes for that element within a total run time of 8 minutes 43 seconds.

I have now changed plan runs to 20 minutes and that is making my issue with plan changes being applied less impactful, but can still occur and also predbat is now not very responsive to changes in inputs, such as an IOG slot as I'm discussing in #2368.

cyberkryten, 2025-05-23:
When I said turn off all debug, I meant including Predbat - it writes a LOT of data, so if you have an IO bottleneck then that could be the cause of your issues.  iostat may show what's going on.

For comparison, my grep for those same start/end terms you showed are a lot quicker, but I'm running on a VM host with a Quad Core i5 and NVMe storage.

```
tail -F predbat.log | | grep "PredBat.-.update\|Info..record_status\|Region opt\|Calculated end_record" 
2025-05-23 12:30:00.698255: --------------- PredBat - update at 2025-05-23 12:30:00+01:00 with clock skew 0 minutes, minutes now 750
2025-05-23 12:30:40.320284: Info: record_status Demand
2025-05-23 12:35:00.431159: --------------- PredBat - update at 2025-05-23 12:35:00+01:00 with clock skew 0 minutes, minutes now 755
2025-05-23 12:35:17.804475: >> Region optimisation pass width 960
2025-05-23 12:35:21.859518: >> Region optimisation pass width 480
2025-05-23 12:35:26.312742: >> Region optimisation pass width 240
2025-05-23 12:35:30.452974: >> Region optimisation pass width 120
2025-05-23 12:35:37.172461: >> Region optimisation pass width 60
2025-05-23 12:35:46.798275: Calculated end_record as 05-24 23:30:00 based on best_price 7.0 next_charge_start 05-23 23:30:00 max_windows 70
2025-05-23 12:36:17.084333: Info: record_status Demand
2025-05-23 12:40:00.179973: --------------- PredBat - update at 2025-05-23 12:40:00+01:00 with clock skew 0 minutes, minutes now 760
2025-05-23 12:40:18.553724: Info: record_status Demand
2025-05-23 12:45:00.660115: --------------- PredBat - update at 2025-05-23 12:45:00+01:00 with clock skew 0 minutes, minutes now 765
2025-05-23 12:45:17.589329: >> Region optimisation pass width 960
2025-05-23 12:45:21.688109: >> Region optimisation pass width 480
2025-05-23 12:45:26.160907: >> Region optimisation pass width 240
2025-05-23 12:45:30.262053: >> Region optimisation pass width 120
2025-05-23 12:45:37.012849: >> Region optimisation pass width 60
2025-05-23 12:45:46.493530: Calculated end_record as 05-24 23:30:00 based on best_price 7.0 next_charge_start 05-23 23:30:00 max_windows 70
2025-05-23 12:46:16.424851: Info: record_status Demand

```

johnwb87, 2025-05-23:
Is that your full predbat log? I have a lot more than that going on and I do not have debug turned on.

cyberkryten, 2025-05-23:
No, it's the output of the Grep for the relevant entires.  I've got full Predbat debug on which generates about 30M/hour, so it will probably be faster still without all that.

I don't know how to turn off logging totally, or change it to minimal in Predbat/HA - if I have issues that I think are logging related, I usually just build /var/log to a Ramdisk or link to /dev/null !!

Not much else I can do to help - unless someone comes along with a Green comparison that is way faster, I think you've just hit the max performance of the device.