springfall2008/batpred #2368: Immediately change battery mode if watch_list: - '{octopus_intelligent_slot}' change detected.

johnwb87, 2025-05-15:
Problem:
Predbat is not very robust to a late change to an IOG slot. Often resulting in significant periods of battery discharge whilst the plan is fully recalculated..

Enhancement:
If a IOG slot changes at short notice then change battery mode as a holding position before recalculating the entire plan. Change to PauseDischarge if a slot arrives and Eco if a slot disappears. I think these are the lowest risk options to preserve battery charge and minimise grid usage.

Alternative:
I've considered an automation that would check Predbat status when a slot in the current 30 minutes changes and then make the change in GivTCP to bridge the gap to Predbat reacting.

pacemaker82, 2025-05-22:
I think it could be even more "intelligent" without re calculating the plan. 

I think before any recalculation, predbat should respond to two events and do _SOMETHING_ 

1) Car charging entity changed - do something
2) IOG dispatches changed - do something

That "something" should be related to telling the battery to charge/discharge almost instantly based on the state change and whatever logic makes most sense depending on 1 or 2 happening. 

I also think (2) should trigger a recalculation after its done "something", rather than waiting for the 5 min or whatever timeframe to elapse.

pacemaker82, 2025-05-22:
> So I believe this bit is already in place

Thanks - I had no idea there was a watch list function - my apps.yaml doesn't mention it, so I suppose it was added somewhere along the lines in a release notes. Will give it a whirl.

gcoan, 2025-05-22:
> Thanks - I had no idea there was a [watch list function](https://springfall2008.github.io/batpred/apps-yaml/#watch-list-automatically-start-predbat-execution) - my apps.yaml doesn't mention it, so I [...]

watch list isn't in the default template but I have added it to the documentation https://springfall2008.github.io/batpred/apps-yaml/#watch-list-automatically-start-predbat-execution

springfall2008, 2026-10-28:
The watch list already triggers a new plan; the 10-minute plan run behind the request (#2367) is gone.

closed 2026-10-28 by springfall2008