springfall2008/batpred #2380: Recent Predbat versions planning export when car charging

sammort, 2025-05-20:
**Describe the bug**
In recent versions of Predbat (from around 8.19), the plan for my system is both not following configuration (planning to export during car charge even though `switch.predbat_car_charging_from_battery` is `false`) and sub-optimal (exporting earlier than expected, instead of waiting until rate change).

**Expected behaviour**
Plan should follow explicit settings 
- not planning to export during car charging periods if `switch.predbat_car_charging_from_battery` is `false`
- holding export until late as possible if `switch.calculate_export_high_import` if `true`

**Predbat version**

v8.19.14

**Environment details**

- AIO
- Docker
- Intelligent Octopus Go

**Screenshots**

[image]

**Log file**
I haven't captured a log file that goes with this plan and debug. But let me know if that would help

**Predbat debug yaml file**
[predbat_debug.yaml.txt](https://github.com/user-attachments/files/20357154/predbat_debug.yaml.txt)

pacemaker82, 2025-05-28:
+1 on this - I am getting the half hour charge/export when the car will charge for the entire 30 mins.

[image]

pacemaker82, 2025-05-29:
I think it's a plan thing, because Predbat's logic should "hold for car" when the binary sensor is ON. 

Looking through my data I see yesterday "some" evidence of having charge and export within the same 30 min windows, but in reality the car could be turned on and off for 10 mins and export for 20 mins.

[image]

pacemaker82, 2025-05-29:
Just looking more broadly over past 72 hours, it seems to confirm a plan thing not a "reality" thing. 

My car can charge max ~1.2 kWh, so I would expect to see if both export and car charge was happening that the total kWh would be higher than that. I tried looking at the battery discharge data but its not very reliable and causes the charts to go haywire

[image]

sammort, 2025-05-29:
I can also confirm that this is only an issue in the future plan and not what is actualy executed by Predbat.

However, I would say the value of the plan itself is almost zero if it doesn't actually represent a realistic version of planned events. These aren't plans that are 24-48 hours away, they are within the next 2 to 3 hours. I'd also expect the plan to respect explicit settings such as car charging not being allowed from the battery.