springfall2008/batpred #2381: Car kWh in plan doesn't make sense (IOG)

pacemaker82, 2025-05-21:
**Describe the bug**
In IOG, I have a planned dispatch from 15.29 - 17.00 with a total of 3.56 kWh into the car.

[image]

The car is on a slow charger, so maximum throughput is 2.45 kWh - makes sense and reflects the IOG data (2.45 x 1.5 hours is ~3.6 kWh vs 3.56 kWh reported by the dispatch.

However, for the current 30 min slot, the kWh in the car column is consistently wrong. Right now its reporting 4.06 kWh within the slot, which is physically impossible, nor aligned at all with what the dispatch is saying it will provide. At most, a 30 min slot can only ever transfer 1.22 kWh, this slot only has 25 mins remaining.

[image]

It appears some over-calculation is happening.

**Expected behaviour**
Current slot should reflect the actual energy transfer to car within that time slot's remaining time. 

**Predbat version**

8.19.14 but has happened forever.

Timeflies1980, 2025-05-21:
Do you have the input_number.predbat_car_charging_rate setup correctly?

pacemaker82, 2025-05-21:
> Do you have the input_number.predbat_car_charging_rate setup correctly?

According to docs that’s for predbat led charging not IOG https://springfall2008.github.io/batpred/car-charging/

pacemaker82, 2025-05-22:
@springfall2008 

The strange thing is that the plan reports the **_future_** 30 min slots correctly based on the dispatch, just not the CURRENT slot. Example this morning showing the dispatch 7.00-7.30 window as a 1.03 kWh charge, however plan shows 1.58 kWh in the remaining 10 mins of that slot (7.20-7.30): 

[image]

[image]

As I have set 
```
  car_charging_energy:
    - sensor.shellyem_c45bbee1fdee_channel_1_energy
```

I assume you should be able to figure out how much went in, and how much is remaining in the current slot.

pacemaker82, 2025-05-22:
Just a note, that now that **5** mins has passed, the current car KWh has **halved** from 1.58 kWh remaining in the slot, to 0.79 kWh remaining in the slot. I only actually charged about 0.204 kWh within the last 5 mins. 

[image]

gcoan, 2025-05-22:
> Just a note, that now that 5 mins has passed, the current car KWh has halved from 1.58 kWh remaining in the slot, to 0.79 kWh remaining in the slot

On the 5 minute mark Predbat doesn't recalculate the plan, only on the 10 minute mark (by default).  So would expect a reduction for 'charge still remaining' but (predbat led charging) the full recalc based on 'charge currently applied' would only happen on the 10 minute mark

pacemaker82, 2025-05-22:
> On the 5 minute mark Predbat doesn't recalculate the plan, only on the 10 minute mark (by default). So would expect a reduction for 'charge still remaining' but (predbat led charging) the full [...]

Good to know ... either way - the remaining charge in the slot is wrong :)

gcoan, 2025-05-22:
> Good to know ... either way - the remaining charge in the slot is wrong :)

Don't disagree with that point!  I can't explain the calculation either

pacemaker82, 2025-05-26:
Small update on this @springfall2008 - the current slot kWh remaining is **only incorrect** when the **car is actively charging**

I noticed this morning in the 7.30-8.00 slot (the current slot) that it was showing the correct amount for the slot until the car actually started, at which point the value was ~4x what it should be.

springfall2008, 2025-06-07:
I'm struggling with this one without any logfile or debug yaml, if you can post both when it happens I'm sure I can find the issue

pacemaker82, 2025-06-07:
YAML:
[predbat_debug.yaml.txt](https://github.com/user-attachments/files/20639338/predbat_debug.yaml.txt)

LOG:
[predbat_car_log.txt](https://github.com/user-attachments/files/20639341/predbat_car_log.txt)

Examples - two log entries for the 7.30 slot, only the first one is actually correct:

```
2025-06-07 07:30:10.701099: Octopus Intelligent slot at 06-07 07:30:00-06-07 08:00:00 assumed price 7.0 amount 1.33 kWh location None source smart-charge octopus_slot_low_rate True
2025-06-07 07:30:10.701460: Octopus Intelligent slot at 06-07 07:30:00-06-07 08:00:00 assumed price 7.0 amount 3.7 kWh location  source  octopus_slot_low_rate True
```

pacemaker82, 2025-06-08:
Attaching more logs (based on Predbat v 8.20.12)

At 0700 today

[predbat_log_0700.txt](https://github.com/user-attachments/files/20643287/predbat_log_0700.txt)

Plan:

[image]

At 0715 today

[predbat_log_0715.txt](https://github.com/user-attachments/files/20643289/predbat_log_0715.txt)

Plan:

[image]

springfall2008, 2025-06-08:
The debug yaml you provided doesn't show the issue, can you provide a debug yaml from when the HTML report is incorrect?

pacemaker82, 2025-06-12:
attaching the debug yaml while it is happening

[predbat_debug.yaml.txt](https://github.com/user-attachments/files/20717380/predbat_debug.yaml.txt)