springfall2008/batpred #2382: Random mid-day exports on Octopus Go

nbullus, 2025-05-21:
**Describe the bug**
The plan now seems to randomly export battery at various points of the day, including in the middle of Export Freeze periods. This increases the risk of running out of battery should solar be lower than forecast and/or load is higher than forecast.

Today was a good example where it exported twice in the day (22% SOC) within the middle of an Export Freeze period. PV ended up being less than forecast and my load will increase as we normally go out on Wed but decided to stay in and cook. My battery will likely run out and I will be forced to pull from the grid at peak rate.

[image]

Last occurrence randomly exported at 17:55 which was captured in the attached debug log.

[image]

**Expected behaviour**
On Octopus Go there is no cost benefit to export battery early. Better to wait until the end of the peak period in preparation for the cheap off-peak charge period. The earlier you start to export (and the number of time you repeat this) the higher the risk of running out of battery later on.

**Predbat version**

8.19.14

**Environment details**

- GE AIO
- Standard HAOS installer 

**Screenshots**

See description

**Log file**

Debug attached below.

**Predbat debug yaml file**

[predbat_debug_17_50_00.yaml.txt](https://github.com/user-attachments/files/20372763/predbat_debug_17_50_00.yaml.txt)

nbullus, 2025-05-23:
Is there any short term workaround I could use to minimise mid-day exports as it seems to be happening regularly now that I am monitoring it more?

gcoan, 2025-05-23:
> Is there any short term workaround I could use to minimise mid-day exports as it seems to be happening regularly now that I am monitoring it more?

You could put a rate adjustment in apps.yaml for the daytime of say -3p and this would encourage Predbat to not export during the day.  Would mean the battery is more likely to be filled up rather than freeze export though

Alternatively run an automation to put a freeze export manual override during the daytime,  or an automation to set control mode to Charge only during the day and Charge and Discharge for the evening.

nbullus, 2025-05-23:
> You could put a rate adjustment in apps.yaml for the daytime of say -3p and this would encourage Predbat to not export during the day. Would mean the battery is more likely to be filled up rather [...]

Thanks Geoff. I'll have a play with automating the Freeze Export manual override so I can keep the Freeze Export going.

johnwb87, 2025-05-23:
Or you could use the rate override via the manual API and only set a load scaling so that on days and for windows that  you know load is going to change you can scale predbat's load forecast. That way you are fixing the input deficiency that is leading to the plan allowing the battery to run out.

mpartington, 2025-05-27:
The latest version on main, discharge should all be at the end of the day

[predbat_debug-8.yaml.txt](https://github.com/user-attachments/files/20455824/predbat_debug-8.yaml.txt)

nbullus, 2025-06-01:
Upgraded to v8.20.3 but still getting random mid-day exports. Some are planned, but others just happen on the spur on the moment. Today it exported at 06:50 which is just over an hour after it charged the battery in the off-peak period. Will attach debug file when next occurs.

[image]

springfall2008, 2025-06-07:
https://github.com/springfall2008/batpred/releases/tag/v8.20.11

mpartington, 2025-06-07:
Great to see a fix, but curious as to why later export should make a plan worse for us on AC coupled systems, where there is no clipping. I would have thought no impact on plan cost, but just increased risk of running out if exporting early

closed 2026-08-20 by chalfontchubby