springfall2008/batpred #2383: Predbat repeatedly sets the discharge slot end time, increasing register writes

gcoan, 2025-05-21:
**Describe the bug**
Predbat is set to force discharge to empty the batteries via an automation just before sunrise that sets predbat.select_manual_force_export to "05:00:00,05:30:00,06:00:00,06:30:00,07:00:00,07:30:00,08:00:00"

I would expect Predbat to set a single long forced discharge but from the givtcp log I can see it set each slot in turn every half hour, which isn't optimal for inverter register writes:

```
2025-05-21 05:00:15,279 - G - write       -  [INFO    ] - Setting Discharge Slot 1 to: 05:00 - 05:31 was a success
2025-05-21 05:00:16,028 - G - write       -  [INFO    ] - Setting Timed Export mode was a success
2025-05-21 05:00:27,092 - H - write       -  [INFO    ] - Setting Discharge Slot 1 to: 05:00 - 05:31 was a success
2025-05-21 05:00:27,821 - H - write       -  [INFO    ] - Setting Timed Export mode was a success
2025-05-21 05:30:15,178 - G - write       -  [INFO    ] - Setting Discharge Slot 1 to: 05:00 - 06:01 was a success
2025-05-21 05:30:26,102 - H - write       -  [INFO    ] - Setting Discharge Slot 1 to: 05:00 - 06:01 was a success
2025-05-21 05:30:28,680 - H - read        -  [INFO    ] - Saving current energy stats at start of day rate tariff (Dynamic)
2025-05-21 05:30:28,803 - G - read        -  [INFO    ] - Saving current energy stats at start of day rate tariff (Dynamic)
2025-05-21 06:00:15,673 - G - write       -  [INFO    ] - Setting Discharge Slot 1 to: 05:00 - 06:31 was a success
2025-05-21 06:00:26,238 - H - write       -  [INFO    ] - Setting Discharge Slot 1 to: 05:00 - 06:31 was a success
2025-05-21 06:30:15,030 - G - write       -  [INFO    ] - Setting Discharge Slot 1 to: 05:00 - 07:01 was a success
2025-05-21 06:30:26,006 - H - write       -  [INFO    ] - Setting Discharge Slot 1 to: 05:00 - 07:01 was a success
2025-05-21 07:00:14,670 - G - write       -  [INFO    ] - Setting Discharge Slot 1 to: 05:00 - 07:31 was a success
2025-05-21 07:00:26,269 - H - write       -  [INFO    ] - Setting Discharge Slot 1 to: 05:00 - 07:31 was a success
2025-05-21 07:25:05,397 - G - write       -  [INFO    ] - Setting Eco mode was a success
2025-05-21 07:25:16,612 - H - write       -  [INFO    ] - Setting Eco mode was a success
2025-05-21 07:30:15,125 - G - write       -  [INFO    ] - Setting Discharge Slot 1 to: 05:00 - 08:01 was a success
2025-05-21 07:30:15,864 - G - write       -  [INFO    ] - Setting Timed Export mode was a success
2025-05-21 07:30:26,912 - H - write       -  [INFO    ] - Setting Discharge Slot 1 to: 05:00 - 08:01 was a success
2025-05-21 07:30:27,650 - H - write       -  [INFO    ] - Setting Timed Export mode was a success
2025-05-21 08:00:15,707 - G - write       -  [INFO    ] - Setting Discharge Slot 1 to: 05:00 - 08:31 was a success
2025-05-21 08:00:26,271 - H - write       -  [INFO    ] - Setting Discharge Slot 1 to: 05:00 - 08:31 was a success
2025-05-21 08:34:14,050 - G - write       -  [INFO    ] - Setting Eco mode was a success
```

**Expected behaviour**
A single forced discharge slot from 05:00 to 08:00 (or 08:01)

**Predbat version**
8.19.14

**Environment details**
2x Gen 1 hybrid inverters
HAOS 2025.5.1

**Log file**

[predbat.4.log](https://github.com/user-attachments/files/20375186/predbat.4.log)
[predbat.5.log](https://github.com/user-attachments/files/20375185/predbat.5.log)

**Predbat debug yaml file**
[predbat_debug.yaml.txt](https://github.com/user-attachments/files/20375196/predbat_debug.yaml.txt)

cyberkryten, 2025-05-22:
Different Inverter (SolarEdge) but I've noticed that Predbat seems to reset it to Maximise Self Consumption mode before setting it to Discharge to Maximise Export - is that what your Eco mode is as having those at 07:25 seems odd.,  

I can't see the service setup entries (maybe different) but this could be related to the 'always' flag that was added to the discharge_start_service?

The other thought is maybe there's no way to read out the start/end times that are already set (or that Predbat doesn't try to do that) and so it sets them again each time.  

From your log, it looks like on most of those writes, the end time is being increased but the start is staying the same, so even at 08:00 it is setting 05:00-08:31 which seems the wrong way around, maybe the code is incrementing the wrong entry?

gcoan, 2025-05-22:
> I can't see the service setup entries (maybe different) but this could be related to the 'always' flag that was added to the discharge_start_service?
>
> The other thought is maybe there's no way to [...]

The GivTCP inverter logic is built into predbat, its not via service start definitions in apps.yaml, so it might be due to the new 'always' logic or it might not

Predbat is semi doing it right, its reading the start and end times (which are separate HA entities), realising the next discharge slot is an extension of the current one, so its only setting the end time, so we see the pattern in the givtcp log.
I think the problem may be in the way the series of force export commands is processed, I would have thought Predbat overwrites these into the plan, then processes the plan as a whole, setting a single long discharge slot,  but it appears to be processing each force discharge at the start of the half hour slot, so hence the pattern of time changes I see.

And as for the extra Eco, no idea, when it's in force export for several hours that shouldn't be there either!

springfall2008, 2026-10-28:
Manual windows are deliberately not merged (plan.py:3931-3932).

closed 2026-10-28 by springfall2008