springfall2008/batpred #2385: Calibration curves

interceptador, 2025-05-23:
Hi 

I have recently installed predate am trying to make sense of it all, should I be concerned that the calibration curves are as wild as they are, if so is there any improvement I can make to change them for the better. 

[image]

gcoan, 2025-05-23:
your load calibration curve (top right) looks to be quite a long way off

here's mine for comparison 

[image]

you've not provided any details of your setup so hard to provide detailed advice, but I would start by looking at your losses (battery/inverter) and your load sensor, does it include all load, have you got EV charging that could be excluded, have you included enough days_previous to give a reasonable average load to predbat

interceptador, 2025-05-23:
Thank you for your reply. This is my setup. 
Giv Energy gen 3 3.6 inverter
Giv Energy 9.5 gen 2 battery
8 430kw West facing panels
6 430kw East facing panels
Giv Energy Car charger

I have used 7 days as the history, I don't have any more stored yet than that as I have not figured out how to keep it longer than the HA 10 days. 

I have kept the load sensor pointing to the default one the givenergy inverter in apps.yaml

I have pointed the EV charger to car_charging_energy: re:(sensor.givevc_{geserial}_charge_session_energy) which looks to be an incremental load for the day. 

I have used 0.06 as the battery losses charge and discharge ( as the inverter data sheet seems to imply 94%) and 0.03 as the inverter loss. 

Interestingly today however the load curve seems complete nonsense. but the giv energy load today is ok. see attached screen shots. But also the car charging has not been subtracted from the predicted load. 

Really scratching my head over this one. So would really appreciate any help that can be offered. 

[image]

[image]

gcoan, 2025-05-23:
Your setup sounds OK and your losses are reasonable.

If you want to increase the amount of history retained in HA, you'll need to set purge_keep_days in configuration.yaml to a value more than 10, details in https://springfall2008.github.io/batpred/apps-yaml/#days_previous

If you have your days_previous just set to the value 7 then this will take the same day's load from a week ago.  Obviously depends on your home usage pattern but averaging over more days can usually give a better result as it reduces the impact of spikes such a doing a load of washing one day. https://springfall2008.github.io/batpred/apps-yaml/#understanding-how-days_previous-works

I personally have mine set to average days 2 through to 8.

If your car charging isn't being eliminated from the load then something is usually wrong with the car setup in predbat https://springfall2008.github.io/batpred/car-charging/ - double check you have followed all the steps and set the sensors up correctly.

If you can attach a copy of your apps.yaml (you will have to rename it apps.txt) then I can see if there is anything obviously wrong with it

interceptador, 2025-05-26:
I am having a look at your suggestions today. But here is the apps.yaml file. Thank you for you assistance. 

[appsyamlpredbat.txt](https://github.com/user-attachments/files/20439822/appsyamlpredbat.txt)

gcoan, 2025-05-26:
> I am having a look at your suggestions today. But here is the apps.yaml file. Thank you for you assistance.
> 
> [appsyamlpredbat.txt](https://github.com/user-attachments/files/20439822/appsyamlpredbat.txt)

I don't see anything obviously wrong with your apps.yaml, it looks to be configured correctly.  Unfortunately details of setting predbat up for the givenergy EV charger are not in the predbat documentation https://springfall2008.github.io/batpred/devices/ so I can't check the sensors you are using are correct.

One query, the car_charging_energy: which is set to sensor.givevc_11288853517737_charge_session_energy (you could by the way just set it to 'sensor.givevc_11288853517737_charge_session_energy', the 're' regular expression isn't doing anything).  Is this sensor in kWh or Wh?  Needs to be in kWh.

If you look at the load column in your predbat plan, does it look sensible and match what you see on your givtcp load_energy_today_kwh sensor?

interceptador, 2025-05-30:
Thank you for taking the time to look at that. Yes it's definitely a kWh sensor. I think predbat is setup correctly but the data iits getting is bad data from givtcp, the data over the last few weeks looks questionable. for example over the last 2 days the charger had just read a constant 2kWh. I have reset givtcp. Hopefully that may fix it. I will keep digging and report if I get a fix.

gcoan, 2025-05-30:
If you're getting bad data from the EVC then this is going to be a problem. 

Givtcp just reads what the inverter and EVC report so it should just work.  If you look in the givtcp logs does it indicate any errors when reading the data?
I would try restarting the EVC and inverter as well.

If the kwh sensor is bad, is the instantaneous EV charging power ok? It's possible with a Reimann integration helper to convert power to energy - as long as the power sensor is ok

charliejvs, 2025-06-04:
So update. 

Checked the GivTCP logs, can't see any errors reading data. I did a reinstall of predbat and GivTCP and have done some restarts of the actual devices as well. 

After all this I think I have got down to the bottom of it. I am on the EON smart drive V2 tariff. This works in a similar way to the octopus intelligent tarrif. I think this was one of the things throwing out all the calculations. The data I was using from the GivEVC was pointing to Energy sensor which resets after each charging period. Because the charging periods were not one whole period. i.e. multiple micro periods. It reset to 0 each time resulting in incorrect figures. So I have now made a utility meter that adds up the daily energy usage as well as a Reimann integral on the power sensor and I will over the next few weeks look which is best. 

This tariff also has another problem. It sometimes charges outside the off-peak periods.  I have now pointed the car_charging_now: term to a sensor that knows the charging status of charger. I think this may be a work around to this in the car charging plan to make sure the battery doesn't discharge to the car. Would be interested if this is something sensible to do? 

Long and short I think the predictions are now more sensible. I have attached some pictures. Although I do get lots of warnings so say 
Warn: Historical day 4 has 205 minutes of gap in the data, filled from 8.62 kWh to make new average 10.05 kWh (percent 86%) I see this has been mentioned on previous threads as being "normal" 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

closed 2026-08-20 by chalfontchubby