springfall2008/batpred #2386: Integrate Predheat to use Open Energy Monitor sensors?

zarch1972, 2025-05-23:
**Is your feature request related to a problem? Please describe.**

I already use Open Energy Monitor to monitor my heat pump with an electricity meter and heat meter.

https://emoncms.org/energystatsuk

Along with 100's of others at https://heatpumpmonitor.org/

I have all the sensors in play then pulled into Home Assistant using the emoncms integration

Electric in (instant and day)
heat out (instant and day)
Both of above split by DHW and heat only 
Outdoor temp
Indoor temp
Flow Temp
Return Temp
Flow Rate
COP / Instant COP
etc

**Describe the solution you'd like**

So rather than making calculations to predict forthcoming electricity (or heat) required, could Predheat use these actual figures?  To give solid predictions based on recent fact?

Obviously, if you can't split your usage down into separate heating / DHW usage numbers, then some accommodation for that would be required.

Note: not everyone using Open Energy Monitor has separate heating and hot water values as it can require and extra bit of kit to monitor the 3 way valve (whether it's doing heating or hot water)

I would be happy to help with any trials / implementation

Cheers, Mick

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

[image]

johnwb87, 2025-05-24:
+1 on this. I’m expecting to have OEM on a heat pump by the end of the year and want to use all the data to drive predheat.

zarch1972, 2025-08-05:
In the predheat config I've put in the internal room and outside temps as well as internal target temp from the Vaillant heat pump controller.

I also put the heat output from the heat pump into the 'heating_energy' 
put flow temp in and and some other stuff.

But cos there's no heat at the moment I have no idea what predheat is doing.  And because I've only just starting setting up in the summer I have no reference to what it can do in the winter.

Also, because i'm doing two hot water runs a day and the flow temp / heat output is used in those two, who knows.
Hence it would be useful to have a heating / hot water binary sensor perhaps?

I think predheat using heat pump actuals from open energy mon needs some thought. :-)

zarch1972, 2025-10-15:
ignore me, I found predheat.heat_energy$external, which i'd not added in as a load_forecast. RTFM.
Although this is probably 'heat only', so I need to find a way to add daily hot water usage onto the total.

johnwb87, 2025-10-15:
Yes, I think it is predheat.heat_energy. I think the hot water can be predicted fairly well by predbat with the one assumption being that your scheduling is consistent. For example when I have been using immersion heater overnight because it is a consistent schedule predbat sees that in the historic data and so it can forecast it.

Do you have any of the Predheat charts set up?

My own OEM set up is just days away now so I will be actively playing with this myself very soon.

zarch1972, 2025-10-15:
I need to do some digging around graphs and get my head around all this now i've found this predheat.heat_energy

Our hot water usage through the heat pump is quite consistent through the year, so I could easily just add 5kWh electricity each day (although half of that usually overnight in the off peak period).

This also piqued my interest: https://github.com/springfall2008/batpred/discussions/837

johnwb87, 2025-10-15:
> I need to do some digging around graphs and get my head around all this now i've found this predheat.heat_energy
>
> Our hot water usage through the heat pump is quite consistent through the year, so I [...]

Yes, OK I agree something to think about and look into. I was hoping to work with @springfall2008 once my system is up and running to, as a minimum, improve the documentation. For example all heat pump installs should have a very good heat loss calculation, so how can we use that.

johnwb87, 2025-10-16:
@zarch1972 oh wow they’re your articles, I’ve read many of them already! I will have a 5kW Vaillant and am planning on implementing “sweet spot” mode for heating a 250l tank.

zarch1972, 2025-10-16:
Infamy! Infamy! They've all got it in for me! LOL

Email me via the site if you want any help, i'd be happy to go through stuff I've learnt. 

I'm currently writing an article about 'how much electricity does a heat pump use', which is very much tied to what we are talking about here.  So many variables!!!!

closed 2026-10-28 by springfall2008