springfall2008/batpred #2394: Predbat execution time has increased significantly in recent versions (i.e. its slower)

gcoan, 2025-05-26:
**Describe the bug**
In recent versions of Predbat it seems to be taking a lot longer to evaluate the plan than it used to be, and its probably much much worse on complex tariffs like Agile.  Anecdotally Predbat used to take a couple of minutes to do a full plan evaluation but from the predbat compare timing I can see its taking sometimes up to 12 minutes:

[image]

The compare has been taking more than 20 minutes for the last couple of nights, causing the predbat error monitor to think that predbat has stalled execution.

Can see this in normal runs, 18:05 update took 5 minutes, 18:15 took 8 minutes, 18:25 took nearly 6 minutes:

[image]

Downgrading to 8.19.9, execution time is not much better:

[image]

Likewise 8.19.6:

[image]

Other people are reporting similar slow behaviour #2367 #2368 https://github.com/springfall2008/batpred/issues/2384#issuecomment-2910145251 https://github.com/springfall2008/batpred/issues/2384#issuecomment-2910245420

As suggested in 2367, problem seems to be in the region optimisation passes from 240 downwards:

```
2025-05-26 18:25:00.530944: --------------- PredBat - update at 2025-05-26 18:25:00+01:00
...
025-05-26 18:25:17.561659: >> Region optimisation pass width 960
...
2025-05-26 18:26:42.557193: >> Region optimisation pass width 240
2025-05-26 18:27:37.116339: >> Region optimisation pass width 120
2025-05-26 18:28:59.592035: >> Region optimisation pass width 60
2025-05-26 18:29:05.998759: Optimise all charge found best buy/sell price band 13.6 best price at cost -1925.594 metric -1858.7075 keep 0 cycle 97.2994 carbon 0 import 
...
2025-05-26 18:30:46.486726: Swap export optimisation finished metric -1156.45 cost -1176.81 metric_keep 0.06 cycle 52.47 carbon 0 import 44.03
...
2025-05-26 18:30:46.573474: Previous plan best metric is -1163.92 (cost -1184.76) and new plan best metric is -1157.72 (cost -1178.79)
2025-05-26 18:30:46.573679: New plan metric is not significantly better (metric_min_improvement_plan 2.0) than previous plan, using previous plan
2025-05-26 18:30:46.574368: Plan valid is now true after recompute was False
...
2025-05-26 18:30:49.188526: Completed run status Demand
2025-05-26 18:30:49.207191: Info: record_status Demand

**Expected behaviour**
Execution times more in the 1-2 minute range that they used to be

**Predbat version**
8.19.14

**Environment details**
HAOS 2025.5.2

**Log file**

[predbat.1.log](https://github.com/user-attachments/files/20444735/predbat.1.log)

**Predbat debug yaml file**

[predbat_debug.yaml.txt](https://github.com/user-attachments/files/20444749/predbat_debug.yaml.txt)

pacemaker82, 2025-05-27:
+1 on this issue too. I noticed yesterday that IOG plan caused a 15min delay to the new plan rendering. Looking at Predbat Active status at the time, it seemed to try to calculate (+5 min duration) turned off for 1 second, then ran again (+5min duration), 1 second off, ran again for +5min - rendering a new plan after 15 mins. 

[image]

Plan used to take around 2 mins (on HA Green device)

pacemaker82, 2025-05-27:
I dont see a significant timing issue between the region optimisations, but I do see a +2m wait after width 60 before it finally completes JUST before the next run at 10:55. (HA GREEN, IOG)

```
2025-05-27 10:50:27.104137: Optimise all windows, total charge 96 export 96

2025-05-27 10:50:32.687743: >> Region optimisation pass width 960
...
2025-05-27 10:50:55.976908: >> Region optimisation pass width 480
...
2025-05-27 10:51:20.290722: >> Region optimisation pass width 240
...
2025-05-27 10:51:56.653752: >> Region optimisation pass width 120
...
2025-05-27 10:52:36.885232: >> Region optimisation pass width 60
...
2025-05-27 10:54:50.958249: Completed run status Demand
2025-05-27 10:54:50.990422: Info: record_status Demand
2025-05-27 10:54:51.022429: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json
```

pacemaker82, 2025-05-27:
I do see significant time loss here:

sheer volume of these (200+), sometimes only 3 per sec

```
Optimise export window ...
```

Seems a big lag/wait between these final steps too:

```
2025-05-27 10:54:07.223831: Optimise price set 31.5 pass low price 29.37 ...
2025-05-27 10:54:11.687670: Swap export optimisation started metric -359.84 cost -398.92 battery_value 2.91
2025-05-27 10:54:26.690454: Swap export optimisation started metric -359.88 cost -399.04 battery_value 2.86
2025-05-27 10:54:41.396048: Swap export optimisation finished metric -359.88 cost -399.04 metric_keep 0.15 cycle 35.61 carbon 0 import 28.42
```

pacemaker82, 2025-05-28:
some positive signs with 8.20.0 on main @springfall2008 @gcoan 

I am seeing times reduced from around 5m30s to do a plan, to now consistently 4m10s, so the 5 and 10 min plans are actually no longer "clashing/overlapping" like they did on 8.19.x

Red = 8.20.x Blue = 8.19.x

[image]

Be interested to see what others are seeing

gcoan, 2025-05-28:
Thanks @pacemaker82 on a sample size of 2 runs, 8.20.0 has done a plan update in 3.5 minutes twice, versus 5-6 minutes for the three preceding runs on 8.19.14 so 👍 from me @springfall2008 

[image]

johnwb87, 2025-05-29:
Are you guys on main to get that?

pacemaker82, 2025-05-29:
yes just select main from the predbat option dropdown

[image]

johnwb87, 2025-05-29:
Worse for me. I was getting consistent 9 minute runs. The last 90 minutes since going to main they’ve all been 12 minutes. So now I’m missing 2 normal runs each time the plan calculates.

Again I have carbon metric on.

johnwb87, 2025-05-29:
8.20

2:53 with Carbon metric off. 10:24 with carbon metric on.

I was watching the processor utilisation once on 8.20 and it was running at much higher utilisation throughout with carbon metric off.

Conclusion then is that some aspect of the carbon metric calculation creates a bottleneck that leads to less CPU utilisation and slower run times.

johnwb87, 2025-05-29:
Watching the log now also.

The "Current best metric" print outs are taking around 5 seconds each with carbon metric on. Compared to 0.5 seconds with carbon metric off.

WyndStryke, 2025-05-30:
I'm seeing the same.  2 minutes with carbon turned off, 5 minutes with carbon turned on.    This is on a Pi 4.

I'll be interested to see how long the overnight tariff comparison takes to run tonight.  Previously it was taking 1 hour 44 min to do the full list of tariffs.

springfall2008, 2025-05-31:
Can you try the latest release as there should be some speedup?

gcoan, 2025-05-31:
> Can you try the latest release as there should be some speedup?

Not noticeably faster for me, if anything slower

8.20.0 (main pre-release) was getting more typically 3-minutes but some outliers 

[image]

8.20.1 (latest release) getting 7/8 minutes

[image]

pacemaker82, 2025-05-31:
Seeing over 5 min runs on 8.20.2 but was getting 4.30 ish on 8.20.0.

gcoan, 2025-05-31:
I'm on 8.20.2 now and will go to .3 when predbat realises its out there, but looking at the logfile it is still taking a long time (5 minutes) in the region optimisation 120 and 240:

```
2025-05-31 22:00:54.402976: Optimise all charge found best buy/sell price band 8.2 best price at cost -2364.8641 metric -2268.2825 keep 0 cycle 89.7146 carbon 0 import 29.76 cost -2364.8641 battery_value 6.7716 limits [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 0, 13.696, 0, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.45, 13.45, 0, 13.45, 0, 0, 0, 0, 0, 13.696, 13.696, 0, 0, 0, 0, 0, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0] export 169.0283
2025-05-31 22:00:54.403053: Optimise all charge found best buy/sell price band 8.2 best price at cost -2366.9429 metric -2270.3925 keep 0 cycle 89.7146 carbon 0 import 27.84 cost -2366.9429 battery_value 6.7716 limits [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 0, 13.696, 0, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.45, 13.45, 0, 13.45, 0, 0, 0, 0, 0, 13.696, 13.696, 0, 0, 0, 0, 0, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0] export 169.0283
2025-05-31 22:00:59.195470: >> Region optimisation pass width 480
2025-05-31 22:01:12.492066: Optimise all charge found best buy/sell price band 14.8 best price at cost -2387.2092 metric -2285.9526 keep 0 cycle 85.5342 carbon 0 import 27.65 cost -2387.2092 battery_value 6.7716 limits [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 0, 13.696, 0, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.45, 13.45, 0, 13.45, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 0.55, 0.55, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0] export 172.3484
2025-05-31 22:01:12.513882: Optimise all charge found best buy/sell price band 14.8 best price at cost -2387.2092 metric -2285.9579 keep 0 cycle 85.5342 carbon 0 import 27.65 cost -2387.2092 battery_value 6.7716 limits [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 0, 13.696, 0, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.45, 13.45, 0, 13.45, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0] export 172.3484
2025-05-31 22:01:23.442621: Optimise all charge found best buy/sell price band 13.4 best price at cost -2390.4431 metric -2293.5842 keep 0 cycle 88.4549 carbon 0 import 28.93 cost -2390.4431 battery_value 6.7716 limits [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 0, 13.696, 0, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 0, 13.696, 0, 0, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0] export 173.5085
2025-05-31 22:01:29.303961: Optimise all charge found best buy/sell price band 13.4 best price at cost -2396.8665 metric -2300.3615 keep 0 cycle 88.4549 carbon 0 import 26.55 cost -2396.8665 battery_value 6.7716 limits [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 0, 13.696, 0, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 0, 13.696, 0, 0, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0] export 174.2537
2025-05-31 22:01:29.304061: Optimise all charge found best buy/sell price band 13.4 best price at cost -2399.6769 metric -2303.2232 keep 0 cycle 88.4549 carbon 0 import 24.94 cost -2399.6769 battery_value 6.7716 limits [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 0, 13.696, 0, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 0, 13.696, 0, 0, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0] export 174.2537
2025-05-31 22:01:37.694131: Update event install {'domain': 'update', 'service': 'install', 'service_data': {'entity_id': 'update.target_timeframes_update', 'version': 'v1.1.1'}} None
2025-05-31 22:02:02.681255: >> Region optimisation pass width 240
2025-05-31 22:02:14.032532: Update event install {'domain': 'update', 'service': 'install', 'service_data': {'entity_id': 'update.solcast_pv_solar_update', 'version': 'v4.3.5'}} None
2025-05-31 22:02:22.062811: Update event install {'domain': 'update', 'service': 'install', 'service_data': {'entity_id': 'update.octopus_energy_update', 'version': 'v15.1.0'}} None
2025-05-31 22:03:58.293581: >> Region optimisation pass width 120
[... cut ...]
2025-05-31 22:05:01.744493: Optimise all charge found best buy/sell price band 14.8 best price at cost -2415.4563 metric -2319.233 keep 0 cycle 88.3072 carbon 0 import 23.75 cost -2415.4563 battery_value 6.7716 limits [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 0, 13.696, 0, 13.696, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 0, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 2.2, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 0, 13.696, 0, 0, 13.696, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 13.696, 0.0, 0.0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0] export 169.8544

```

pacemaker82, 2025-06-01:
Wanted to reflect on my plan for 8.20.3. After midnight the times seems to dramatically drop from over 5m to around 3m30s consistently. To note, I didn't change anything - I am on IOG and fixed export. There was no car charging last night (no car plan) so I am not sure if that's related. 

[image]

gcoan, 2025-06-01:
I purge my predbat.active sensor at 2am as part of a daily housekeeping, so I can't see what happened at midnight without looking at the predbat logfiles, but the 2am runs were 5 minutes consistently on 8.20.3:

[image]

plans at 10:30 were 3-4 minutes but the most recent run after this screenshot was 6 1/2 minutes with 3 minutes being in the region optimisation 120 part:

[image]

for me I'd say the opposite behaviour to @pacemaker82 

[image]

the shorter executions (smaller yellow sections) were when predbat was exporting this morning from 5-8:30, the longer executions were when charging up to 5am and now from 10am

I'm on Agile and there were overnight cheap prices last night and plunge negative rates during the day today

pacemaker82, 2025-06-02:
The pattern for me was somewhat similar overnight, in that at 5.30am this morning things slow down again:

[image]

At around 3am the plans are blazingly fast, around 3m.

At 6.30am they have over doubled to 6m42s, 6m28s, then averaging 5m30s

pacemaker82, 2025-06-02:
> For those of us on IOG the first plan after the cheap period ends likely is the longest plan window. As I understand it, detailed planning runs up-to a set number of charge periods ahead.

Interesting - so should expect an outlier after 5.30am charge period, but would subsequent ones need the same or similar processing time if nothing else is changing?

WyndStryke, 2025-06-02:
I don't know if this is any interest, but here are the times that it takes to process the 'compare' on different tariffs, with carbon off.

- E-on Next Drive  4m 54s
- Go/Fixed 3m 1s
- Go/Agile 6m 48s
- Agile/Fixed 5m 17s
- Agile/Agile 13m 31s
- Flux 2m 46s
- Snug/Fixed  3m 5s
- iFlux 2m 9s
- Cosy/Fixed 2m 41s
- Cosy/Agile 9m 4s
- iGo/Fixed 3m 10s
- iGo/Agile 5m 4s

Note that this does not take account of background activity, like the backup or nightly database purge.

<html>
<body>

current | E-on Next Drive | 2025-06-02 00:04:54 | £-6.84 | £-7.22 | £-5.47 | 93.21 | 51.58 | 2.49 | Existing
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
go_fixed | GO import/Fixed export | 2025-06-02 00:07:55 | £-5.68 | £-5.71 | £-4.21 | 87.32 | 44.72 | 5.63 |  
go_agile | GO import/Agile export | 2025-06-02 00:14:03 | £-3.66 | £-3.60 | £-2.79 | 49.88 | 3.54 | 5.57 |  
agile_fixed | Agile import/Fixed export | 2025-06-02 00:19:20 | £-4.83 | £-4.27 | £-2.88 | 69.58 | 24.9 | 13.66 |  
agile_agile | Agile import/Agile export | 2025-06-02 00:32:51 | £-3.61 | £-3.66 | £-2.71 | 58.14 | 12.49 | 5.71 |  
flux | Flux import/Export | 2025-06-02 00:35:05 | £-6.50 | £-6.68 | £-5.30 | 49.89 | 3.81 | 3.49 |  
snug_fixed | Snug import/Fixed export | 2025-06-02 00:38:10 | £-5.63 | £-5.60 | £-4.06 | 91.32 | 49.71 | 6.01 |  
iflux | Intelligent Flux import/Export | 2025-06-02 00:40:19 | £-8.75 | £-8.72 | £-6.07 | 53.15 | 14.51 | 5.01 | Best
cosy_fixed | Cosy import/Fixed export | 2025-06-02 00:43:00 | £-4.94 | £-3.11 | £-1.53 | 38.03 | 3.07 | 15.99 |  
cosy_agile | Cosy import/Agile export | 2025-06-02 00:52:04 | £-3.65 | £-3.05 | £-2.29 | 36.28 | 1.57 | 7.82 |  
igo_fixed | Intelligent GO import/Fixed export | 2025-06-02 00:55:14 | £-5.99 | £-6.06 | £-4.46 | 88.4 | 46.12 | 5.75 |  
igo_agile | Intelligent GO import/Agile export | 2025-06-02 01:00:18 | £-3.87 | £-3.94 | £-3.08 | 64.1 | 19.85 | 4.37

</body>
</html>

[current](https://7y4iqzf2h6lrayqyjthj49za2xttc7t4.ui.nabu.casa/api/hassio_ingress/9PzFX1Yv_JGJvK0sN1aZin9OaoQRV_Uw6dXadD-GXss/compare#heading-current)	E-on Next Drive	2025-06-02 00:04:54	£-6.84	£-7.22	£-5.47	93.21	51.58	2.49	Existing
[go_fixed](https://7y4iqzf2h6lrayqyjthj49za2xttc7t4.ui.nabu.casa/api/hassio_ingress/9PzFX1Yv_JGJvK0sN1aZin9OaoQRV_Uw6dXadD-GXss/compare#heading-go_fixed)	GO import/Fixed export	2025-06-02 00:07:55	£-5.68	£-5.71	£-4.21	87.32	44.72	5.63	
[go_agile](https://7y4iqzf2h6lrayqyjthj49za2xttc7t4.ui.nabu.casa/api/hassio_ingress/9PzFX1Yv_JGJvK0sN1aZin9OaoQRV_Uw6dXadD-GXss/compare#heading-go_agile)	GO import/Agile export	2025-06-02 00:14:03	£-3.66	£-3.60	£-2.79	49.88	3.54	5.57	
[agile_fixed](https://7y4iqzf2h6lrayqyjthj49za2xttc7t4.ui.nabu.casa/api/hassio_ingress/9PzFX1Yv_JGJvK0sN1aZin9OaoQRV_Uw6dXadD-GXss/compare#heading-agile_fixed)	Agile import/Fixed export	2025-06-02 00:19:20	£-4.83	£-4.27	£-2.88	69.58	24.9	13.66	
[agile_agile](https://7y4iqzf2h6lrayqyjthj49za2xttc7t4.ui.nabu.casa/api/hassio_ingress/9PzFX1Yv_JGJvK0sN1aZin9OaoQRV_Uw6dXadD-GXss/compare#heading-agile_agile)	Agile import/Agile export	2025-06-02 00:32:51	£-3.61	£-3.66	£-2.71	58.14	12.49	5.71	
[flux](https://7y4iqzf2h6lrayqyjthj49za2xttc7t4.ui.nabu.casa/api/hassio_ingress/9PzFX1Yv_JGJvK0sN1aZin9OaoQRV_Uw6dXadD-GXss/compare#heading-flux)	Flux import/Export	2025-06-02 00:35:05	£-6.50	£-6.68	£-5.30	49.89	3.81	3.49	
[snug_fixed](https://7y4iqzf2h6lrayqyjthj49za2xttc7t4.ui.nabu.casa/api/hassio_ingress/9PzFX1Yv_JGJvK0sN1aZin9OaoQRV_Uw6dXadD-GXss/compare#heading-snug_fixed)	Snug import/Fixed export	2025-06-02 00:38:10	£-5.63	£-5.60	£-4.06	91.32	49.71	6.01	
[iflux](https://7y4iqzf2h6lrayqyjthj49za2xttc7t4.ui.nabu.casa/api/hassio_ingress/9PzFX1Yv_JGJvK0sN1aZin9OaoQRV_Uw6dXadD-GXss/compare#heading-iflux)	Intelligent Flux import/Export	2025-06-02 00:40:19	£-8.75	£-8.72	£-6.07	53.15	14.51	5.01	Best
[cosy_fixed](https://7y4iqzf2h6lrayqyjthj49za2xttc7t4.ui.nabu.casa/api/hassio_ingress/9PzFX1Yv_JGJvK0sN1aZin9OaoQRV_Uw6dXadD-GXss/compare#heading-cosy_fixed)	Cosy import/Fixed export	2025-06-02 00:43:00	£-4.94	£-3.11	£-1.53	38.03	3.07	15.99	
[cosy_agile](https://7y4iqzf2h6lrayqyjthj49za2xttc7t4.ui.nabu.casa/api/hassio_ingress/9PzFX1Yv_JGJvK0sN1aZin9OaoQRV_Uw6dXadD-GXss/compare#heading-cosy_agile)	Cosy import/Agile export	2025-06-02 00:52:04	£-3.65	£-3.05	£-2.29	36.28	1.57	7.82	
[igo_fixed](https://7y4iqzf2h6lrayqyjthj49za2xttc7t4.ui.nabu.casa/api/hassio_ingress/9PzFX1Yv_JGJvK0sN1aZin9OaoQRV_Uw6dXadD-GXss/compare#heading-igo_fixed)	Intelligent GO import/Fixed export	2025-06-02 00:55:14	£-5.99	£-6.06	£-4.46	88.4	46.12	5.75	
[igo_agile](https://7y4iqzf2h6lrayqyjthj49za2xttc7t4.ui.nabu.casa/api/hassio_ingress/9PzFX1Yv_JGJvK0sN1aZin9OaoQRV_Uw6dXadD-GXss/compare#heading-igo_agile)	Intelligent GO import/Agile export	2025-06-02 01:00:18	£-3.87	£-3.94	£-3.08	64.1	19.85	4.37

gcoan, 2025-06-02:
> Interesting - so should expect an outlier after 5.30am charge period, but would subsequent ones need the same or similar processing time if nothing else is changing?

Predbat does a full replan every 10 minutes, and in between on the 5 minute run it just executes the previous plan. 
You can change these frequencies if you want to, but "nothing else is changing" isn't true, your house load could be more or less than predicted, likewise your solar generation can vary. This is why predbat does a full replan every 10 minutes.

WyndStryke, 2025-06-02:
> You can change these frequencies if you want to, but "nothing else is changing" isn't true, your house load could be more or less than predicted, likewise your solar generation can vary. This is why predbat does a full replan every 10 minutes.

I wonder if there would be any benefit in comparing the various actuals vs forecasts,  and only doing a full replan if there was a noteworthy difference.  

Probably somewhat pointless since the Pi (or whatever) is going to be running 24/7 anyway, so does it actually matter if it runs the replan more than strictly necessary.

pacemaker82, 2025-06-02:
I had some thought that maybe the long times were something related to predicted Vs actual solar

it was a choppy day

[image]

I overlayed the PV Power onto today's Active Status 

[image]

5.30am thru to ~5pm the plan time is high. The post 5pm time is less, but not as good as the pre 5.30am average.

johnwb87, 2025-06-03:
I’m confident that I debugged it down to the charge optimisation steps. @gcoan was broadly in agreement.

So I think inputs that affect charging changing plan generation times is consistent with that.

I also still stand behind my understanding of how far out predbat does detailed charge planning. If you have the home battery prediction graph I believe that is what the black record cut of is showing. I can’t find it in the documentation but I’m sure Trefor has said that is the case in a previous issue. For those on IOG that jumps forward more than twelve hours for the first run after 5:30 as predbat shifts from courser planning to detailed planning for a much greater portion of the 48 hour plan window.

Add in the apparent poor utilisation of the HA Green processor during charge optimisation steps and I think we have narrowed down to some sort of bottleneck, unexpected wait or ineffective threading during charge calculation that is exacerbated by adding additional metrics such as carbon.

If I thought I could add any value in the actual code I would have a look, however it’s beyond me. I’m not even sure which file to look at, plan or prediction?

pacemaker82, 2025-06-03:
Just adding that the plan duration hasn’t decreased today since I got an IOG plan like it did yesterday. Still well over 5m tonight since charge window finished about 5.30am. I’m giving up looking for any other patterns

johnwb87, 2025-06-05:
My post 5:30 run times were 13 and 14 minutes today. So I thought time for a run with debug. Well those runs have been taking 18 minutes as I flicked the switch and carried on with life for a while.

Here's a debug and log file from a plan calculation run.

[predbat_debug_08_40_00.yaml.txt](https://github.com/user-attachments/files/20607634/predbat_debug_08_40_00.yaml.txt)

[predbat.1 (2).log](https://github.com/user-attachments/files/20607638/predbat.1.2.log)

springfall2008, 2025-06-07:
Have you tried the latest v8.20.x ?

johnwb87, 2025-06-08:
Hi. Yes I’m on the latest release, I have auto update switched on.
Today I’m between 8-10 minutes. Which is some improvement compared to the 10+ minutes I was seeing a few days ago.

pacemaker82, 2025-06-08:
Run times exactly the same here, no improvements in recent updates. No carbon optimisation, just IOG.

Today below. Still averaging ~3.7 minutes during the charge window and over 5-6 minutes rest of day.

WyndStryke, 2025-06-08:
With 8.19, carbon on, each run was about 5 minutes (and 1h 44m for the overnight tariff comparison).

With 8.20.12, carbon on, the runs are now about 4m 10s.

With carbon off, the runs are about 3m 15s.

pacemaker82, 2025-06-17:
Just FYI really, but during the overnight charge window (when plan calculation is much shorter), the pattern is the same - just a bit tighter in terms of time taken, and therefore allowing the "short run" to have a clear gap between (see 7.15am as example of that short run)

[image]

johnwb87, 2025-06-17:
I need to enable this level of tracking to be sure. But it makes sense as what is driving the difference between 6am and 6pm is the duration being planned. So there is just more for the charge planning to look at and because it’s the charge optimisation that is happening in that low period it simply has more to do.

PhillWalker, 2025-06-28:
+1 for much slower after recent release updates

GnomedDev, 2025-06-29:
Is there a particularly heavy and self contained section of the planning code? If so, it might be worth trying out Cython or PyO3 to avoid the 100x performance overhead of using Python for hot paths.

johnwb87, 2025-07-05:
Really big strides with performance from 8.22.1.

Before update around 7 minutes, after solidly at 4 minutes and sometimes close to 3.

[image]

closed 2025-09-04 by gcoan