springfall2008/batpred #2396: soc_max using input_number

dannytsang, 2025-05-26:
**Is your feature request related to a problem? Please describe.**
I have a helper (input_number) holding my soc_max value. I don't think it's able to read the entity:

[image]

**Describe the solution you'd like**
Ability to use input_number.

**Describe alternatives you've considered**
Hard coding the value in predbat.yaml.

**Additional context**
I believe it's not reading the input_number because there is no value next to it but I could be wrong.

springfall2008, 2025-05-31:
This should work, does the input number exist?

dannytsang, 2025-05-31:
> This should work, does the input number exist?

I’m sure it did but let me double check next week.

dannytsang, 2025-06-02:
The helper definitely exists with a value:

[image]

dannytsang, 2025-06-23:
I can't see any read out of values, only when there's an error reading them for example:
```
16057 | 2025-06-23 18:00:11.996479: Warn: Validation of apps.yaml found configuration item 'pv_power' element sensor.growatt_sph_pv_power is not a valid type float
```
although I haven't seen any related to the input_number helper I'm using.

Do I need to turn on debug?

chalfontchubby, 2026-10-19:
_Posted by Claude on behalf of @chalfontchubby._

Closing as working as designed and stale. `soc_max` is read through the same entity lookup as every other setting, so an `input_number` helper is supported, and nobody has reproduced it not being read. To check on your system, look for the startup log line `Inverter 0 with soc_max X kWh ...`, which shows the value Predbat actually used. If that shows the wrong value with an `input_number` on a current release, please open a new issue with the log.

closed 2026-10-19 by chalfontchubby