springfall2008/batpred #2397: Negative Agile Rates

weevil23, 2025-05-29:
Ideally when Agile is negative we should disable solar generation and full charge from mains ( not exactly green but economically sound).
Can Giv 5.0 Gen 1 be commanded to actually turn off solar?
If not can a notification of start and finish negative rates be issued so the array can be isolated?

weevil23, 2025-05-29:
OK
We use 30-40kWh per day. I have 2 x 9.5 batts that on a full sun day are just about charged by 4pm (ie we rarely sit with full batteries), 1 x Giv 5.0 and 6kWh solar.
If we are being paid to use electricity how can it not make sense to charge from the mains?

gcoan, 2025-05-29:
Predbat will charge from the mains if it can, but negative rates during sunny windy days as we had today are only cost effective if you can consume more than your solar panels are generating, otherwise you're continuing to export and in most cases your export income will be higher than what you get paid for negative rate import. 

You should be able to create an automation that checks import rates and alerts when goes negative and goes positive.

closed 2025-11-05 by gcoan