springfall2008/batpred #2407: 8.20.2 suboptimal plan, debug requested

PianSom, 2025-05-31:
**Describe the bug**
As requested on fb

**Expected behaviour**
later discharge this evening

**Predbat version**

8.20.2

**Environment details**

AIOx2+Gateway
IOG
Docker HA + Predbat

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Debug attached

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

[predbat_debug.yaml.txt](https://github.com/user-attachments/files/20535434/predbat_debug.yaml.txt)

mpartington, 2025-06-01:
Tagging on to this issue to provide some requested debug for version .5 - shown the non continuous discharge (rather than 1 long slot)

[image]

[predbat_debug.yaml (3).txt](https://github.com/user-attachments/files/20541036/predbat_debug.yaml.3.txt)

mpartington, 2025-06-01:
Also adding debug for an early discharge

[predbat_debug.yaml.txt](https://github.com/user-attachments/files/20541353/predbat_debug.yaml.txt)

[image]

springfall2008, 2025-06-01:
New release with another fix: https://github.com/springfall2008/batpred/releases/tag/v8.20.6

mpartington, 2025-06-01:
> Not necessarily a problem, but a slightly eccentric change of state at noon tomorrow. 8.20.6
> 
@springfall2008 , worth noting that firmware from 3018 onwards does all the balancing calibration at the top end. So ideally if the battery reaches 100%, it should be left to 'top up' charge as necessary. Setting freeze export disrupts this process, as it should carry on charging until it hits the true full.point.  Demand mode would be more appropriate.

closed 2025-06-22 by springfall2008