springfall2008/batpred #2411: Plan since v.8.19.9 has been strange for my setup. Early discharging and holding charge

ma9mwah, 2025-06-01:
**Describe the bug**
Ever since v8.19.9 the plan for my battery just doesn't seem right. I'm on Octopus Go so a pretty simple tariff. The new versions discharge the battery early during the day which then might then cause you to run out of batery early if an unexpected load occured. It also holds the battery at 100% during the night where the previous plan would discharge then charge.

**Expected behaviour**
I would expect the new version to be similar to old, or even better if it could push the final discharge to a latter slot by an hour or so

**Predbat version**
v8.19.9 and v8.20.4

**Environment details**

- GivEnergy AIO
- Standard HAOS install 

**Screenshots**
OLD:

[image]

NEW:

[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

ma9mwah, 2025-06-01:
[predbat_debug_v8.20.5.yaml.txt](https://github.com/user-attachments/files/20541059/predbat_debug_v8.20.5.yaml.txt)
[predbat_debug_v8.19.9.yaml.txt](https://github.com/user-attachments/files/20541058/predbat_debug_v8.19.9.yaml.txt)

springfall2008, 2025-06-01:
So the problem here is you have 1 minute of zero rate at midnight, your rates_export should just set:

```yaml
rates_export
   - rate: 4.1
```
As you had an end time of 1 minute before midnight.

springfall2008, 2025-06-01:
That said I'll do a simple fix to ignore anything not on a 5 minute boundary when setting rate bands

closed 2025-06-03 by ma9mwah