springfall2008/batpred #2417: Overrides being ignored until Predbat add-on is restarted

mpartington, 2025-06-02:
**Describe the bug**
Myself and another user as seeing overrides being selected ok, but not being shown in a plan until the add on is restarted.

It's worse if you select multiple slots at the same time, but I have just captured debug where it ignores a single slot

Seems like a bug has crept in recently 

**Expected behaviour**
overrides to be actioned correctly in the plan

**Predbat version**
8.20.7

Standard installation 

Will add screenshots and debug to comments

**Predbat debug yaml file**
This is important for any plan related issues.

Once you have captured the issue go to the Web UI, in the Dash tab click on **predbat_debug.yaml** and wait 30 seconds for it to download. Select 'keep' if your Web Browser thinks the file is dangerous.

Rename the download file to predbat_debug.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

mpartington, 2025-06-02:
I missed the screenshot here, but got the debug.

An export at 20.30 was selected, but not planned (I did wait for it to refresh)

[predbat_debug1.yaml.txt](https://github.com/user-attachments/files/20558216/predbat_debug1.yaml.txt)

qyaunos1, 2025-06-02:
my version is 8.20.6 but I have noticed the problem start over a month ago.

gcoan, 2025-06-03:
@mpartington @qyaunos1 can you confirm that the problem is (only?) with manually selecting Force Export Freeze?

I have a daily automation that sets Force Export every morning on my predbat and from the logs that works perfectly.  From your screenshots it too is working, the problem is Freeze Export sometimes doesn't 'take' and sometimes results in a Force Export?

mpartington, 2025-06-03:
I think its any override. I just tried to create some debug using
FreezeExport. Maybe it's using the selector Vs automation

gcoan, 2025-06-03:
> I think its any override. I just tried to create some debug using
> 
> FreezeExport. Maybe it's using the selector Vs automation
> 

I use an automation to set my force export from 5:30 through to 8:00, and as I said these work OK.

If I do repeated manual selects I think I have noticed that sometimes predbat doesn't recognise all the changes made, I think it starts processing the first one and then if you make a second soon after it gets interrupted.

Manually turning on predbat.active I found would cause all the slots entered to be picked up, but I don't often make lots of manual overrides so didn't really look into this.

springfall2008, 2025-06-07:
I think I may have a fix for the reason why it doesn't update again, but it will have to re-calculate the plan a second time after the first selection triggered the update

qyaunos1, 2025-06-07:
Thanks Trefor. Will you message on here with the version number when you have released a fix please? Out of interest, what was causing the issue?

springfall2008, 2025-06-07:
Try this new release: https://github.com/springfall2008/batpred/releases/tag/v8.20.10

mpartington, 2025-06-08:
> Try this new release: https://github.com/springfall2008/batpred/releases/tag/v8.20.10

I'm afraid it hasn't worked. It's now over 30mins since I selected the overrides and 14-15.00 is still missing.

Additionally, I think there is a new bug. If you turn export freeze off, it still plans it

[predbat_debug-1.yaml.txt](https://github.com/user-attachments/files/20643760/predbat_debug-1.yaml.txt)

[image]

springfall2008, 2025-06-21:
>Additionally, I think there is a new bug. If you turn export freeze off, it still plans it

You debug yaml had set_export_freeze enabled, when I turned it off then the freeze slots went away.

closed 2026-08-21 by mpartington