springfall2008/batpred #2423: How to clear PredBat Historical usage data?

Sycamore967, 2025-06-05:
A while ago, I updated my apps.yaml file to a later version (from the web page) to take advantage of all the new features that had been introduced since I installed PB 18months or so ago.

(I don't know of a better way of doing this please tell me if there is).

I missed adding the car charge sensor back in and didn't spot it for a while.

Predbat now insists the house load is over 3kwh for each slot on most days, anywhere from 19:30 right up until 05:30.  Obviously, it's seeing the car charge as house load as it didn't have the sensor configured.

Several weeks ago, I fixed the sensor and expected the history and PB's anticipated load to gradually sort itself out, accepting that the figures would be wide of the mark until then.

However, it's been well over the 22 day history that HA is set to hold and PB is still showing the 3Kwh + anticipated load against the house.

It's a long explanation for a short question:  How do I sort PB's anticipated load?

(So far I've dropped HA's history purge to 6 days and PB's history to 7 days, in the hope of creating an empty history, but it doesn't seem to have worked, so my logic must be wrong).

Can anyone suggest help?

gcoan, 2025-06-05:
Predbat runs a new calculation every 10 minutes. In each run it takes house load and subtracts car charging based on the HA sensors. 

So if predbat is not correctly subtracting car load then either the underlying sensors don't show that load correctly or your apps.yaml is misconfigured.

And HA history should always be higher than predbat days_previous, if you are purging HA history too early that means predbat has nothing to forecast from

Sycamore967, 2025-06-05:
Hi, thanks for responding.

I think you've miss understood.  I know that in my history (until I fixed it) that PB had no way to subtract the car load.  That has been fixed for over 22 days which is the HA setting and PB is set to 21 days.  What I'm saying is that PB is STILL forecasting a house load based on old data that just can't exist any more (or at least shouldn't).  I don't believe there are any config errors in my apps.yaml and I've attached it just in case.  PB now correctly shows forecast car load in the plan and Car Charging Hold as the status when the car is actively charging, so it looks ok from what I can see.  It's just the forecast load that wrong and I doesn't seem to be clearing up as I'd expect.

[apps.yaml.txt](https://github.com/user-attachments/files/20608997/apps.yaml.txt)

gcoan, 2025-06-05:
I have not misunderstood, but as I said it looks like you have not correctly set predbat up.

Predbat forecasts based upon the information it is given so if its is forecasting an incorrect house load then the information you have given it must be incorrect.

From your apps.yaml you are using the givtcp load sensor to give your historical house load, which if you haven't arranged the CT clamps for your inverter appropriately will include car charging information.

You therefore need to remove the car charging information from the house load https://springfall2008.github.io/batpred/apps-yaml/#car-charging-filtering

You need to set car_charging_energy to a sensor that gives the historical car charging and there's details in the above link about ensuring it is set to the correct sensor.

You have set it to 

car_charging_energy: 're:(sensor.hypervolt_session_energy)'

Have you checked that this is the correct sensor?

According to the predbat documentation for Hypervolt https://springfall2008.github.io/batpred/devices/#hypervolt you should have used 're:(sensor.hypervolt_session_energy_total_increasing)'

Also the car charging filtering says you need to turn switch.predbat_car_charging_hold on.  Is it on?

closed 2025-06-07 by Sycamore967