springfall2008/batpred #2443: When using Predbat-led EV charging the input_number charge rate is ignored

PianSom, 2025-06-10:
**Describe the bug**
I am using Predbat-led EV charging on IOG (so not looking to use any bonus slots). My EV charges at the same time as my battery, so the voltage drops, so the EV charge rate I can achieve is about 6.3kW. I have set input_number.predbat_car_charging_rate to 6.0kW. 

However, this input is ignored for the plan - which shows the EV charging at 7.2kW

**Expected behaviour**
I expect Predbat to use the specified charge rate

**Predbat version**
8.20.14

**Environment details**

- AIOx2+GW
- Docker
- Zappi charger/IOG

**Screenshots**

[image]

**Log file**

[predbat_debug.yaml.txt](https://github.com/user-attachments/files/20673386/predbat_debug.yaml.txt)

**Predbat debug yaml file**
This is important for any plan related issues.

Once you have captured the issue go to the Web UI, in the Dash tab click on **predbat_debug.yaml** and wait 30 seconds for it to download. Select 'keep' if your Web Browser thinks the file is dangerous.

Rename the download file to predbat_debug.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

chalfontchubby, 2026-08-20:
Thanks for the report. This issue has been open for over a year with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-08-28).

PianSom, 2026-08-21:
I believe this is still an issue

chalfontchubby, 2026-08-21:
Thanks for confirming @PianSom - keeping this open.

PianSom, 2026-08-24:
Correction, sorry @chalfontchubby  - this is not an issue any more. Closing.

closed 2026-08-24 by PianSom