springfall2008/batpred #2444: Growatt via Solax Modbus Addon: Unable to read inverter time string

MuadDib007, 2025-06-10:
**Describe the bug**
When using the rtc sensor from the Solax Modbus Integration (Connected to my Growatt MOD10KTL3-XH(BP)) as input for inverter_time in apps.yaml I receive the following error in the logs:

` Warn: Unable to read inverter time string 2025-06-10 15:44:42 using formats ['%Y-%m-%dT%H:%M:%S%z', '%Y-%m-%d %H:%M:%S%z', '%Y-%m-%dT%H:%M:%S']`

**Expected behaviour**
Expect Predbat to use this time string format without issue

**Predbat version**
v8.20.14

**Environment details**

- Growatt MOD10KTL3-XH(BP) with 30 kWh APX Batteries connected via Solax Modbus integration.
- HAOS running inside Vmware ESXi

**Screenshots**

[image]

Entity state of rtc:

[image]

**Log file**
[predbat (2).log](https://github.com/user-attachments/files/20674062/predbat.2.log)

**Predbat debug yaml file**
[predbat_debug.txt](https://github.com/user-attachments/files/20674053/predbat_debug.txt)

springfall2008, 2025-06-17:
You can probably just comment out the inverter time for now, but its just a different time format

chalfontchubby, 2026-09-19:
From the issue post this looks like a space separator, no fractional seconds, a combination missing from the list of strings. The bot should be able to confirm that and add the required string.

springfall2008, 2026-09-19:
🤖 Automated first-pass triage — a maintainer will review before any action is taken.

**Classification:** bug (a small format-coverage gap in the inverter-time parser), `priority_low`. The existing `enhancement` framing fits too — it's a "support another time format" fix. The maintainer's workaround stands: remove `inverter_time` from apps.yaml (it only feeds clock-skew detection) or reformat the sensor via a template.

**What I checked** (against main at v9.0.3-35-g002f6b86, 2026-09-16):

- The attached log + debug yaml confirm the setup: `inverter_type: SA`, `inverter_time: [sensor.g10k_rtc]` (the Solax Modbus `rtc` sensor), `timezone: Europe/Amsterdam`.
- The reported string `2025-06-10 15:44:42` (space separator, **no** timezone offset) matches none of the three formats Predbat tries: `TIME_FORMAT` (`%Y-%m-%dT%H:%M:%S%z`), `TIME_FORMAT_OCTOPUS` (`%Y-%m-%d %H:%M:%S%z`), and — for inverter type SA — the per-type `clock_time_format` `%Y-%m-%dT%H:%M:%S` (apps/predbat/config.py:2123). Verified with a direct strptime check: exactly `%Y-%m-%d %H:%M:%S` is the missing combination, confirming the collaborator's reading of the issue.
- Not fixed on main since the report: the reporter's v8.20.14 log shows a format list identical to current main's, so the warning still reproduces on v9.0.3.
- Failure consequences (parse chain in `Inverter.__init__`, apps/predbat/inverter.py:508-521): a warn every 5-minute cycle, `inverter_time` stays `None` so clock-skew detection never runs for that inverter, and an `auto_restart("Unable to read inverter time")` trigger each cycle — with `auto_restart` unconfigured (as here) that only logs a warning; with it configured it restarts inverter control.
- Possibly related: #3599 logs the same warning but with the string literally `unknown` on inverter type Solax — a different cause.

**Verified fix (local throwaway clone, not committed):** reproduced the failure with a probe against the test fixture — the string falls through all three formats and lands in the auto-restart trigger — then added `TIME_FORMAT_SOLIS` (`%Y-%m-%d %H:%M:%S`, already defined at apps/predbat/const.py:24) as a fourth fallback in the chain, and the probe plus the full `inverter` test module passed. Suggested fix for the BOT_PR flow: extend the fallback chain at apps/predbat/inverter.py:508-521 with `TIME_FORMAT_SOLIS` (a small loop over the formats keeps the try/except stack readable).

Nothing missing from the bug template (version, environment, log and debug yaml all present).

closed 2026-09-19 by springfall2008