springfall2008/batpred #2445: PV Forecast differs from solcast and pv scaling has no effect

raldred, 2025-06-10:
**Describe the bug**
Solcast has been predicting much higher forecasts than I've been seeing the last few days.
In looking into this I have noticed that predbat's PV forecast is higher than what solcast has itself.
I have tried to use `predbat_pv_scaling`, dropping it right down to see if that can adjust what predbat sees but it doesn't appear to change anything.

**Expected behaviour**
Predbat shouldn't be showing pv forcast above what solcast has, it should also reduce pv forecast adjusting using pv scaling

**Predbat version**
v8.20.14

**Environment details**

- Luxpower
- Standard HAOS installer
- Predbat addon

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

[image]
[image]

**Log file**

[predbat.log](https://github.com/user-attachments/files/20676596/predbat.log)

johnwb87, 2025-06-11:
`Warn: PV Forecast data adds up to 352.28 kWh but total sensors add up to 219.60000000000002 kWh, this is unexpected and hence data maybe misleading (factor 1.6)`

That's interesting. Also the preceding lines show that predbat is seeing the sensor correctly. I think it suggests that there is something going on with the detailed forecast. That is what predbat uses to build the 30 min solar forecast.

springfall2008, 2025-06-14:
I'd need to see what's in detailedforecast to compare this against the total in the sensor, this is what its saying is mismatched.

raldred, 2025-06-15:
I've switched to main, warning is still present and PV forecast is still showing more than solcast 

The warning shows...
```
18264	2025-06-15 21:20:04.138005: Warn: PV Forecast today adds up to 78.05 kWh but total sensors add up to 54.6 kWh, this is unexpected and hence data maybe misleading (factor 1.4)
18314	2025-06-15 21:20:05.915070: PV: [ 36.6, 36.61, 36.61, 36.61, 36.61, 36.61, 36.61, 36.61, 36.61, 36.61, 36.61, 36.61, 36.61, 36.61, 36.61, 36.61, 36.61, 36.66, 36.77, 37.02, 37.5, 38.3, 39.6, 41.47, 43.93, 46.91, 50.35, 54.12, 58.03, 62.02, 66.06, 70.11, 74.09, 77.95, 81.63, 85.08, 88.23, 91.04, 93.46, 95.48, 97.09, 98.28, 99.03, 99.61, 100.04, 100.32, 100.53, 100.69, 100.79, 100.83, 100.83, 100.83, 100.83, 100.83, 100.83]
18281	2025-06-15 21:20:05.645235: PV Forecast for day d3 is 75.23 (43.2 10% 82.7 90% 75.23 CL) kWh
18280	2025-06-15 21:20:05.637850: PV Forecast for day d2 is 45.95 (9.37 10% 78.41 90% 31.46 CL) kWh
18279	2025-06-15 21:20:05.630675: PV Forecast for day tomorrow is 64.22 (41.79 10% 80.47 90% 43.96 CL) kWh
18278	2025-06-15 21:20:05.612892: PV Forecast for today is 55.75 (26.56 10% 78.01 90% 38.11 calib) kWh and left today is 0.02 (0.01 10% 0.03 90% 0.01 calib) kWh
18277	2025-06-15 21:20:05.592496: PV Calibration: PV production: 73.14 kWh, Total forecast: 105.81 kWh adjustment 0.6912x slot adjustments {0: 1.0, 30: 1.0, 60: 1.0, 90: 1.0, 120: 1.0, 150: 1.0, 180: 1.0, 210: 1.0, 240: 1.0, 270: 1.0, 300: 1.0, 330: 1.0, 360: 1.0, 390: 0.9672, 420: 0.9293, 450: 0.996, 480: 1.0654, 510: 0.7789, 540: 0.9391, 570: 0.8116, 600: 1.0493, 630: 1.126, 660: 1.1472, 690: 0.9585, 720: 0.6998, 750: 0.8639, 780: 1.1532, 810: 1.0594, 840: 1.0678, 870: 1.039, 900: 0.9467, 930: 0.9089, 960: 1.0785, 990: 1.1605, 1020: 1.0843, 1050: 1.0368, 1080: 0.9866, 1110: 1.0619, 1140: 1.0, 1170: 1.0, 1200: 1.0, 1230: 1.0, 1260: 1.0, 1290: 1.0, 1320: 1.0, 1350: 1.0, 1380: 1.0, 1410: 1.0}
18276	2025-06-15 21:20:05.592192: PV Calibration: Worst day scaling factor 0.57 best day scaling factor 1.5
18275	2025-06-15 21:20:05.592159: PV Calibration: Past day 1 had 45.87 kWh of PV forecast data and actual 71.61 kWh
18274	2025-06-15 21:20:05.592123: PV Calibration: Past day 2 had 65.26 kWh of PV forecast data and actual 41.73 kWh
18273	2025-06-15 21:20:05.592065: PV Calibration: Past day 3 had 42.45 kWh of PV forecast data and actual 30.0 kWh
18272	2025-06-15 21:20:05.592028: PV Calibration: Past day 4 had 72.27 kWh of PV forecast data and actual 60.01 kWh
18271	2025-06-15 21:20:05.591992: PV Calibration: Past day 5 had 58.44 kWh of PV forecast data and actual 40.81 kWh
18270	2025-06-15 21:20:05.591950: PV Calibration: Past day 6 had 38.2 kWh of PV forecast data and actual 29.22 kWh
18269	2025-06-15 21:20:05.591907: PV Calibration: Past day 7 had 52.93 kWh of PV forecast data and actual 30.06 kWh
18268	2025-06-15 21:20:05.591870: PV Calibration: Past day 8 had 40.83 kWh of PV forecast data and actual 23.28 kWh
18267	2025-06-15 21:20:05.591831: PV Calibration: Past day 9 had 59.91 kWh of PV forecast data and actual 38.84 kWh
18266	2025-06-15 21:20:05.591725: PV Calibration: Past day 10 had 0.01 kWh of PV forecast data and actual 0.13 kWh
18265	2025-06-15 21:20:04.161550: PV Calibration: Fetching PV data for calibration
18264	2025-06-15 21:20:04.138005: Warn: PV Forecast today adds up to 78.05 kWh but total sensors add up to 54.6 kWh, this is unexpected and hence data maybe misleading (factor 1.4)
18263	2025-06-15 21:20:04.137962: PV Data for pv_forecast_d4 total 57.89 kWh

```

[image]

Here's detailedForecast from the solcast sensor

```
[{'period_start': datetime.datetime(2025, 6, 15, 0, 0, tzinfo=zoneinfo.ZoneInfo(key='Europe/London')), 'pv_estimate': 0.0, 'pv_estimate10': 0.0, 'pv_estimate90': 0.0}, {'period_start': datetime.datetime(2025, 6, 15, 0, 30, tzinfo=zoneinfo.ZoneInfo(key='Europe/London')), 'pv_estimate': 0.0, 'pv_estimate10': 0.0, 'pv_estimate90': 0.0}, {'period_start': datetime.datetime(2025, 6, 15, 1, 0, tzinfo=zoneinfo.ZoneInfo(key='Europe/London')), 'pv_estimate': 0.0, 'pv_estimate10': 0.0, 'pv_estimate90': 0.0}, {'period_start': datetime.datetime(2025, 6, 15, 1, 30, tzinfo=zoneinfo.ZoneInfo(key='Europe/London')), 'pv_estimate': 0.0, 'pv_estimate10': 0.0, 'pv_estimate90': 0.0}, {'period_start': datetime.datetime(2025, 6, 15, 2, 0, tzinfo=zoneinfo.ZoneInfo(key='Europe/London')), 'pv_estimate': 0.0, 'pv_estimate10': 0.0, 'pv_estimate90': 0.0}, {'period_start': datetime.datetime(2025, 6, 15, 2, 30, tzinfo=zoneinfo.ZoneInfo(key='Europe/London')), 'pv_estimate': 0.0, 'pv_estimate10': 0.0, 'pv_estimate90': 0.0}, {'period_start': datetime.datetime(2025, 6, 15, 3, 0, tzinfo=zoneinfo.ZoneInfo(key='Europe/London')), 'pv_estimate': 0.0, 'pv_estimate10': 0.0, 'pv_estimate90': 0.0}, {'period_start': datetime.datetime(2025, 6, 15, 3, 30, tzinfo=zoneinfo.ZoneInfo(key='Europe/London')), 'pv_estimate': 0.0, 'pv_estimate10': 0.0, 'pv_estimate90': 0.0}, {'period_start': datetime.datetime(2025, 6, 15, 4, 0, tzinfo=zoneinfo.ZoneInfo(key='Europe/London')), 'pv_estimate': 0.0, 'pv_estimate10': 0.0, 'pv_estimate90': 0.0}, {'period_start': datetime.datetime(2025, 6, 15, 4, 30, tzinfo=zoneinfo.ZoneInfo(key='Europe/London')), 'pv_estimate': 0.0043, 'pv_estimate10': 0.0, 'pv_estimate90': 0.0043}, {'period_start': datetime.datetime(2025, 6, 15, 5, 0, tzinfo=zoneinfo.ZoneInfo(key='Europe/London')), 'pv_estimate': 0.0522, 'pv_estimate10': 0.0218, 'pv_estimate90': 0.0827}, {'period_start': datetime.datetime(2025, 6, 15, 5, 30, tzinfo [... cut ...] start': datetime.datetime(2025, 6, 15, 22, 30, tzinfo=zoneinfo.ZoneInfo(key='Europe/London')), 'pv_estimate': 0.0, 'pv_estimate10': 0.0, 'pv_estimate90': 0.0}, {'period_start': datetime.datetime(2025, 6, 15, 23, 0, tzinfo=zoneinfo.ZoneInfo(key='Europe/London')), 'pv_estimate': 0.0, 'pv_estimate10': 0.0, 'pv_estimate90': 0.0}, {'period_start': datetime.datetime(2025, 6, 15, 23, 30, tzinfo=zoneinfo.ZoneInfo(key='Europe/London')), 'pv_estimate': 0.0, 'pv_estimate10': 0.0, 'pv_estimate90': 0.0}]
```

springfall2008, 2025-06-17:
So the data you posted adds up to 78.0526 but it doesn't match the sensor value which is 54.

Something is wrong with your Solcast setup as the data is bogus.

You can even see it on the card itself, it has the estimate for the two roof aspects but the total isn't the two combined.

I'm wondering if you have an AC Limit set in the Solcast web site that is causing it to clip the data here? If so I'd set it to something large like 99 and retry?

closed 2026-08-20 by chalfontchubby