springfall2008/batpred #2446: Add Zero export control option (Solar)

crcfree, 2025-06-10:
**Is your feature request related to a problem? Please describe.**
Allow Predbat to control and calculate/plan for zero solar export when export price is negative.

**Describe the solution you'd like**
Most inverters support zero export to be enabled.(typically a switch entity in HA).
When export prices are below a set threshold predbat will enable the zero export - but also adjust the plan according.

**Describe alternatives you've considered**
I already have a automation setup to handle this, but the plan is not adjusted as Predbat thinks it will still export at a negative rate.

Getting Predbat in control will present a more precise plan when solar export is disabled.

springfall2008, 2025-06-14:
You want an option to assume if export rates are negative then the price will be zero instead?

crcfree, 2025-06-15:
Most inverters has a function to limit(zero) export (only charge battery, and support the house load), it will adjust the solar input to match that. 
This way export can be limited(close to zero) when for example the export price is negative and lots of solar is available.
In many countries we will actually loose money if we export at negative prices - add a export tarif to that, and it will add up to be a very bad business case.
For most inverters this is just an on/off function - no other input is needed.
Predbat needs be able to control this function and adjust the plan according to the above.

Have a look at:
https://growattinverters.com/Export-Limitation-Solution.html
https://growatt.pl/wp-content/uploads/2020/01/Growatt-Export-Limitation-Guide.pdf

springfall2008, 2026-08-29:
This is an automated first-pass triage. A maintainer will review before any labels/actions are treated as final.

**Classification:** enhancement (feature request). No priority applied — per triage guidelines, priority is skipped for feature requests.

**Investigated against:** current `main` (`v8.53.5-25-ge126ad35`).

Confirmed the gap still exists: `export_limit` (`apps/predbat/config.py:2566`, consumed in `apps/predbat/inverter.py`) is a static per-inverter value with no link to the export rate, and there is no existing mechanism for Predbat to drive an inverter's zero-export/export-limit HA entity. So today the plan will indeed keep assuming normal export during negative-price slots even if your automation is curtailing it, exactly as reported.

**Closely related — please see #3986** ("model a rate-conditional export limit (curtailment under a price threshold)"), which covers the *plan-adjustment* half of this request in detail and already has an open PR: #4023. That PR adds `select.predbat_curtail_on_negative_export_price` (modes `curtail_excess` / `solar_production_off`, triggered whenever the export rate goes negative) so the plan correctly models curtailed export instead of booking a phantom export credit, plus a fix so PV calibration doesn't mistake curtailment for panel underperformance.

One difference worth flagging: #3986/PR #4023 is deliberately **modelling-only** — it does not have Predbat drive the inverter's zero-export switch itself (that was considered and rejected there as inverter-specific and a new control failure mode, left to the user's own automation). Your request as titled also asks for Predbat to *control* that switch, which isn't covered by #4023. So I'm not closing this as a straight duplicate — the "assume/plan around curtailment" half looks like it will be resolved by #3986 once merged, but the "Predbat actively controls the zero-export entity" half remains a distinct, unaddressed ask here.

In the meantime, the existing partial workaround is the [manual API](https://springfall2008.github.io/batpred/manual-api/) to override `export_limit` from your own automation (as noted on #3986) — with the same caveat that Predbat won't know the override will later be reverted, so it plans as if it's permanent.

No missing-information request — this is a clear feature request with sufficient detail.

springfall2008, 2026-10-29:
Closing in favour of #3986, which covers the same need: modelling export curtailment when the export price is negative so that the plan matches what the inverter actually does. PR #4023 implements it. Please follow and comment there.

As noted in that thread, an automation can already switch the inverter's zero-export mode and use the [manual API](https://springfall2008.github.io/batpred/manual-api/) to override `export_limit`, so that Predbat plans with the limit in force.

closed 2026-10-29 by springfall2008