springfall2008/batpred #2447: Mismatch of load forecast - or my misunderstanding

jrviz, 2025-06-10:
For a while now I have been seeing high levels of load forecast, especially in the evening (but high at other times too). I have a load power sensor set up and this is working and set up in apps.yaml. A lot of the time my load is 5-700w but I see regular forecasts in predbat of 2-5kW.

Now, it mat be a misunderstanding of what the load column means and this is including export but I see high levels of load forecast even when not exporting.

So with these plans

[image]

[image]

and underlying forecast graph

[image]

they do not seem to line up with the power sensor history (this one set in apps.yaml)

[image]

Predbat seems to show high load forecast but my power history (which is typical) is way below that. Have I got something set up wrongly or am I misunderstanding the load numbers in Predbat? Or is there something wrong with the data that is stored for the forecast? (set at 10 days previous, weight of 1 and no load scaling)

My concern is these high (but wrong) load forecast figures mean that suboptimal plans are being generated.

johnwb87, 2025-06-11:
Predbat requires energy for the load sensors. To be specific load_today in apps.yaml should be a kWh sensor, not W.

johnwb87, 2025-06-11:
What are you using for days previous? Are you doing any car charging?

jrviz, 2025-06-11:
10 or 14 days previous (I'm keeping 14 days). No car charging.

I'm looking at a possible issue with the sensor used for load_today. I'll investigate further but if it is the case then that means the template for solaredge will need to be changed (if you have batteries). I'll report back.

jrviz, 2025-06-12:
OK, I found it. There is a known issue (just not by me :-)) with one of the power sensors in Solaredge where power has the battery discharge added into it. There is an approach using a calculated value as an alternative and the documentation/template should be updated. I'll close this now and re-open another thread which has been used to update Solaredge documents.

closed 2025-06-12 by jrviz