springfall2008/batpred #2448: Plans exporting during the day

markbw999, 2025-06-10:
As posted in the Facebook group... I'm really struggling with the plans predbat is creating since v8.20. It used to charge my batteries overnight, maintain during the day, export excess solar, and discharge in the evening. Now it is exporting at least 60% in the morning and sometimes much more than that leaving very little contingency for poor Solcast or an unexpected laundry load. Also, I'm getting no iBoost in the plan because all the solar is going to recharging the batteries. What do I need to tweak to get closer to the old plans?

As requested I have attached predbat_debug but saved as a text file as I couldn't upload it with a yaml extension.

[predbat_debug.txt](https://github.com/user-attachments/files/20680364/predbat_debug.txt)

johnwb87, 2025-06-11:
That is a particularly bad solar prediction. Is it often that far off?

My next question would be do you have carbon enable turned on? System carbon intensity was very high this morning so predbat may have taken that into consideration for it's export choices. It certainly did for me, which personally is what I want and I can rely on my solar forecast quite well.

markbw999, 2025-06-11:
I get about one day a month on average where Solcast is that far off. I'm doing six API calls a day and it has now updated itself to something more reasonable.
I agree with you that my priorites aren't all about cost, hence enabling carbon intensity. Also, it doesn't make much economic sense to run my solar diverter (iBoost) instead of exporting the excess solar but I prefer to sacrifice a few pence not to burn gas to heat the hot water.
I'm on Intelligent Octopus Go.

markbw999, 2025-06-11:
I used to have my own automation to call Solcast but since Trefor built it into predbat I rely on that. If I created a separate call at sunrise will that work with the predbat functionality?

johnwb87, 2025-06-11:
Hmm. Ok, I’m not familiar with how predbat handles solcast calls, sorry about that. I stayed with what I had and was working.

johnwb87, 2025-06-11:
> I have the same issue. I would like to see a metric for just holding battery charge, so you could set something like "1p per kWh.hour of battery charge kept". That would add an intrinsic preference to plans which discharged in the evening over the morning, and disfavour plans that meant the battery would have a low level for long periods of time, which isn't great for contingencies (especially for those of us who have batteries capable of Island Mode/EPS operation)

I believe Metric Battery Value Scale achieves this. I’ve not played with it but probably start at 1.1 and see how you go.

peter1rhodes, 2025-06-11:
> I believe Metric Battery Value Scale achieves this. I’ve not played with it but probably start at 1.1 and see how you go.

I thought Metric Battery Value Scale only affects how leftover energy in the battery at the end of the plan (e.g. 24 hours ahead) is valued, and has nothing to do with valuing holding a higher SoC during the plan period? Happy to be corrected!

johnwb87, 2025-06-12:
Reduce the metric carbon value and see what happens. I’m surprised you are seeing this behaviour at 5p/Kg I have to be up at 30+ to get it to do morning export.
And your load scaling metrics are all above default so that should be giving you more headroom.

markbw999, 2025-06-13:
Here's some more evidence. I took this snapshot of my plan for today at about 17:00 yesterday while on v8.20.14.

[image]

I then reverted back to v8.19.14. Without changing any other parameters I got this plan.

[image]

Clearly there is a significant difference in the way the two versions model my specific setup.

WyndStryke, 2025-06-13:
> One thought given what you said about solcast forecast. Are you using the templated automation to refresh Solcast? That doesn't actually do a sunrise refresh with the way the logic works.

I removed the sunset / sunrise check from the automation, and that fixed it.  It's OK in winter, but the problem occurs after March, because it is actually comparing the next day's sunrise against the current time, and it'll be a minute or two earlier.

In my variant, I do one forecast in the early morning (near the start of the cheap overnight import period), and then the remaining 4 forecasts get spread over the day.  The early morning forecast is so that it can get a more recent forecast when deciding how much to import overnight.

I want to adjust it to take account of 419 errors, so that if the early morning forecast does not work, it'll do one additional forecast during the day, and if a daytime forecast got missed, it'll do an additional forecast towards the end of the day (and skip the next early morning forecast).

springfall2008, 2025-06-14:
Looking at this the export in the day is because you have a AC coupled inverter there is quite a bit of round trip losses, exporting solar as its generated avoids all of that. I might ask if you have accounted for losses in the PV inverter in your forecast or pv_scaling however?

Setting best_soc_keep to say 1.0 will prevent the flat battery if that's what you prefer also.

markbw999, 2025-06-15:
So where do we go from here? v8.19 is working for me at the moment but clearly I don't want to be on an outdated version for long.

markbw999, 2025-06-17:
Here is my last comment as I'm going to close the issue. I tried upgrading to the latest version (v8.21.2) yesterday and miraculously the plans are back to how I want them. The only factor now different is my carbon intensity forecast broke when I upgated HA to 2025.6.1 so predbat is not taking that into account. I can only conclude that carbon intensity was screwing up my plans when I opened this issue. If I ever get it working again and the problem comes back I'll open an issue specific to that problem.

closed 2025-06-17 by markbw999

markbw999, 2025-06-17:
> > > I'll open a new issue for my related problem, as this is not solved for me
> > 
> > 
> > Sorry to hear you are still having issues. Are you using carbon intensity? If so, maybe you could test my theory that it's that is causing the problem. Could you temporarily comment it out in your apps.yaml and see if that makes a difference to your plan?
> 
> I am, but with quite a low value (20p/kg, which in the north-east where the difference across the day is usually just 10g/kWh to 20g/kWh, is not a lot), and anyway, the plan routinely discharges in the morning even when the grid is cleaner than later in the day. If the carbon metric is to blame, then there's a fault with it!

I was on an even lower value and also not experiencing high intensity when having problems with the plan. The only difference in my setup now is the carbon intensity integration is broken. My unproven suspicion is a bug was introduced in v8.20. You might be able to prove that (or otherwise).

peter1rhodes, 2025-06-17:
For example, right now:

[image]

markbw999, 2025-06-20:
Hi Peter, predbat is back to producing crap plans for me (exporting down to 4% during the day). In case you are interested I have come up with the following contingency plan via automations...
1 - At sunrise set reserve min to 25%
2 - If I start importing more than 250W from the grid at peak rate, set reserve min to 4%
3 - An hour before sunset set reserve min to 4%
This should leave me with some contingency during the day if solcast is way off and still export the remaining energy in my batteries in the evening. I shouldn't have to do this but needs must.