springfall2008/batpred #2449: import_limit setting requirement

Noodleyman, 2025-06-11:
**Is your feature request related to a problem? Please describe.**
I have a SolarEdge system with two batteries and a large 16kW array. Both can be charged in parallel with each battery having a maximum charge rate of 5kWh, thus the total charge rate is 2x5kWh, thus 10kWh. However, due to DNO requirements due to very rural with poor grid capabilites, I need to keep my grid import limit at 7.2kWh. This number keeps me from pulling too much power from the grid when charging, considering other background power requirements from the grid. PredBat seems to only have the setting "inverter_limit_charge" and no import_limit setting. 

inverter_limit_charge limits the rate at which batteries can be charged via PV, not only grid. Thus on bright days the battery charge rate is is being limited to 7.2kWh even though it can charge at 10kWh.

**Describe the solution you'd like**
A setting to explicitly define import_limit, to work like th export_limit setting, but for imports as the grid import limit may vary from inverter charge limits.

**Describe alternatives you've considered**
Have not been able to find a workaround without putting Predbat into read only mode and taking control of the system via automations.

**Additional context**
I think the above describes the requirement, althogh happy to add more details if any specific questions are raised.

arturonaredo, 2026-10-11:
Adding a concrete use case for this request, also relevant to #1199's proposed first step of a flat site-import limit.

## Environment and observed limitation

- Predbat **v9.3.7** (`303d592bd5f1b707201c4da70b9ab3308fd10758`).
- Deye hardware configured as **`inverter_type: SK`**, through local Solarman. **Not Deye Cloud.**
- Native Predbat-led EV planning; a single Home Assistant automation follows its charging-slot signal and applies physical safety guards. No external battery/EV optimiser.
- Whole-site import ceiling: **6,000 W**.
- Inverter limit: 16,000 W; battery charge/discharge model limits: 3,000/8,192 W.
- Battery capacity: 32.1536 kWh, with a **10% reserve (3.21536 kWh, not 0.1 kWh)**.
- EV input is approximately **6.77 kW at a fixed rate**; there is no validated current-control interface. Lowering the configured prediction rate would not throttle the actual charger.
- EV load is inside the house CT and `car_charging_from_battery` is enabled. Export compensation is zero.

I observed a native **forecast**, not a measured overload, with five future grid-power points above 6 kW and a peak of 6.333 kW. The installed Python prediction path accumulates residual grid demand without a whole-site import constraint: [v9.3.7 prediction.py, lines 1359–1394](https://github.com/springfall2008/batpred/blob/v9.3.7/apps/predbat/prediction.py#L1359-L1394).

The separate `inverter_limit_charge`, `inverter_limit_discharge` and `battery_rate_max` settings are not a joint grid-import cap. Hardware peak shaving or a relay cutoff may prevent execution, but that does not make the forecast or its advertised EV completion feasible.

There was also a **separate household-history quality problem** in this installation. I have removed its contaminated ML forecast from active planning and selected recent measured native history. The observed forecast above is therefore supporting context, not a controlled quantitative reproduction of this feature in isolation. Correcting load history does not introduce the missing constraint. This report also makes **no claim about the cause of an inverter backup/communication fault**.

## Expected configurable behaviour

An optional, whole-site parameter such as `import_limit: 6000` (W; disabled by default) should model the combined demand of the house, EV and battery charging, after PV and physically available battery support:

1. Preserve energy balance. Do **not** simply clip the displayed import curve or silently discard EV/house demand.
2. Respect battery charge/discharge rates, losses and reserve. Curtailing battery grid charging must not unnecessarily reduce otherwise available PV charging.
3. Prefer a feasible battery/EV plan over a cheaper infeasible one; prepare battery energy automatically where possible, without a fixed arbitrary reserve or manual SoC override.
4. Do not assume the battery always supplies only the portion above the grid limit: dispatch/hold behaviour must match the inverter control model.
5. For fixed-rate EV charging, move or defer executable slots as necessary. If the connection, power, stored energy or deadline makes the request impossible, report an explicit shortfall/infeasibility rather than promising completion.
6. Keep Python and C++ predictions consistent, or explicitly disable an unsupported kernel path when the feature is enabled. An enabled setting must never be silently ignored.

For example, with no PV and the battery at its reserve, 1 kW of household demand plus a fixed 6.77 kW EV requires 7.77 kW of import: it cannot fit within 6 kW. This is a simple feasibility bound, **not a claimed simulator test result**. A viable alternative needs earlier battery preparation, a different EV slot or an explicit unmet request.

A first-stage draft can expose configuration and truthful feasibility diagnostics, but should be labelled as such: diagnostics alone do not enforce the limit, optimise EV slots, account for an executor's 15%/25% pause/restart hysteresis, or provide real-time electrical protection. The draft must not close this request until its stated scope is actually implemented and verified.

I am testing the proposal in an isolated fork; it is **not installed on the live inverter controller**. Tests use synthetic data, not private HA dumps, and no load, battery-disconnection or outage experiments are involved.

arturonaredo, 2026-10-11:
Draft follow-up: **#5486 — configurable whole-site import diagnostics (no enforcement)**.

The initial implementation adds a scalar `import_limit` threshold in W and reports
the best forecast's raw peak import, excess energy and violating duration. It does
not alter battery dispatch, fixed EV demand, costs or candidate ordering, and it
does not hide an over-limit plan by clipping the grid graph.

The draft includes synthetic reserve/energy-balance and EV tests, real cache/kernel
checks, mocked HA publication tests and a proposal for the later joint constraint.
The focused tests and full quick suite passed locally; four slow tests were skipped.

This is **not a complete solution to #2449 or #1199**, is not a real-time protective
limit and has not been deployed to live hardware. Native EV rescheduling, feasible
battery preparation and executable grid-charge curtailment remain follow-up work.
Both issues should remain open. Feedback on the diagnostics-first split is welcome.

gcoan, 2026-10-11:
Just one comment on the original ask

> Describe the solution you'd like
> A setting to explicitly define import_limit, to work like th export_limit setting, but for imports as the grid import limit may vary from inverter charge limits.

I think the solution being asked for is not like export_limit.  Predbat relies on the inverter to enforce export_limit and the configuration in Predbat is so that Predbat can model this behaviour and include it in the plan.

Ideally an inverter would deliver similar import side behaviour because to do this properly requires dynamic control, throttling battery and/or solar in response to changing house load.  Predbat with its 5 minute cycle time would simply not be able to do this.

@arturonaredo thanks for the work so far on this, the above provided as food for thought as to how accurate and time-responsive Predbat can be in managing import load.
Adding such an enhancement does make sense, its just that a Predbat-based solution is going to have limitations and not be able to totally constrain import energy to the defined limit

arturonaredo, 2026-10-25:
Thanks @gcoan, I agree that Predbat cannot provide an instantaneous import limiter or replace electrical protection.

My home illustrates the planning problem: I use a **6 kW site-import threshold**, while my EV charges at a fixed approximately **6.77 kW**. I want to charge overnight, meet the car’s departure target, and retain enough domestic battery energy for household demand.

For example, with **1 kW of household consumption**, the house and car together need **7.77 kW**. Keeping grid import around 6 kW would require approximately **1.77 kW of battery support**. Over three hours, that represents **5.3 kWh of battery energy before losses**, in addition to preserving the domestic reserve. If the battery starts at reserve, that schedule cannot work as planned.

What I want Predbat to anticipate is that energy requirement: charge the domestic battery beforehand where feasible, avoid competing battery grid charging while the car needs that headroom, or choose different EV slots. If no suitable schedule is found, report the unmet target rather than promise completion.

Unexpected peaks will still happen, but historical household demand gives us a useful baseline for much of the overnight consumption. We cannot predict every appliance switching on, but we can avoid scheduling a **predictable, sustained overload**.

Since your comment, the work has progressed across three separate PRs:

- **#5486** remains diagnostics-only: report forecast import violations without changing the plan.
- **#5498** now implements experimental joint EV and domestic-battery planning, including modelled grid/PV sharing, battery reserve and power limits, and independent nominal/PV10 replay checks. This is the contribution addressing the overnight example above.
- **#5505**, which you already know from your documentation review, provides the separate Deye/Solarman template and HA-observed service sequences.

The distinction is **planning ahead for sufficient energy and power**, not guaranteeing an instantaneous ceiling. Fast local regulation and electrical protection remain separate, and the experimental planner does not guarantee physical departure-target completion.

Would that separation address your concern?

gcoan, 2026-10-25:
this all sounds good @arturonaredo, a set of logically complete enhancements that add to the core Predbat 'forecast/plan/schedule' architecture and don't try to bend Predbat into something it cannot easily be such as a dynamic import balancer.

janmiddelink, 2026-10-29:
The peak shaving on my Deye has two parameters a switch "Grid peak shaving" enabled/disabled and a scalar "Grid peak shaving power" in watt.  When enabled de import from grid will be limited to de grid peak shaving power. The actual power drawn from the grid during charge wil be dependent on the power from PV and the load.

mgazza, 2026-11-06:
Adding a verified Fox Cloud case where an enforced import ceiling makes the predicted battery refill unattainable alongside EV demand. This supports the existing modelling request; it is not a request for Predbat to replace instantaneous electrical protection.

**Kit and readbacks (6 November 2026, read-only):**

- Fox H1-5.0-E-G2-WL, EP12 battery with 11.52 kWh reported capacity. Battery charge capability approximately 5 kW.
- Fox's capability-first API exposes `basicParameter2.importLimitedPower` and `basicParameter2.importLimitCurrent` as readable fields.
- `/op/v1/device/setting/get` returned `importLimitedPower = 9200 W` and `importLimitCurrent = 100 A`. The separate peak-shaving read also returned 9200 W. These are current readbacks; we have not established who configured them or when.
- The older `/op/v0/device/setting/get` call using key `ImportLimit` returns errno 40257 on this device. That failure must not be interpreted as an absent/unlimited import ceiling. Capability-first reads succeed with errno 0. Documentation: https://www.foxesscloud.com/public/i18n/en/OpenApiDocument.html and https://ho0mrhk5pr.apifox.cn/497227750e0

**Observed overnight behaviour:**

| Time, Europe/London, 6 November | Grid import kW | Total house + EV load kW | Battery charging kW |
| --- | ---: | ---: | ---: |
| 04:00 | 9.201 | 8.471 | 0.729 |
| 05:00 | 9.199 | 8.740 | 0.458 |

These are aligned five-minute Fox telemetry samples, rounded as reported. EV demand was approximately 7 kW and is already included in the load column. Do not add it again. No PV was being produced at those times. The battery charge request/rating was 5 kW, but actual charging backed off as measured import approached 9.2 kW. End-window SOC was approximately 55%, rather than the requested 100%.

There were also separate economic planning decisions: Predbat scheduled force export around 00:30–01:30, before the recorded EV charging began around 02:00, and subsequently deferred battery charging for roughly an hour with `Hold for car`. The evidence does **not** show battery export concurrent with the EV in the aligned samples. The issue is that earlier export and later refill decisions need to account for attainable charging power, rather than assuming the battery can always refill at its standalone maximum.

**Requested modelling behaviour / acceptance:**

1. Represent an explicit whole-site grid-import allowance, distinct from inverter AC/DC ratings, battery charge capability, and export limits. Support a manual value and automatic provider readback where available, with units, provenance and freshness handled explicitly.
2. During each forecast interval, constrain battery grid charging using the remaining import headroom after house and EV demand. Include PV and conversion losses correctly; preserve legitimate PV charging above the grid allowance. For multiple inverters sharing a meter, apply the shared allowance once rather than once per inverter.
3. Feed attainable charge energy into SOC prediction, charge-window selection and export decisions. Prefer earlier refill or less export where necessary; report an infeasible target rather than promising completion.
4. Retain the inverter/charger's fast local limits and electrical protection. Do not silently raise physical limits or treat a current-limit configuration value as proof of the installed fuse/wiring rating.
5. Replay a case with a 9.2 kW site allowance, about 7 kW EV demand, variable household demand and a 5 kW battery charger, checking predicted SOC against constrained delivered energy.

A constant battery-rate reduction is only a conservative workaround: it cannot represent varying house/EV headroom and can unnecessarily constrain charging when the car is off. No customer settings were changed during this investigation, and no delivery date is implied.