springfall2008/batpred #2460: Use of additional weather sources for Predheat temperature forecast (specifically Met Office)

Timeflies1980, 2025-06-15:
**Is your feature request related to a problem? Please describe.**
In Predheat the only weather source for forecast temps is openweathermap. This is not accurate for my location. I would like to be able to use the data provided via the UK Met Office integration. 

**Describe the solution you'd like**
# Weather data
    weather: weather.met_office_stafford

I would like to be able to use the weather entity as provided my the met office integration as above 

**Describe alternatives you've considered**
None

**Additional context**
The Met office integration provides hourly temp data as standard. This could be extended to the local weather data providers. 

[image]

TCWORLD, 2025-11-19:
Don't know if you've tried this, but seems to work fine for me.

I have the met office integration in HA, and then In predheat config I put:

`weather: weather.met_office_chapel_allerton`

Seems to work fine - predheat gets the needed external temperature forecast.

gcoan, 2025-11-22:
Documentation and predheat template updated to explain how other weather services such as Met Office can be used

closed 2025-11-22 by gcoan